MichaIng/DietPi #7286: Bundled 4G module doesn't work on NanoPi NEO3 (aarch64) (but does on NanoPi NEO)

mszyndel, 2024-11-22:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.8.0` fresh off the downloads
- Distro version | `bookworm`
- Kernel version | `Linux DietPi 6.6.56-current-rockchip64 #1 SMP PREEMPT Thu Oct 10 10:50:06 UTC 2024 aarch64 GNU/Linux`
- SBC model | `NanoPi NEO3 (aarch64)`
- Power supply used | 5V 1A RAVpower
- SD card used | SanDisk 32G

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) N/A
- Was the software title installed freshly or updated/migrated? N/A
- Can this issue be replicated on a fresh installation of DietPi? YES

- Bug report ID | N/A
- 
#### Steps to reproduce

Boot DietPi on NanoPi NEO3 bundled with 4G module. Usb 4G device is not detected, no eth1.

#### Expected behaviour

DietPi should detect USB and run it using cdc_ether/rndis_host driver
It does work on Debian-based distro provided by the manufacturer, also works on DietPi on another board out of the box (see below).

Doesn't look like manufacturer bundles any special drivers, but I wasn't able to solve this myself. I tried stuff like `modprobe ...` adding all modules that I see used on NanoPI NEO, but to no avail.

#### Actual behaviour

USB device/4G modem is not detected at all

#### Extra details

Photo of the setup

[image]

**lsusb**
```
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**lsmod**
```
Module                  Size  Used by
snd_soc_pcm5102a       12288  1
snd_soc_simple_card    20480  0
snd_soc_simple_card_utils    24576  1 snd_soc_simple_card
snd_soc_rockchip_spdif    12288  0
snd_soc_rockchip_i2s    24576  2
snd_soc_core          208896  5 snd_soc_pcm5102a,snd_soc_rockchip_spdif,snd_soc_simple_card_utils,snd_soc_rockchip_i2s,snd_soc_simple_card
snd_compress           24576  1 snd_soc_core
snd_pcm_dmaengine      12288  1 snd_soc_core
snd_pcm               106496  4 snd_compress,snd_soc_simple_card_utils,snd_soc_core,snd_pcm_dmaengine
snd_timer              36864  1 snd_pcm
rk_crypto              28672  0
snd                    77824  4 snd_timer,snd_compress,snd_soc_core,snd_pcm
rng_core               16384  1 rk_crypto
soundcore              12288  1 snd
cpufreq_dt             16384  0
dm_mod                126976  0
ip_tables              28672  0
x_tables               36864  1 ip_tables
autofs4                40960  2
[... 494 lines cut ...]
cdc_ether              16384  1 rndis_host
usbnet                 32768  2 rndis_host,cdc_ether
sunxi                  12288  0
phy_generic            16384  2 sunxi
gpio_keys              16384  0
ac200_phy              12288  1
```

**dmesg -6 (selected lines)**
```
[    5.214210] usb 1-1: New USB device found, idVendor=19d1, idProduct=0001, bcdDevice= 2.00
[    5.222631] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    5.229878] usb 1-1: Product: EigenComm Compo
[    5.234295] usb 1-1: Manufacturer: EigenComm
[    5.238606] usb 1-1: SerialNumber: 000000000001
[    6.605343] usbcore: registered new interface driver cdc_ether
[    6.666027] rndis_host 1-1:1.0: rndis media connect
[    6.735450] rndis_host 1-1:1.0 eth1: register 'rndis_host' at usb-1c1b000.usb-1, RNDIS device, 20:89:84:6a:96:ab
[    6.746377] usbcore: registered new interface driver rndis_host
[   14.193924] cdc_acm 1-1:1.2: ttyACM0: USB ACM device
[   14.249228] cdc_acm 1-1:1.4: ttyACM1: USB ACM device
[   14.284910] cdc_acm 1-1:1.6: ttyACM2: USB ACM device
[   14.285106] usbcore: registered new interface driver cdc_acm
[   14.285117] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

MichaIng, 2024-11-22:
On the NEO3, the USB device is not detected at all, hence missing driver is not the issue. I guess the pins need to be configured. Let me check for device tree overlays.

mszyndel, 2024-11-22:
Thanks for blazing fast response! Please let me know if I can provide any more details. Documentation for 4g module is sparse to nonexistent but here’s friendlyelec page for the board https://wiki.friendlyelec.com/wiki/index.php/NanoPi_NEO3

MichaIng, 2024-11-22:
Sadly here is no overlay coming with the kernel, to toggle USB support on GPIO: 
[image]

The Allwinner kernel has interfaces for usbhosts, which we enable by default for some SBCs, including the NanoPi NEO. See `/boot/dietpiEnv.txt`.
[image]

I'll check the source of those overlays, maybe we can try to create similar ones for the RK3328.

MichaIng, 2024-11-22:
Please try this:
```sh
apt install device-tree-compiler
mkdir -p /boot/overlay-user
cd /boot/overlay-user
cat << '_EOF_' > usbhost.dts
/dts-v1/;
/plugin/;
/ {
        compatible = "rockchip,rk3328";
        fragment@0 {
                target = <&usb20_otg>;
                __overlay__ {
                        dr_mode = "host";
                };
        };
};
_EOF_
exec dtc -I dts -O dtb -o usbhost.dtbo usbhost.dts
G_CONFIG_INJECT 'user_overlays=' 'user_overlays=usbhost' /boot/dietpiEnv.txt
reboot
cat /proc/device-tree/usb@ff580000/dr_mode; echo
```
The last command is to check whether this USB port has been switched to host mode.

It is weird:
- This node is defined in mainline Linux NanoPi R2S device tree (which is used as basis for the NEO3 device tree), and while it is called `usb20_otg`, in it is configured to be in host mode, instead of OTG:
   - https://github.com/armbian/build/blob/main/patch/kernel/archive/rockchip64-6.6/dt/rk3328-nanopi-neo3-rev02.dts 
   - https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3328-nanopi-r2s.dts?h=linux-6.6.y#n385
- Armbian patches this device tree to not set it to host mode anymore, which results in it being in OTG mode: https://github.com/armbian/build/blob/main/patch/kernel/archive/rockchip64-6.6/board-nanopi-r2s.patch#L706
- OTG mode should generally be able to automatically act as both, host or device mode. But there are cases/devices where it does not work. So maybe it is the case here, and I hope it is the correct node, as the pins actually serve two USB 2.0 ports, with one of them clearly intended to be in OTG mode by default.

But let's see ...

mszyndel, 2024-11-23:
```
root@DietPi:~# cat /proc/device-tree/usb@ff580000/dr_mode; echo
otg
root@DietPi:~# lsusb
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

MichaIng, 2024-11-23:
Hmm, it worked fine here, and the `/proc/device-tree/usb@ff580000/dr_mode` output verified the mode switch. Also even if an overlay is broken, and causes an error when being applied, the boost script just reverts to the original device tree 🤔.

... ah, I see there was an `exec` in my command, which replaces the shell process with the `dtc` command. Copy&paste from another script 😅. However, it terminates the shell session, but another SSH login did work in my case, to repeat the steps without `exec`, resp. from the `dtc` command on.

mszyndel, 2024-11-24:
Sorry, I didn't have time to connect via UART. Your fix works, as in USB device/network card is detected successfully. However, 4G modem is detected before LAN and therefore I couldn't ssh into the box.

Here's some more info

First of all, it works!
```
cat /proc/device-tree/usb@ff580000/dr_mode; echo
host
```

```
# lsusb
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 19d1:0001 BYD EigenComm Compo
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Welcome message shows following
```
curl: (28) Resolving timed out after 3000 milliseconds
```

```
# dmesg | grep eth
[    3.357006] rndis_host 1-1:1.0 eth0: register 'rndis_host' at usb-ff580000.usb-1, RNDIS device, 20:89:84:6a:96:ab
```

```
ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 20:89:84:6a:96:ab brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.2/24 brd 192.168.10.255 scope global dynamic eth0
       valid_lft 86125sec preferred_lft 86125sec
    inet6 fe80::2289:84ff:fe6a:96ab/64 scope link
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether d2:8a:a2:8b:b5:86 brd ff:ff:ff:ff:ff:ff
```

ping works when I specify the interface

```
root@DietPi:~# ping 8.8.8.8
ping: connect: Network is unreachable
root@DietPi:~# ping -Ieth0 8.8.8.8
PING 8.8.8.8 (8.8.8.8) from 192.168.10.2 eth0: 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=113 time=151 ms
64 bytes from 8.8.8.8: icmp_seq=2 ttl=113 time=44.8 ms
64 bytes from 8.8.8.8: icmp_seq=3 ttl=113 time=44.4 ms
64 bytes from 8.8.8.8: icmp_seq=4 ttl=113 time=51.8 ms
```

I'll poke around a little more now that we're safely in userland 😅

mszyndel, 2024-11-25:
Okay, to make it work I did following changes to network interfaces

```
root@DietPi:~# cat /etc/network/interfaces
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

auto eth1

# 4G
allow-hotplug eth0
iface eth0 inet dhcp
address 192.168.10.2
netmask 255.255.255.0
gateway 192.168.10.1
# dns-nameservers 9.9.9.9 149.112.112.112

# Ethernet
allow-hotplug eth1
iface eth1 inet dhcp
address 192.168.0.101
netmask 255.255.255.0
gateway 192.168.0.1
dns-nameservers 8.8.8.8 9.9.9.9
```

MichaIng, 2024-12-12:
Sorry for the late reply. So The 4G module shows up as `eth0`, that is interesting. That also broke the network connection, I guess, as previously `eth0` was the Ethernet adapter.

`auto eth1` should not be needed, as `allow-hotplug eth1` configures it already, once detected.

Since you have both interfaces configured with DHCP, is an Ethernet cable even connected, and if so, which of the two is then used is default gateway?
```sh
ip r l 0/0
```

So the solution to get the $G module detected as USB device was the overlay, right
https://github.com/MichaIng/DietPi/issues/7286#issuecomment-2495112813

mszyndel, 2024-12-12:
> So the solution to get the $G module detected as USB device was the overlay, right [#7286 (comment)](https://github.com/MichaIng/DietPi/issues/7286#issuecomment-2495112813)

Yes

mszyndel, 2025-11-02:
Sorry for late reply. I reinstalled most recent DietPi from scratch on two boards (I think 1GB and 2GB RAM, although they are visually different as well).

Results are the same as previously.
1. Upon install eth0 is ethernet and it works.
2. After applying the overlay from https://github.com/MichaIng/DietPi/issues/7286#issuecomment-2495112813, after reboot ethernet doesn't work
3. I connect via UART I find ethernet is `eth1` and `eth0` is now 4G board.

```
dietpi@DietPi:~$ ip r l 0/0
default via 192.168.10.1 dev eth0
dietpi@DietPi:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 20:89:84:6a:96:ab brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.2/24 brd 192.168.10.255 scope global dynamic eth0
       valid_lft 85829sec preferred_lft 85829sec
    inet6 fe80::2289:84ff:fe6a:96ab/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether d2:8a:a2:8b:b5:86 brd ff:ff:ff:ff:ff:ff
```

I bring eth1 up
```
dietpi@DietPi:~$ sudo ip link set eth1 up
dietpi@DietPi:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ether 20:89:84:6a:96:ab brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.2/24 brd 192.168.10.255 scope global dynamic eth0
       valid_lft 85826sec preferred_lft 85826sec
    inet6 fe80::2289:84ff:fe6a:96ab/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether d2:8a:a2:8b:b5:86 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::d08a:a2ff:fe8b:b586/64 scope link tentative proto kernel_ll
       valid_lft forever preferred_lft forever
```
but default route is still through eth0 (4G modem)
```
dietpi@DietPi:~$ ip r l 0/0
default via 192.168.10.1 dev eth0
```
Maybe I was unclear in my previous comment https://github.com/MichaIng/DietPi/issues/7286#issuecomment-2498468385. by default `etc/network/interfaces` looks like this (missing eth1 entirely)
```
# Location: /etc/network/interfaces
# Please modify network settings via: dietpi-config
# Or create your own drop-ins in: /etc/network/interfaces.d/

# Drop-in configs
source interfaces.d/*

# Ethernet
allow-hotplug eth0
iface eth0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 9.9.9.9 149.112.112.112

# WiFi
#allow-hotplug wlan0
iface wlan0 inet dhcp
address 192.168.0.100
netmask 255.255.255.0
gateway 192.168.0.1
#dns-nameservers 9.9.9.9 149.112.112.112
wireless-power off
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf`
```

I admit I'm bit out of my depth here, but please ask if there's any more info I can provide.

MichaIng, 2025-11-15:
I'll mark the issue as closed, since the device tree overlays have been added. But I bookmarked this for later reference regarding `ifplugd`. Feel free to test and report back how it goes, so we may fine tine the instructions for later use.