MichaIng/DietPi #7288: package ascii-image-converter says color is not supported

voidnonymous, 2024-11-24:
### Creating a bug report/issue
- [yep] I have searched the existing open and closed issues

#### Required Information
- DietPi version | Core 9 Sub 8
- Distro version | Bookworm
- Kernel version | Linux DietPi 6.1.75-vendor-rk35xx #1 SMP Thu Jul 25 14:36:35 UTC 2024 aarch64 GNU/Linux

- SBC model | Orange Pi 5 Plus (aarch64)
- Power supply used | No-Name 30W Power Brick (Works fine)
- SD card used | TEAM 128GB

#### Additional Information (if applicable)
- ascii-image-converter
- Freshly Installed
- Can be replicated on fresh install

#### Steps to reproduce

1. Do `sudo apt install ascii-image-converter`
2. Run it with the -c flag for color not via ssh
3. Shows the error that your terminal does not support 24-bit color or 8-bit color

#### Expected behaviour
Should allow me to get color on a Local HDMI Screen
#### Actual behaviour
Gives me an error about color saying it is not supported by my terminal.
#### Extra details
I'm not using a Desktop, just using the terminal.

MichaIng, 2024-11-24:
The supported colors depend on the used terminal, i.e. what the `TERM` variable contains. That one is passed through from the SSH client, at that one needs to interpret color codes and be able to display related colors.

PuTTY uses `xterm` by default, which supports 8 ANSI colors only. It can be adjusted in its settings, but I never tried whether PuTTY does support a wide range, when using `xterm-256color`.

voidnonymous, 2024-11-24:
SSH Works fine, just using ubuntu's built-in ssh command. It's locally where it gives the error

MichaIng, 2024-12-12:
Sorry for the late reply. So the terminal on the Ubuntu client seems to be limited then. Which one is it?
```sh
echo $TERM
```