MichaIng/DietPi #7290: Various external USB WiFI dongles. Fresh rip 4 image

oxoocoffee, 2024-11-24:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `bookworm 0`
- Kernel version | `uname -a`
- SBC model | `RPi 4 Model B (aarch64)` or (EG: RPi3)
- Power supply used | (Power supply that also did come wt my rPi5 kit)
- SD card used | (SanDisk)

#### Additional Information (if applicable)
- Software title | `kernel`
- Was the software title installed freshly or updated/migrated? `Fresh Install`
- Can this issue be replicated on a fresh installation of DietPi? `Yes`

- Bug report ID | `d714ba79-6dc9-4de3-9515-fd8e92223326`

#### Steps to reproduce

1. Insert different USB Wifi Dongles to USB (blue) port which should not matter
2. `ifconfig -a` shows no `wlan0`

#### Expected behaviour

- I would expect `wlan0` show up as a un-configured device

#### Actual behaviour

- No devices found. I will post dongle models, insert kernel log lines, and `lsusb` for each

#### Extra details

I am experimenting with a different WiFi Dongles, since I a trying to steal away from hacking the external WiFI antenna on PCB for a better signal in metal cases.

I did not make any configuration changes in `dietpi-config / Network Options: *`.
Under `Adapter Options` I have default settings and for below scenarios they do stay as they are below.

```
    Ethernet     : Available | [On] | Connected
    WiFi         : Not Found | [Off] | Disconnected
    Onboard WiFi : [Off]
```

I have few USB WiFI dongles that after insertion, automatically show `wlan` in `ifconfig`

Below is the list of what looks like are devices which running kernel does not support.
All links are to Amazon store, and links been sensitized :)

**Would it be to merge some to next update  so we do not have to look for a correct source version and compile the submodules?**

- [AC600 Mbps Dual Band Internet USB WiFi Adapter](https://www.amazon.com/600Mbps-Internet-Wireless-Receiver-Compatible/dp/B0CGJGTZLQ/) 🔴 **Not Working**
    ```bash
    $ sudo dmsg -T --follow-new
    [Sun Nov 24 22:19:23 2024] usb 1-1.2: new high-speed USB device number 7 using xhci_hcd
    [Sun Nov 24 22:19:23 2024] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
    [Sun Nov 24 22:19:23 2024] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [Sun Nov 24 22:19:23 2024] usb 1-1.2: Product: 802.11ac NIC
    [Sun Nov 24 22:19:23 2024] usb 1-1.2: Manufacturer: Realtek
    [Sun Nov 24 22:19:23 2024] usb 1-1.2: SerialNumber: 123456
    ```

    ``` 
    $ lsusb
    Bus 001 Device 006: ID 0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC
    ````

- [USB-AX55 Nano AX1800 USB WiFi6  ](https://www.amazon.com/ASUS-802-11ax-Frequency-USB-AX55-Nano/dp/B0BQ1DLLCM)
   ```bash
   $ dmesg -T --follow-new 
   [Sun Nov 24 22:47:35 2024] usb 1-1.1: new high-speed USB device number 9 using xhci_hcd
   [Sun Nov 24 22:47:35 2024] usb 1-1.1: New USB device found, idVendor=0b05, idProduct=1a62, bcdDevice= 0.00
   [Sun Nov 24 22:47:35 2024] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
   [Sun Nov 24 22:47:35 2024] usb 1-1.1: Product: 802.11ax WLAN Adapter
   [Sun Nov 24 22:47:35 2024] usb 1-1.1: Manufacturer: Realtek
   [Sun Nov 24 22:47:35 2024] usb 1-1.1: SerialNumber: 00e04c000001
   ```
   
   ```bash
   $ lsusb
   Bus 001 Device 009: ID 0b05:1a62 ASUSTek Computer, Inc. 802.11ax WLAN Adapter
   ```

- [Edimax Wi-Fi 4 802.11n Adapter for PC N150 Nano USB Adapter](https://www.amazon.com/dp/B08F2ZNC6J) ❤️ **Working**
   ```bash
   $ dmesg -T --follow-new 
   [Sun Nov 24 22:59:10 2024] usb 1-1.2: new high-speed USB device number 10 using xhci_hcd
   [Sun Nov 24 22:59:10 2024] usb 1-1.2: New USB device found, idVendor=7392, idProduct=b811, bcdDevice= 0.00
   [Sun Nov 24 22:59:10 2024] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
   [Sun Nov 24 22:59:10 2024] usb 1-1.2: Product: Edimax N150 Adapter
   [Sun Nov 24 22:59:10 2024] usb 1-1.2: Manufacturer: Realtek
   [Sun Nov 24 22:59:10 2024] usb 1-1.2: SerialNumber: 08BEAC441FD9
   [Sun Nov 24 22:59:10 2024] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
   [Sun Nov 24 22:59:10 2024] usbcore: registered new interface driver r8188eu
   ```
   
    🔴 **Important:** Interesting line here in above output
    ```
    [Sun Nov 24 22:59:10 2024] r8188eu: module is from the staging directory, the quality is unknown, you have been warned. 
   ```
   
   ```bash
   $ lsusb
   Bus 001 Device 010: ID 7392:b811 Edimax Technology Co., Ltd Edimax N150 Adapter
   ```
   
   ```bash
   wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
   ```

- [EDUP USB WiFi Adapter for PC](https://www.amazon.com/dp/B0827LG8L2) ❤️ **Working**
    ```bash
    $ sudo dmsg -T --follow-new
    [Sun Nov 24 22:33:27 2024] usb 1-1.2: new high-speed USB device number 8 using xhci_hcd
    [Sun Nov 24 22:33:27 2024] usb 1-1.2: New USB device found, idVendor=148f, idProduct=7601, bcdDevice= 0.00
    [Sun Nov 24 22:33:27 2024] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [Sun Nov 24 22:33:27 2024] usb 1-1.2: Product: 802.11 n WLAN
    [Sun Nov 24 22:33:27 2024] usb 1-1.2: SerialNumber: 1.0
    [Sun Nov 24 22:33:28 2024] cfg80211: Loading compiled-in X.509 certificates for regulatory database
    [Sun Nov 24 22:33:28 2024] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
    [Sun Nov 24 22:33:28 2024] usb 1-1.2: reset high-speed USB device number 8 using xhci_hcd
    [Sun Nov 24 22:33:28 2024] mt7601u 1-1.2:1.0: ASIC revision: 76010001 MAC revision: 76010500
    [Sun Nov 24 22:33:28 2024] mt7601u 1-1.2:1.0: Firmware Version: 0.1.00 Build: 7640 Build time: 201302052146____
    [Sun Nov 24 22:33:28 2024] mt7601u 1-1.2:1.0: EEPROM ver:0c fae:00
    [Sun Nov 24 22:33:29 2024] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
    [Sun Nov 24 22:33:29 2024] usbcore: registered new interface driver mt7601u 
    ```
    
    ```
    $ lsusb
    Bus 001 Device 008: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
    ```
    
    ```
    $ ifconfig -a
    wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
    ```

I have few others from other brand on the way. I will update the list when Iet them as well few others I did not have time yet to check.
I am not sure about `rPI 5 fresh` image

Joulinar, 2024-11-25:
best to test with new RPI kernel https://github.com/MichaIng/DietPi/issues/6676
This might have an improvement for some USB WiFi adapter

oxoocoffee, 2024-11-25:
Quick glance @ #7253 mentions Raspberry Pi 5: and I am running Raspberry Pi 4, which there are two different images AFAIK. Is #7253 is still applicable here?

Joulinar, 2024-11-25:
> is still applicable

Yes

oxoocoffee, 2024-11-26:
Ok, couple of things. This issue is in a way related to my ##7253 issue, where I did try to install AP using dietpi-software on rPI 5 clean image and I got similar errors.

As far this issue goes, The system is still booted as it was with all steps above.
I would like to remove external USB dangle and try internatl WIFI.
Should I post my findings to the other thread which is internal WIIFI and AP? And indicate that the information if for rPi 4? Start a similar issue like the ##7253 but for rPI 4?

Please note, all software was installed strictly using dietpi-software.

### On current running system rPI 4
-  Uninstall AP software using dietpi-software, I should uncheck the `WiFi Hotspot` option in `Advanced Networking`? Does this under cover include `purge` flag to apt? to remove all files? If not, which files I should remove to revert to clean version?
- When I uninstall `WiFi Hotspot` this should uninstall its dependent packages? I just want to make sure everything which was modified by installing `WiFi Hotspot` is removed.
- Remove external WIFI dongle, and reboot
- `sudo dietpi-config` and enable `Onboard WiFi`
- Verify that `wlan0` is enabled using `ifconfig -a`
- Install `WiFi Hotspot` and capture all output.
- If errors encountered, post if here or to the other thread?

I can make a fresh image of rPI 5 and capture all output when installing WiFi Hotspot on it. 
I should be able to do it tonight.

Any other notes or suggestions you recommend? For now without using external WiFI?

MichaIng, 2024-12-12:
@oxoocoffee 
Sorry for my late reply. To your questions:
- Uninstalling software via `dietpi-software` indeed purges all packages, hence removes all config files. Also other possible related files are removed.
- Yes, all dependent packages are purged.

I guess the issues are discussed in the new topic you opened 🙂.

StephanStS, 2025-01-07:
@oxoocoffee: Nice list of adapters. :+1: 
I have some more experiences listed there: https://dietpi.com/blog/?p=3266#issues_with_external_wifi_adapters (see them on the photo on top of the blog post).

MichaIng, 2025-01-08:
Closing this. Most will be solved once we migrate everyone to the new kernel stack. Will be offered with next DietPi release, and then enforced 1-2 releases later.

closed 2025-01-08 by MichaIng