MichaIng/DietPi #7291: Kernel version?

danergo, 2024-11-25:
My kernel version is 6.1.21-v8+ (DietPi v9.7.1).

I have a problem with an SPI-UART converter, for which I got a recommendation of updating the kernel.

I ran dietpi-update, it found a newer (9.8.0) version, but I couldn't find anywhere what kernel version is coming along with 9.8.0.

How can I know which kernel belongs to a certain DietPi version?

And how can I update the kernel on DietPi OS?

Joulinar, 2024-11-25:
DietPi is not a separate operating system. It is a set of bash scripts on a Debian-based image. And there is no relationship between a DietPi version and the kernel. The kernel version depends on your SBC and the Debian version. And kernel updates are done with `apt` like on any other system. I assume you have a Raspberry PI device? Is that correct? In that case, the kernel is the original kernel provided by the RPi guys. However, there have been major changes to the structure of the kernel and firmware that we are not yet making available to all users. Actually something you can do manually if you want to switch to the new kernel structure. See https://github.com/MichaIng/DietPi/issues/6676

Joulinar, 2024-11-26:
128MB are pretty much ok. And as a first step, you would need to use our script I linked above to migrate to new structure for kernel and firmware. Just using `rpi-update` will not do that.

danergo, 2024-11-26:
Huh, I was faster: just did a quick `rpi-update next`. It succeeded, now:

```
# uname -a
Linux rpi 6.12.1-v8+ #1822 SMP PREEMPT Mon Nov 25 14:19:09 GMT 2024 aarch64 GNU/Linux
```

But, now, my gpios are not working anymore:

```
# echo 0 > /sys/class/gpio/gpio25/value
-bash: /sys/class/gpio/gpio25/value: No such file or directory

# echo 25 > /sys/class/gpio/export
-bash: echo: write error: Invalid argument
```

What shall I do?

Joulinar, 2024-11-26:
> What shall I do?

Undo whatever you have done

danergo, 2024-11-26:
I got it now!

For others might be interesting:

```
Older kernels (6.1):
# echo 25 > /sys/class/gpio/export

Newer kernels (6.12):
# echo 537 > /sys/class/gpio/export
```

You can know which gpio is which with

```
cat /sys/kernel/debug/gpio
```

I have no idea on why is it modified like this :)

closed 2024-11-27 by danergo