MichaIng/DietPi #7296: Radxa Zero does not boot #2

way5, 2024-11-29:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
```
bookworm
```
- Kernel version | `uname -a`
```
Linux RZERO 6.6.54-current-meson64 #1 SMP PREEMPT Fri Oct  4 14:30:05 UTC 2024 aarch64 GNU/Linux
```
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
```
Radxa Zero (aarch64)
```
- Power supply used | (EG: 5V 1A RAVpower)
```
5V 3A
```
- SD card used | (EG: SanDisk ultra)
```
N/A
```

#### Steps to reproduce

1. Erase mmc0 and boot with rz-udisk-loader.bin or radxa-zero-erase-emmc.bin.
2. Install any of the latest compatible Linux distributions
3. At every boot it shows the following and keeps waiting until I manually type boot command.

The source topic is published here: https://forum.radxa.com/t/rzero-autoboot-from-emmc/24764

The issue supposedly already have a solution: https://github.com/CodeChenL/build/commit/12477803c5e20ea4fae0f22dc13817c8a9831deb#diff-7c8d1c058500641ab9f9a5b0d38785a5b72eec2f904b9d11774d5d6c60d0e0cdR17

Could these patches be applied to the current Dietpi build?

thank you

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2024-12-06:
To understand the issue better:
- The two binaries are used to flash images to the onboard eMMC?
- Does it work when you boot DietPi or Armbian from SD card, download and flash the image to the eMMC from there?

Because the two binaries use a different bootloader version and source, and the patches are as well for a different version, so I suspect the issue does neither exist with the Armbian bootloader builds, nor those of DietPi (which are the mostly the same). If so, then this is something which can only be solved by those who provide those `rz-udisk-loader.bin` or `radxa-zero-erase-emmc.bin` maintainers, respectively in the U-Boot sources they use. But if booting from internal eMMC does not work either, when flashing an Armbian or DietPi image directly, then it is in our U-Boot sources as well.

We maintain an Armbian fork to implement own fixes, if needed, hence can add those patches. But I do not have the board to test. @StephanStS you do, right?

And btw, do not believe everything Igor says about DietPi. He has an extreme aversion against DietPi, since it exists, and spreads a lot of outdated information, misinformation and plain lies about it, wherever he can. Like that we would not do kernel development, while we maintain a set of own kernel sources and builds for a number of SBCs, and a handful patches on top of Armbian sources, to fix issues, which he sadly is too proud to accept in their sources, at least not openly. But other than that, Armbian is great, of course: Mostly same underlying bootloader and kernel sources for most SBCs, we just have a different priority and different aims on the userland part, providing smaller images, but trying to simplify/speedup software installations with our scripts.

way5, 2024-12-06:
> To understand the issue better:
> 
> * The two binaries are used to flash images to the onboard eMMC?

I do not understand your question. I was flashing onto eMMC either Armbian or DietPi the results were the same. However @CodeChenL has fixed this issue for Armbian. There is no bootloader on both HW partitions (mmc0boot0, mmc0boot1) since they were erased by `radxa-zero-erase-emmc.bin` and were never flashed again. However it boots somehow, I suspect MBR from /mmc0/mmcblk1 takes the lead, but I already not sure (nobody could answer me) . 

> * Does it work when you boot DietPi or Armbian from SD card, download and flash the image to the eMMC from there?

I didn't try this, just because it is crucial to me it boots from eMMC. If it's somehow would help in debugging this issue I will try booting from SD.  

> Because the two binaries use a different bootloader version and source, and the patches are as well for a different version, so I suspect the issue does neither exist with the Armbian bootloader [...]

As I mentioned before it works already on Armbian. [This commit](https://github.com/CodeChenL/build/commit/12477803c5e20ea4fae0f22dc13817c8a9831deb#diff-7c8d1c058500641ab9f9a5b0d38785a5b72eec2f904b9d11774d5d6c60d0e0cdR17) has fixed the thing.

> 
> We maintain an Armbian fork to implement own fixes, if needed, hence can add those patches. But I do not have the board to test. @StephanStS you do, right?
> 
> And btw, do not believe everything Igor says about DietPi. He has an extreme aversion against DietPi, since it exists, and spreads a lot of outdated information, misinformation, up to plain lies about it, wherever he can. Like that we would not do kernel development, while we maintain a set of own kernel sources and builds for a number of SBCs, and a handful patches on top of Armbian sources, to fix issues, which he sadly is too proud to accept in their sources, at least not openly. But other than that, Armbian is great, of course: Mostly same underlying bootloader and kernel sources for most SBCs, we just have a different priority and different aims on the userland part, providing smaller images, but trying to simplify/speedup software installations with our scripts.

Alright guys :) I am taking noone's side here, it is important to me to fix those issues to have fully working board at last. The next issues after fixing the boot process is https://github.com/armbian/build/issues/7544.

StephanStS, 2024-12-07:
@MichaIng: I only own an Radxa Zero w/o eMMC. Would this help?
(Should I get one with eMMC? Would at least take until end of december 2024)

way5, 2024-12-07:
@StephanStS, I just successfully installed DietPi on SD and it boots without issues. The problem is only with eMMC.

CodeChenL, 2024-12-09:
https://github.com/armbian/build/pull/7538 This change has been accepted by Armbian. In short, some patchs are to repair the issue of EMMC particles compatibility from different manufacturers.

MichaIng, 2024-12-12:
I am rebuilding our kernel and bootloader packages and push them to the APT server ASAP.

closed 2024-12-25 by MichaIng