MichaIng/DietPi #7297: Robosats

brunocorisco86, 2024-11-30:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- ...

#### Describe the solution you'd like:

- ... I want a p2p bitcoin trading on. Based on docker and Nginx

#### Describe alternatives you've considered:

- ...

#### Additional context

- ...

Joulinar, 2024-11-30:
Usually we don't add Docker containers to our software installation script

MichaIng, 2024-12-02:
Seems to be possible to install via pip: https://github.com/RoboSats/robosats
But all in alpha state, nothing stable, and there are zero install instructions. I am not even sure what exactly this Python package installs. Based on the the quick start docs, you need a dedicated wallet, and a Tor browser, nothing else: https://learn.robosats.com/docs/quick-start/

... ah, so the liked repo contains an alternative "client", so no Tor browser needed then, but you can compile the client as Android app or desktop client, also for Linux. And there are pre-compiled binaries for x86 at least, which makes using the Docker image pretty ridiculous, IMO. Unless, the Python version is a problem.

MichaIng, 2024-12-02:
Interesting, the pre-compiled archive is fully self-contained, it seems, no Python scripts at all. It seems to be based on Chromium: 
[image]

Also the sources contain a lot of blobs: https://github.com/RoboSats/robosats/tree/main/desktopApp/tor/tor-linux/tor

brunocorisco86, 2024-12-03:
I'd love to have an instance running on a Raspberry Pi at home or wherever it's hosted, accessing remotely via Tailscale or SSH. However, I couldn't configure nginx after running docker-compose in the nodeapp folder. It keeps giving me a 403 nginx error. But apparently, Tor and Robosats were running.