MichaIng/DietPi #7298: zoraxy

xabirequejo, 2024-11-30:
### Creating a software request

[image]

#### Formal software information
- Software title | Zoraxy
- Short description | A general purpose HTTP reverse proxy and forwarding tool.
- Official URL | https://zoraxy.aroz.org/
- GitHub URL | https://github.com/tobychui/zoraxy
- (Official) install documentation | https://geekscircuit.com/installing-zoraxy-reverse-proxy-your-gateway-to-efficient-web-routing/

#### Are there similar/alternative software titles available with DietPi-Software?
- [frp](https://dietpi.com/docs/software/advanced_networking/#frp)

#### What makes your requested software better than the above solutions, if available?

- UI is easier on the eye.
- Much easier to use for newbies.
- There's no need for separated admin and dashboard panels.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- Installing Zoraxy is easy — integrating the service with already existing ones is what makes it tricky for noobs like me.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. `mkdir zoraxy`
2. `cd zoraxy`
3. `wget https://github.com/tobychui/zoraxy/releases/download/XYZ/zoraxy_linux_a**XY`
4. `mv zoraxy_linux_a**XY zoraxy`
5. `chmod +x zoraxy`
6. automagically configure lighttpd so that Zoraxy is available.

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Unfortunately I lack the skills needed to maintain software.

fow0ryl, 2025-01-27:
Hello,
there are some pitfalls with zoraxy. Especially when you start zoraxy from CLI.
Zoraxy will create some files and folders, just where you are. (cwd)
So it is very useful to use a systemd service for startup. And of course yue d dedicated directory.
Here is my zoraxy.service I'm using within LXC on Proxmox. (all on bookworm)
I have put the zoraxy binary @ /usr/bin/. The rest resides @ /opt/zoraxy and Output is piped to logger, so that we can look at this with journald.

If you are using a webserver on port 80 already zoraxy did not start (Should be solved with zoraxy 3.1.17).
You have to decide if you want your running webserver to serve port 80 or if you want zoraxy with port 80 redirect disabled ...

<pre><code>[Unit]
Description=Zoraxy Reverse Proxy Server

# start not befor network is online
After=network-online.target
Wants=network-online.target

[Service]
Type=simple

# start zoraxy as root
User=root
Group=root

# folder where zoraxy config & runtime data are located
WorkingDirectory=/opt/zoraxy

# use absolute path for zoraxy always
# options:
# -fastgeoip=true               faster GeoIP-lookup, but more RAM ...
#                               faster GeoIP-lookup, but uses more RAM ...
# -log /var/log/zoraxy          log to file / dont create local log folder
# -port=:8008                   use port instead of defalut (8000)

ExecStart=/usr/bin/zoraxy -log /var/log/zoraxy -port=:8000 2>&1 | logger &
ExecStop=/usr/bin/kill "$MAINPID"

[Install]
WantedBy=multi-user.target
</code></pre>

fow0ryl, 2025-02-11:
I decided to bypass some problems.

1. Do not put zoraxy in any folder that is in your exec path -> move binary to /opt/zoraxy/bin/
2. Do not run zoraxy as root -> create conf file & create them with systemd-users
3. Use "systemctl" only

/lib/sysusers.d/zoraxy-sysuser.conf
<pre><code>
#Type Name  ID          GECOS                       Home directory
g	zoraxy	-
u	zoraxy	-:zoraxy 	"User for zoraxy proxy"		/var/lib/zoraxy	-
</code></pre>

/lib/systemd/system/zoraxy.service
<pre><code>
[Unit]
Description=Zoraxy Reverse Proxy Server

# start not before network is online
After=network-online.target
Wants=network-online.target

[Service]
Type=simple

# start zoraxy as root
User=zoraxy
Group=zoraxy

# folder where zoraxy config & runtime data are located
WorkingDirectory=/opt/zoraxy

# Create log directory & allow access by zoraxy
ExecStartPre=!/bin/mkdir -p /var/log/zoraxy
ExecStartPre=!/bin/chmod -R 775 /var/log/zoraxy
ExecStartPre=!/bin/chown -R zoraxy:zoraxy /var/log/zoraxy/
ExecStartPre=!/bin/chown -R zoraxy:zoraxy /opt/zoraxy/

# use absolute path for zoraxy always
# options:
# -fastgeoip=true                               faster GeoIP-lookup, but more RAM ...
#                                                               faster GeoIP-lookup, but uses more RAM ...
# -log /var/log/zoraxy                  log to file / dont create local log folder
# -port=:8008                                   use port instead of defalut (8000)

ExecStart=/opt/zoraxy/bin/zoraxy -log /var/log/zoraxy -port=:8000 2>&1 | logger &
ExecStop=/usr/bin/kill "$MAINPID"

[Install]
WantedBy=multi-user.target
</code></pre>