MichaIng/DietPi #7299: Manual install over Armbian

way5, 2024-11-30:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- Date: 2024-11-30 15:20:48
- Program name: Dietpi-Installer
- Command: apt-get -y dist-upgrade"
- Exit code: 100
- DietPi version: v9.8.0 (MichaIne/master)
- Distro version: bookworm (ID=7, RASFRIAN=0)
- Kernel version: Linux RZERO 6.6.63-current-meson64 #1 MP PREEMPT Fri Nov 22 14:38:37 UTC 2024 aarch64 GNU/Linux
- Architecture: arm64
- Hardware model: (ID=74)
- Power supply: 5V 3A
- SD card: (EG: SanDisk Ultra 16 GB)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

The manual install over Armbian fails since installer script removes WiFi support and then tries to `apt install ...`. How to deal with this issue?

[image]

#### Extra details

- ...

MichaIng, 2024-12-01:
I guess it is due to NetworkManager removal, which can leave back `/etc/resolv.conf` as revised symlink. From the error handler menu, select "Open shell", and run these commands:
```sh
rm -f /etc/resolv.conf
echo 'nameserver 9.9.9.9' > /etc/resolv.conf
exit
```

I just added the workaround to the installer: https://github.com/MichaIng/DietPi/commit/08b20f4

way5, 2024-12-01:
Hi @MichaIng !
I did:

```
git clone -b dev ...
```

then as usual:

```
…#: .build/images/dietpi-install
```

Same issue:

[image]

Thank you.

MichaIng, 2024-12-05:
@way5 
Ah, the workaround was added to the wrong place. I just changed it, now it should work.

closed 2024-12-25 by MichaIng