MichaIng/DietPi #7300: EEPROM still held back...

Dynamic5912, 2024-11-30:
Is this still an issue?

I thought the "hold" was removed in a previous dietpi update?

I still get this when upgrading:

_dietpi@dietpi:~$ sudo apt upgrade -y                      Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  rpi-eeprom
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
dietpi@dietpi:~$ 
_

Joulinar, 2024-12-01:
Simply do

```
apt install --reinstall rpi-eeprom
```

MichaIng, 2024-12-01:
The package is not on hold and there was no change regarding. APT holds it back by default because it requires a replacement of old with new RPi tools package, and APT never removed a package without confirmation.

A reinstall should allow you to confirm, but it might break some legacy hardware features, as the new tools are designed for the new kernel/firmware packages. Same with the new EEPROM bootloader. Hence I'd either ignore this (keep `rpi-eeprom` on old version) or migrate the RPi to new kernel/firmware as well: #6676

Dynamic5912, 2024-12-01:
> The package is not on hold and there was no change regarding. APT holds it back by default because it requires a replacement of old with new RPi tools package, and APT never removed a package [...]

Thanks. I'm running a Pi4 not a Pi5 - so does this make any difference?

MichaIng, 2024-12-01:
No, the migration is needed to all RPi models.

Dynamic5912, 2024-12-02:
> No, the migration is needed to all RPi models.

OK thanks.

So what do I run to do the migration?

Any errors/warnings to look out for during the process?

MichaIng, 2024-12-02:
> So what do I run to do the migration?

The command in the first comment of that issue. No errors expected. When being asked, as you do not plan to run it on RPi 5, no additional kernel needs to be installed.

MichaIng, 2024-12-03:
As of the context change `sudo` does, you cannot pass through a script like that. Either you need to switch to an interactive `root` shell, or do like that:
```sh
sudo bash -c "$(curl -sSf 'https://raw.githubusercontent.com/MichaIng/DietPi/dev/.meta/dietpi-rpi-firmware-migration')"
```

Dynamic5912, 2024-12-03:
Thanks. All done - rebooted and everything seems fine.

It asked me whether I was installing on a Pi5 so I chose "NO" as I'm using a Pi4 and not planning (yet) on jumping to a Pi5.

If however I do plan to move to a Pi5 - is there a process to follow - I.e should I install a new DietPi image and start from scratch again?

I'm only really using my Pi4 for AdGuard Home, Unbound and some HomeAssistant tinkering.

Dynamic5912, 2024-12-03:
So in theory I can just take the SD Card out of my Pi4 and put it into a Pi5 (if/when purchased) and it will work with little to no additional configuration?

Dynamic5912, 2024-12-03:
Great thanks.

Glad I don't have to reinstall or restore everything if I were to upgrade :)

closed 2025-07-18 by MichaIng