MichaIng/DietPi #7301: Support RKNPU on OPI3B

yoni13, 2024-12-03:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
No

#### Describe the solution you'd like:
Add NPU support for 3B

#### Describe alternatives you've considered:
Use the official image from OrangePi or the image from Armbian

Also, is that possible? It looks like we are on the mainline instead of the Rockchip one.

MichaIng, 2024-12-06:
> Use the official image from OrangePi or the image from Armbian

We do own kernel builds with a soft-fork of the Armbian build system. In case of Orange Pi 3B, it is the edge branch, hence mainline kernel, not Orange Pi vendor kernel.

I guess the NPU requires particular drivers, which are not implemented into mainline Linux (yet)? We could provide vendor kernel builds, but I would prefer to see a way to have the needed drivers in mainline Linux. Maybe you have some more information/background, as you actually aim to make use of the NPU?

yoni13, 2024-12-06:
> We do own kernel builds with a soft-fork of the Armbian build system. In case of Orange Pi 3B, it is the edge branch, hence mainline kernel, not Orange Pi vendor kernel.
>
> I guess the NPU requires [...]

Yes, I am currently working on getting Immich's ML running on the RKNPU. At the moment, I am still figuring out how to resolve issues with get_inputs and get_outputs.

For testing, my workflow involves swapping  SD card with Armbian installed. While this setup is a bit inconvenient, it’s manageable for now.

MichaIng, 2024-12-06:
Ah, with Armbian it works? With the Linux v6.1 image, not the v6.6 image, right?

yoni13, 2024-12-06:
6.1.75-vendor-rk35xx
Yes.

MichaIng, 2024-12-06:
Please test this:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/linux-{image,dtb}-vendor-rk35xx.deb
dpkg -i linux-{image,dtb}-vendor-rk35xx.deb
reboot
```

MichaIng, 2024-12-06:
Sure it boots 😄. You you mean the NPU works with it?

yoni13, 2024-12-06:
> Sure it boots 😄. Or you mean the NPU works with it?

Yep 😅

MichaIng, 2024-12-06:
Okay great. I'll add the vendor kernel to the APT repo component for the Orange Pi 3B, so it can then be easier installed and kept updated.

MichaIng, 2024-12-06:
Before we add patches, can you test the latest build of this kernel?
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-vendor-rk35xx.deb
dpkg -i linux-{image,dtb}-vendor-rk35xx.deb
reboot
```

yoni13, 2024-12-07:
> Before we add patches, can you test the latest build of this kernel?
> ```sh
> cd /tmp
> wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-vendor-rk35xx.deb
> dpkg -i linux-{image,dtb}-vendor-rk35xx.deb
> reboot
> ```

The NPU works, what would you also want to test?

yoni13, 2024-12-07:
The RKMPP also works btw.

MichaIng, 2024-12-12:
> The NPU works, what would you also want to test?

Is the ulimit correct with that one?

yoni13, 2024-12-12:
> > The NPU works, what would you also want to test?
> 
> Is the ulimit correct with that one?

No

```
ulimit -u
486
```

MichaIng, 2024-12-12:
Okay, I'll create a build with these two patches:
- https://github.com/Joshua-Riek/linux-rockchip/commit/9b0de6b
- https://github.com/Joshua-Riek/linux-rockchip/commit/c91ccee

MichaIng, 2025-01-04:
Same issue in Orange Pi 5 Plus (same kernel): #7237

Okay, process limit of 486 is not acceptable. I'll do the builds with this patched today, missed it earlier, as it was shortly before I started my vacation trop.

MichaIng, 2025-01-04:
I pushed a new build of the vendor kernel to our APT repo, which should fix the low limit issue. Please test as well on Orange Pi 3B, just to be sure. I did not apply the patch, as a rebuild based on latest upstream sources solved it already for Orange Pi 5 Plus.

yoni13, 2025-01-05:
> I pushed a new build of the vendor kernel to our APT repo, which should fix the low limit issue. Please test as well on Orange Pi 3B, just to be sure. I did not apply the patch, as a rebuild based on latest upstream sources solved it already for Orange Pi 5 Plus.

```
root@orangepi3b:~# cat /sys/kernel/debug/rknpu/load
NPU load:  0%
root@orangepi3b:~# ulimit -u
30568
root@orangepi3b:~#
```
It works!

closed 2025-01-05 by yoni13

yoni13, 2025-07-05:
Hi, just want to confirm that did we switched back to mainline kernel?

MichaIng, 2025-07-05:
We never switched away from mainline Linux. The other way round: I aim to switch also RK3588 boards away from vendor to mainline Linux by default. But you can always install the vendor kernel manually, if needed for NPU.

yoni13, 2025-07-06:
> We never switched away from mainline Linux. The other way round: I aim to switch also RK3588 boards away from vendor to mainline Linux by default. But you can always install the vendor kernel manually, if needed for NPU.

Thanks for clarifying. How do I switch to vendor kernel manually?

MichaIng, 2026-03-01:
Fantastic! I am happy to review and test it as in-tree driver patched into mainline Linux. Makes sense that you clarified it was tested on DietPi with Armbian kernel, but I really see no reason it wouldn't work with mainline Linux in general. That kernel has a buch of patches, but unlikely this affects an entirely new driver for a device type that was not present (in code) before at all.