MichaIng/DietPi #7302: dietpi-drive_manager will start every service after use (even disabled one)

intiplink, 2024-12-09:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm 1

- Kernel version | `uname -a`
Linux DietPi 6.1.21+ 1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi Zero W

- Power supply used | (EG: 5V 1A RAVpower)
5v 1a

- SD card used | (EG: SanDisk ultra)
Lexar

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce
1. run dietpi-drive_manager
2. do anything so itll stop all services
3. exit, dietpi will re-start services

#### Expected behaviour
- only "restart" enabled services that stopped before run

#### Actual behaviour
- will "start" every services

is this intended beaviour?

Joulinar, 2024-12-09:
It's restarting all services that are listed as included within `dietpi-services`

intiplink, 2024-12-09:
so its intended behavour then, can i ask why? 
why its behaviour didnt like other operation? like when updating dietpi, it'll only restart previously stopped services.
also this line https://github.com/MichaIng/DietPi/blob/5f69eebea01ce277d48333ffa21b844d5a068c3b/dietpi/dietpi-drive_manager#L29 make me think only service that "stopped" during operation, not everything.

Joulinar, 2024-12-09:
Usually behaviour should be same on both operations

MichaIng, 2024-12-09:
`dietpi-services` does not `restart` disabled services, but still `start`s disabled services. I agree that this is not the expected behaviour when it is called automatically in other scripts like the drive manager, backup or any such. `dietpi-services` is not really meant (anymore) to be used manually to start/stop/enabled/disable services anyway. One should use `systemctl` directly for this, or use the integrated whipetail menu, if console commands are uncomfortable. The start/stop/restart CLI is hence mainly to be used in our other scripts, and there is makes sense to not ever start disabled services, unless within `dietpi-services` after they were explicitly installed. I hence aligned it so that neither `dietpi-serivces restart` nor `dietpi-services start` will touch disabled services: https://github.com/MichaIng/DietPi/commit/3ffb0b2

closed 2024-12-25 by MichaIng