MichaIng/DietPi #7305: Vaultwarden for armv6 not updated (fail to upload binary?)

intiplink, 2024-12-11:
Looking at build action, it fails to upload the binary, only x86-64  that successfully uploaded.

I don't know if it's just me, since I don't have other systems.

MichaIng, 2024-12-11:
I don't know where you see any upload failures. All green, all succeeded and arrived on the server. Note that we do not import them automatically into the APT repository. We always have a manual look onto it before doing so, the same binaries and images are manually moved out of `testing` sub directories.

There is generally no need to open a new issue on every vaultwarden update. Their releases are subscribed, and builds are triggered usually within hours. Email notifications are sent our on build errors automatically. Give us a day or two, until they are available via APT.

closed 2024-12-11 by MichaIng

intiplink, 2024-12-11:
thankyou for answers, and im sorry i probably should give more context
on armv6 side, it stuck at 1.30.5 (which is released on march), 2 major version late. 

i intend to change my docker instalation to binary that provided by dietpi, but found out that dietpi binary is way too old, it wont work with current bitwarden client.
the latest binary is found on testing folder (as seen on build action), but it seems as you said, it manually moved out to APT, hence my question, is it failed to upload to APT?

[image]

MichaIng, 2024-12-11:
Ah thanks for that info. Indeed our import script did not automatically import ARMv6 packages correctly. I fixed it just now and refreshed the package lists. You should get latest vaultwarden via APT now.