MichaIng/DietPi #7306: LMS has been rebranded Lyrion Music Server (no more Logitech)

mikeysas, 2024-12-11:
With the release of version 9.0, the LMS Community has completed the transition to Lyrion.
https://lyrion.org/

I don't fully understand how DIetPi manages the install, but in addition to updating the name and documentation to refer to Lyrion of Logitech, please be aware that the service name has also changed to lyrionmusicserver for Debian.  The location for preferences and logging as party of the standard Debian install have not changed and the user is still squeezeboxserver.

MichaIng, 2024-12-11:
Right, I recognised it already here: #6992

But with the release, makes sense to rebrand it our end as well 👍.

MichaIng, 2024-12-11:
I hope/guess the service still has an alias `logitechmediaserver`? Would be an unnecessary breaking change to not leave that in side for a while.

MichaIng, 2024-12-11:
Mostly done:
- https://github.com/MichaIng/DietPi/commit/1aff231
- https://github.com/MichaIng/DietPi/commit/dfdbbd2

But the installation tests on Trixie ARM systems currently fail: https://github.com/MichaIng/DietPi/actions/runs/12284464087
I need to check on a physical system whether it is only because of the emulated containers, which run the tests here on GitHub.

Another test is whether the `lyrionmusicserver` package seamlessly replaces the `logitechmediaserver` package with a `Conflicts` or `Breaks` definition, or whether it needs to be uninstalled manually.

mikeysas, 2024-12-12:
I do not believe that logitechmediaserver is left as an alias. The standard Debian install using apt install removes logitechmediaserver and installs lyrionmusicserver.  Some folks who used dpkg had trouble as it did not completely remove logitechmediaserver.  There is a lot of info on this in the Lyrion forum.

MichaIng, 2024-12-12:
And migration works fine + packages are running well on actual ARMv8 hardware.

closed 2024-12-25 by MichaIng