MichaIng/DietPi #7311: dietpi_installer autopurge error

mews-se, 2024-12-17:
### Creating a bug report/issue
- [ x] I have searched the existing open and closed issues

Installing via the script from a ethernetservers.com VM.

```
E: Essential packages were removed and -y was used without --allow-remove-essential.
[FAILED] DietPi-Installer | APT autopurge gawk original-awk
 - Command: apt-get -y autopurge gawk original-awk
```

--allow-remove-essential needs to be added in some instances of running dietpi_installer on barebone Debian

mews-se, 2024-12-17:
More error-log

```
[ INFO ] DietPi-Installer | Generating list of minimal packages, required for DietPi installation
environment: line 1489: mawk: command not found
removed '/etc/apt/apt.conf.d/01autoremove-kernels'
[ INFO ] DietPi-Installer | APT autopurge gawk original-awk, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
Package 'original-awk' is not installed, so not removed
The following packages will be REMOVED:
  acl* apt-utils* base-files* bash* binutils* binutils-common*
  binutils-x86-64-linux-gnu* bsdextrautils* bzip2* ca-certificates*
  cifs-utils* console-common* console-data* cracklib-runtime* cron* curl*
  db-util* db5.3-util* dbus* debconf-i18n* debconf-utils* dialog* dirmngr*
  distro-info-data* dselect* ed* expat* fetchmail* file* finger* fontconfig*
  fontconfig-config* fonts-dejavu* fonts-dejavu-core* fonts-dejavu-extra*
  fonts-freefont-ttf* ftp* gawk* gettext* gnupg* gnupg-l10n* gnupg-utils* gpg*
  gpg-agent* gpg-wks-client* gpg-wks-server* gpgconf* gpgsm* groff-base*
  ifupdown* info* initramfs-tools-core* initscripts* insserv* iproute2*
  iputils-arping* iputils-ping* iputils-tracepath* isc-dhcp-client*
  isc-dhcp-common* kbd* klibc-utils* ldap-utils* less* libapparmor1*
  libargon2-1* libasprintf0v5* libassuan0* libbinutils* libbpf0* libbsd0*
  libc-l10n* libcap2* libcap2-bin* libcbor0* libcrack2* libcryptsetup12*
  libctf-nobfd0* libctf0* libcurl4* libdbus-1-3* libdns-export1110* libedit2*
  libelf1* libestr0* libexpat1* libext2fs2* libfastjson4* libfdisk1*
  libfido2-1* libfl2* libfontconfig1* libgdbm-compat4* libgdbm6* libgomp1*
  libgpm2* libicu67* libidn11* libip4tc2* libisc-export1105* libjson-c5*
  libklibc* libksba8* libldap-2.4-2* liblognorm5* libltdl7* libmagic-mgc*
  libmagic1* libmd0* libmnl0* libmpdec3* libmpfr6* libncurses5* libncurses6*
  libncursesw6* libnewt0.52* libnghttp2-14* libnl-3-200* libnl-genl-3-200*
  libnpth0* libodbc1* libpaper1* libpcap0.8* libpci3* libperl5.32*
  libpipeline1* libpopt0* libprocps8* libpsl5* libpython2-stdlib*
  libpython2.7-minimal* libpython2.7-stdlib* libpython3-stdlib*
  libpython3.9-minimal* libpython3.9-stdlib* libreadline8* librtmp1*
  libsasl2-2* libsasl2-modules-db* libsensors-config* libsensors5*
  libsigsegv2* libslang2* libsnmp-base* libsnmp40* libsqlite3-0* libssh2-1*
  libtalloc2* libtevent0* libtinfo5* libuchardet0* libustr-1.0-1*
  libutempter0* libwbclient0* libwrap0* libxml2* libxtables12* locales*
  logrotate* logsave* lsb-release* lsof* lynx* lynx-common* lzma* m4* mailcap*
  make* makedev* man-db* manpages* media-types* memtester* mime-support*
  mtools* nano* ncurses-term* net-tools* netbase* odbcinst* odbcinst1debian2*
  openssh-client* openssh-server* openssh-sftp-server* openssl* patch*
  pci.ids* perl* perl-modules-5.32* pinentry-curses* procinfo* procmail*
  procps* psmisc* psutils* pwgen* python-is-python2* python2* python2-minimal*
  python2.7* python2.7-minimal* python3* python3-minimal* python3.9*
  python3.9-minimal* quota* readline-common* resolvconf* rpcbind* rsync*
  rsyslog* runit-helper* sasl2-bin* screen* sharutils* snmp* ssl-cert*
  startpar* sudo* syslinux* systemd* systemd-sysv* sysv-rc* tcpdump* tcsh*
  telnet* tofrodos* traceroute* udev* unixodbc* unzip* uuid-runtime* vim*
  vim-common* vim-runtime* wget* whiptail* whois* wide-dhcpv6-client* xinetd*
  xxd* xz-utils*
WARNING: The following essential packages will be removed.
This should NOT be done unless you know exactly what you are doing!
  base-files gawk (due to base-files) bash
0 upgraded, 0 newly installed, 242 to remove and 96 not upgraded.
E: Essential packages were removed and -y was used without --allow-remove-essential.
[FAILED] DietPi-Installer | APT autopurge gawk original-awk
 - Command: apt-get -y autopurge gawk original-awk
./dietpi-globals: line 966: mawk: command not found
./dietpi-globals: line 1329: mawk: command not found
./dietpi-globals: line 1330: mawk: command not found
./dietpi-globals: line 1329: mawk: command not found
./dietpi-globals: line 1330: mawk: command not found
```

mews-se, 2024-12-17:
Ah crap... Just now noticed that it's a OpenVZ VM. Why is this still a thing :D

Well, sorry for interrupting!

MichaIng, 2024-12-17:
Weird, there seems to be no `mawk` installed on your system, while IIRC it is a dependency of an essential package. Obviously any `awk` providing package is now dependency of an essential package.

So we need to assure `mawk` is installed before attempting to purge the other two.

MichaIng, 2024-12-23:
Added as installer dependency here: https://github.com/MichaIng/DietPi/commit/5542179

closed 2024-12-25 by MichaIng