MichaIng/DietPi #7320: Dietpi-backup errors when rsync reaches Perl folder

Tooloco, 2024-12-23:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=8
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | `SPBIAN bookworm`
- Kernel version | `Linux DietPiTrueNas 6.1.0-28-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.119-1 (2024-11-22) x86_64 GNU/Linux`
- SBC model | `Native PC (x86_64)`
- Power supply used | Corsair 600w PC PSU
- SD card used | NVME SSD

#### Additional Information (if applicable)
- Software title | Dietpi-backup
- Was the software title installed freshly or updated/migrated? 
The OS has been installed freshly with almost no configuration changes
- Can this issue be replicated on a fresh installation of DietPi?
- The issue happens on the Native PC and a RPi4 too
 
- Bug report ID | `f32e0b11-9f42-46ac-add0-7287a14b21ca`

#### Steps to reproduce

1. Run dietpi-backup and choose backup
2. Backup location is a NFS share (`/mnt/truenas/hdds/backup/duetpi-truenas`) on a Truenas host (copied /etc/exports below)
3. Rsync throws multiple errors when reaching perl directory E.g. : `rsync: [receiver] mkstemp /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/.Termcap.pm.KMpbwz failed: No such file or directory (2)`
4. Backup finishes unsuccesfully with the message: `[FAILED] DietPi-Backup: Backup | Exited with error`

#### Expected behaviour

- Backup finishes succesfully

#### Actual behaviour

- Rsync throws multiple errors when reaching perl directory: `rsync: [receiver] mkstemp /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/.Termcap.pm.KMpbwz failed: No such file or directory (2)`
- Backup marked as failed: `Backup failed:   - /mnt/truenas/hdds/backup/dietpi-truenas You will see the log file on the next screen. Please check it for information and/or errors.`
- `sent 1,867,594,424 bytes  received 502,767 bytes  1,686,769.47 bytes/sec
total size is 1,869,495,229  speedup is 1.00
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]`

#### Extra details

- Backup location: `/mnt/truenas/hdds/backup/dietpi-truenas`
- Truenas: cat /etc/exports `"/mnt/HDDs/shares"\
        192.168.1.0/24(sec=sys,rw,no_root_squash,insecure,no_subtree_check)`
- Dataset permissions:
[image]

- Example of rsync error messages: 
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.TextContent.pm.85goYg" failed: No such file or directory (2)`
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.TiedOutFH.pm.Kv0zgl" failed: No such file or directory (2)`
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.Transcode.pm.Kkjbjp" failed: No such file or directory (2)`
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.TranscodeDumb.pm.5KGjSX" failed: No such file or directory (2)`
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.TranscodeSmart.pm.e7gGul" failed: No such file or directory (2)`
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.XHTML.pm.cQBCmK" failed: No such file or directory (2)`
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.XMLOutStream.pm.k771zQ" failed: No such file or directory (2)`
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/.Color.pm.kO7Mzd" failed: No such file or directory (2)`
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/.Overstrike.pm.EbgzX2" failed: No such file or directory (2)`
`rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/.Termcap.pm.KMpbwz" failed: No such file or directory (2)`

Joulinar, 2024-12-23:
is there a perl application running that might create temp files during the backup?

Tooloco, 2024-12-23:
Not that I'm aware of, not any that I've programmed of that im sure hehe:
These are the services present in the machine and i've only used dietpi-software to install them:
[image]
How would I check for a perl app?

Joulinar, 2024-12-23:
Wireguard has nothing to do with Perl. Ho do you start/use your Perl app?

Can you have a look to the folder and check if content is changing over time

```
ls -la /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/
```

Just repeat this again a couple of times.

Tooloco, 2024-12-23:
`sudo ls -la /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/`
`ls: cannot access '/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/': No such file or directory`

`sudo ls -la /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/`
`total 423`
`drwxr-xr-x  2 root root     14 Nov 22 16:37 .`
`drwxr-xr-x 52 root root    127 Nov 22 16:37 ..`
`-rw-r--r--  1 root root  33843 Nov 25  2023 Checker.pm`
`-rw-r--r--  1 root root  17888 Nov 25  2023 Escapes.pm`
`-rw-r--r--  1 root root  14347 Nov 25  2023 Functions.pm`
`-rw-r--r--  1 root root  22196 Nov 25  2023 Html.pm`
`-rw-r--r--  1 root root  79750 Nov 25  2023 Man.pm`
`-rw-r--r--  1 root root   6399 Nov 25  2023 ParseLink.pm`
`-rw-r--r--  1 root root 300178 Nov 25  2023 perldiag.pod`
`-rw-r--r--  1 root root  64594 Nov 25  2023 Perldoc.pm`
`-rw-r--r--  1 root root  54797 Nov 25  2023 Simple.pm`
`-rw-r--r--  1 root root  15146 Nov 25  2023 Simple.pod`
`-rw-r--r--  1 root root  36448 Nov 25  2023 Text.pm`
`-rw-r--r--  1 root root  30993 Nov 25  2023 Usage.pm`

Joulinar, 2024-12-23:
does the folder exist on original location?

`ls -la /usr/share/perl/5.36.0/Pod/Text/`

Tooloco, 2024-12-23:
`dietpi@DietPiTrueNas:~$ ls -la /usr/share/perl/5.36.0/Pod/Text/`
`total 36`
`drwxr-xr-x 2 root root 4096 Nov 22 16:37 .`
`drwxr-xr-x 6 root root 4096 Nov 22 16:37 ..`
`-rw-r--r-- 1 root root 6046 Nov 25  2023 Color.pm`
`-rw-r--r-- 1 root root 6259 Nov 25  2023 Overstrike.pm`
`-rw-r--r-- 1 root root 8350 Nov 25  2023 Termcap.pm`

Joulinar, 2024-12-23:
ok rsync can't create the required directory structure for `5.36.0/Pod/Text/` and `perl/5.36.0/Pod/Simple/` 🤔 

Can you create them manually?

```
sudo mkdir /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/
sudo mkdir /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/
```

Stupid question, there is enough disk space available and no limitation on target location?

```
df -h
```

Tooloco, 2024-12-23:
`dietpi@DietPiTrueNas:~$ sudo mkdir /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/`
`dietpi@DietPiTrueNas:~$ sudo mkdir /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Text/`
`dietpi@DietPiTrueNas:~$ df -h`
`Filesystem                     Size  Used Avail Use% Mounted on`
`udev                           1.9G     0  1.9G   0% /dev`
`tmpfs                          391M  5.5M  386M   2% /run`
`/dev/vda2                       32G  1.2G   29G   4% /`
`tmpfs                          2.0G     0  2.0G   0% /dev/shm`
`tmpfs                          5.0M     0  5.0M   0% /run/lock`
`tmpfs                          2.0G     0  2.0G   0% /tmp`
`tmpfs                           50M     0   50M   0% /var/log`
`/dev/vda1                       63M   12M   52M  19% /boot/efi`
`192.168.1.15:/mnt/HDDs/shares   22T  1.3T   21T   6% /mnt/truenas/hdds`
It seems that I could create the folders without problems, and the share has plenty of space left :)
Should I try to run backup now?

Joulinar, 2024-12-23:
> Should I try to run backup now?

Yes give it a try

Tooloco, 2024-12-23:
No luck :(
```
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.Debug.pm.kOGv9l" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.DumpAsText.pm.GOtXxD" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.DumpAsXML.pm.DmI9dB" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.HTML.pm.Lxo2yD" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.HTMLBatch.pm.3IvlQj" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.HTMLLegacy.pm.6PfuVC" failed: No such file or directory (2)

rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.JustPod.pm.4dcRb8" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.LinkSection.pm.YWFRLv" failed: No such file or directory (2)
deleting usr/share/perl5/Git/LoadCPAN/Mail/Address.pm
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.Methody.pm.UK8DXE" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.Progress.pm.6Dwm2K" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.PullParser.pm.UkHln9" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.PullParserEndToken.pm.Qw1A0v" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.PullParserStartToken.pm.2eFplD" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.PullParserTextToken.pm.kOuW8R" failed: No such file or directory (2)
rsync: [receiver] mkstemp "/mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/Simple/.PullParserToken.pm.h50CSc" failed: No such file or directory (2)

rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
[  OK  ] DietPi-Backup | cp -a /etc/debian_version /mnt/truenas/hdds/backup/dietpi-truenas/data/etc/debian_version
[  OK  ] DietPi-Backup | touch /mnt/truenas/hdds/backup/dietpi-truenas/data
[ SUB1 ] DietPi-Services > start
[  OK  ] DietPi-Services | start : cron
[FAILED] DietPi-Backup: Backup | Exited with error
```

Joulinar, 2024-12-23:
folder still exist on target side?

`ls -la /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/`

Tooloco, 2024-12-23:
Seems like it doesn't
```
dietpi@DietPiTrueNas:~$ ls -la /mnt/truenas/hdds/backup/dietpi-truenas/data/usr/share/perl/5.36.0/Pod/
total 188
drwxr-xr-x  2 root root      3 Nov 22 16:37 .
drwxr-xr-x 52 root root    127 Nov 22 16:37 ..
-rw-r--r--  1 root root 300178 Nov 25  2023 perldiag.pod
```

Tooloco, 2024-12-24:
Okay, so I just tried to backup to a different NFS share and it works without errors, I dont know what might be different between the two shares.
The one that doesnt work (192.168.1.15) is a Truenas host (ZFS, and all that) I'm not very experienced with Truenas.
And the one that does work (192.168.1.14) is a NFS share hosted on a RPI4 running Dietpi v9.8
```
dietpi@DietPiTrueNas:/mnt/nfs_client/Backup/dietpi-truenas$ sudo nfsstat -m
/mnt/truenas/hdds from 192.168.1.15:/mnt/HDDs/shares
 Flags: rw,relatime,vers=4.2,rsize=1048576,wsize=1048576,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=192.168.1.16,local_lock=none,addr=192.168.1.15

/mnt/nfs_client from 192.168.1.14:/mnt/hdd
 Flags: rw,relatime,vers=3,rsize=524288,wsize=524288,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=192.168.1.14,mountvers=3,mountport=36028,mountproto=udp,local_lock=none,addr=192.168.1.14

```

MichaIng, 2024-12-25:
Check this directory on the DietPi host:
```console
root@dietpi:~# l /usr/share/perl/5.*.0/Pod/Simple
total 440K
-rw-r--r-- 1 root root  85K Nov 23 16:42 BlackBox.pm
-rw-r--r-- 1 root root 5.3K Nov 23 16:42 Checker.pm
-rw-r--r-- 1 root root 4.5K Nov 23 16:42 Debug.pm
-rw-r--r-- 1 root root 4.0K Nov 23 16:42 DumpAsText.pm
-rw-r--r-- 1 root root 4.5K Nov 23 16:42 DumpAsXML.pm
-rw-r--r-- 1 root root  34K Nov 23 16:42 HTML.pm
-rw-r--r-- 1 root root  40K Nov 23 16:42 HTMLBatch.pm
-rw-r--r-- 1 root root 2.7K Nov 23 16:42 HTMLLegacy.pm
-rw-r--r-- 1 root root 9.5K Nov 23 16:42 JustPod.pm
-rw-r--r-- 1 root root 4.2K Nov 23 16:42 LinkSection.pm
-rw-r--r-- 1 root root 3.5K Nov 23 16:42 Methody.pm
-rw-r--r-- 1 root root 2.4K Nov 23 16:42 Progress.pm
-rw-r--r-- 1 root root  26K Nov 23 16:42 PullParser.pm
-rw-r--r-- 1 root root 2.9K Nov 23 16:42 PullParserEndToken.pm
-rw-r--r-- 1 root root 4.1K Nov 23 16:42 PullParserStartToken.pm
-rw-r--r-- 1 root root 3.3K Nov 23 16:42 PullParserTextToken.pm
-rw-r--r-- 1 root root 4.0K Nov 23 16:42 PullParserToken.pm
-rw-r--r-- 1 root root  23K Nov 23 16:42 RTF.pm
-rw-r--r-- 1 root root  35K Nov 23 16:42 Search.pm
-rw-r--r-- 1 root root 4.5K Nov 23 16:42 SimpleTree.pm
-rw-r--r-- 1 root root  34K Nov 23 16:42 Subclassing.pod
-rw-r--r-- 1 root root 5.0K Nov 23 16:42 Text.pm
-rw-r--r-- 1 root root 2.5K Nov 23 16:42 TextContent.pm
-rw-r--r-- 1 root root 2.7K Nov 23 16:42 TiedOutFH.pm
-rw-r--r-- 1 root root  710 Nov 23 16:42 Transcode.pm
-rw-r--r-- 1 root root 2.6K Nov 23 16:42 TranscodeDumb.pm
-rw-r--r-- 1 root root  693 Nov 23 16:42 TranscodeSmart.pm
-rw-r--r-- 1 root root  28K Nov 23 16:42 XHTML.pm
-rw-r--r-- 1 root root 4.6K Nov 23 16:42 XMLOutStream.pm
```
It should not contain any dot files. If it does, it seems to be remains of a filesystem corruption. If so, try to repair the filesystem on reboot:
```sh
> /forcefsck
reboot
```
After reboot, logs can be found like this:
```sh
 cat /run/initramfs/fsck.log
```
If there are still dot files in this dir, it is safe to remove them (if possible):
```sh
rm /usr/share/perl/5.*.0/Pod/Simple/.??*
```

Tooloco, 2024-12-25:
Sadly theres no dot files there
```
dietpi@DietPiTrueNas:/mnt/truenas/hdds2$ ls -la /usr/share/perl/5.*.0/Pod/Simple
total 448
drwxr-xr-x 2 root root  4096 Nov 22 16:37 .
drwxr-xr-x 6 root root  4096 Nov 22 16:37 ..
-rw-r--r-- 1 root root 86688 Nov 25  2023 BlackBox.pm
-rw-r--r-- 1 root root  5432 Nov 25  2023 Checker.pm
-rw-r--r-- 1 root root  4625 Nov 25  2023 Debug.pm
-rw-r--r-- 1 root root  4035 Nov 25  2023 DumpAsText.pm
-rw-r--r-- 1 root root  4560 Nov 25  2023 DumpAsXML.pm
-rw-r--r-- 1 root root 40656 Nov 25  2023 HTMLBatch.pm
-rw-r--r-- 1 root root  2758 Nov 25  2023 HTMLLegacy.pm
-rw-r--r-- 1 root root 34759 Nov 25  2023 HTML.pm
-rw-r--r-- 1 root root  9637 Nov 25  2023 JustPod.pm
-rw-r--r-- 1 root root  4301 Nov 25  2023 LinkSection.pm
-rw-r--r-- 1 root root  3571 Nov 25  2023 Methody.pm
-rw-r--r-- 1 root root  2413 Nov 25  2023 Progress.pm
-rw-r--r-- 1 root root  2884 Nov 25  2023 PullParserEndToken.pm
-rw-r--r-- 1 root root 25729 Nov 25  2023 PullParser.pm
-rw-r--r-- 1 root root  4146 Nov 25  2023 PullParserStartToken.pm
-rw-r--r-- 1 root root  3363 Nov 25  2023 PullParserTextToken.pm
-rw-r--r-- 1 root root  4004 Nov 25  2023 PullParserToken.pm
-rw-r--r-- 1 root root 23556 Nov 25  2023 RTF.pm
-rw-r--r-- 1 root root 35836 Nov 25  2023 Search.pm
-rw-r--r-- 1 root root  4632 Nov 25  2023 SimpleTree.pm
-rw-r--r-- 1 root root 33710 Nov 25  2023 Subclassing.pod
-rw-r--r-- 1 root root  2518 Nov 25  2023 TextContent.pm
-rw-r--r-- 1 root root  5103 Nov 25  2023 Text.pm
-rw-r--r-- 1 root root  2751 Nov 25  2023 TiedOutFH.pm
-rw-r--r-- 1 root root  2693 Nov 25  2023 TranscodeDumb.pm
-rw-r--r-- 1 root root   736 Nov 25  2023 Transcode.pm
-rw-r--r-- 1 root root   715 Nov 25  2023 TranscodeSmart.pm
-rw-r--r-- 1 root root 26592 Nov 25  2023 XHTML.pm
-rw-r--r-- 1 root root  4666 Nov 25  2023 XMLOutStream.pm
```

I just tried making a new NFS share on truenas and the backup worked correctly 
```
/mnt/truenas/hdds from 192.168.1.15:/mnt/HDDs/shares
 Flags: rw,relatime,vers=4.2,rsize=1048576,wsize=1048576,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=192.168.1.16,local_lock=none,addr=192.168.1.15

/mnt/nfs_client from 192.168.1.14:/mnt/hdd
 Flags: rw,relatime,vers=3,rsize=524288,wsize=524288,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,mountaddr=192.168.1.14,mountvers=3,mountport=36028,mountproto=udp,local_lock=none,addr=192.168.1.14

/mnt/truenas/hdds2 from 192.168.1.15:/mnt/HDDs/shares2
 Flags: rw,relatime,vers=4.2,rsize=1048576,wsize=1048576,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=192.168.1.16,local_lock=none,addr=192.168.1.15

```
```
admin@truenas[~]$ sudo cat /etc/exports
[sudo] password for admin:

"/mnt/HDDs/shares"\
        192.168.1.0/24(sec=sys,rw,no_root_squash,insecure,no_subtree_check)
"/mnt/HDDs/shares2"\
        192.168.1.16(sec=sys,rw,no_root_squash,insecure,no_subtree_check)

```
/mnt/HDDs/shares2 works without a hitch.
I think it has to do with how truenas handles multiprotocol shares (SMB+NFS) I just created this one as NFS with the above settings (no_root_squash).
[image]
With a POSIX ACL Type
[image]
And these dataset settings

Joulinar, 2024-12-25:
Are you ok if we close this issue as it seems to be related to your NAS?

Tooloco, 2024-12-25:
Yeah sure, hopefully it works as a workaround for more people :D
Merry christmas!

closed 2024-12-25 by Joulinar