MichaIng/DietPi #7321: Error on Sonarr v3 to v4

tellice, 2024-12-25:
i have updated to version 9.9.0 of DietPi and migration Sonarr from v3 to v4

When using the internal updater for Sonarr, i get the following error
Cannot install update because startup folder '/opt/sonarr' is not writable by the user 'sonarr'.

I have checked permissions on the folder and all child files and folders, and all have the correct permissions

any ideas?

tellice, 2024-12-25:
I have also uninstalled sonarr and reinstalled it again

MichaIng, 2024-12-25:
Thanks for reporting. The installer sets:
```sh
install_dir='/opt/sonarr'
...
chown -R sonarr:dietpi /mnt/dietpi_userdata/sonarr "$install_dir"
```
Do you see the last command with correct `/opt/sonarr` as last argument in your install/reinstall output?

I'll also try to replicate it.

MichaIng, 2024-12-25:
I found it. We missed to update the path within the systemd unit. Fixed with: https://github.com/MichaIng/DietPi/commit/c7a3411

To fix it for you now:
```sh
sudo sed -i 's|/usr/lib/sonarr|/opt/sonarr|' /etc/systemd/system/sonarr.service
sudo systemctl daemon-reload
sudo systemctl restart sonarr
```
I'll ship a live path to fix it within `dietpi-software`.

closed 2024-12-25 by MichaIng

tellice, 2024-12-25:
Thats great, i tried the fix but it didnt work. I had to apply the live patch and reintall Sonarr. But it is now all working and updates great.

MichaIng, 2024-12-25:
Thanks for testing it ASAP. No idea why the commands did not work for you, but more importantly the live patch did.