MichaIng/DietPi #7323: raspotify fails to start on fresh install

gergokis, 2024-12-25:
### Creating a bug report/issue
- [ x] I have searched the existing open and closed issues

#### Required Information
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'

- Distro version: bookworm 0
- Kernel version | Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | IanCanada PurePI II
- SD card used | Sandisk 64GB

#### Additional Information (if applicable)
- Software title | raspotify
- Was the software title installed freshly or updated/migrated? The whole system is a fresh new install.
- Can this issue be replicated on a fresh installation of DietPi? Yes
787bf0fa-f134-446a-a072-c9c4d4b6650c

#### Steps to reproduce
systemctl start raspotify

#### Expected behaviour
raspotify starts

#### Actual behaviour
raspotify does not start

#### Extra details

root@DietPi:/etc# systemctl status raspotify
× raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Wed 2024-12-25 14:55:44 GMT; 12ms ago
   Duration: 159ms
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 11379 ExecStart=/usr/bin/librespot (code=exited, status=1/FAILURE)
   Main PID: 11379 (code=exited, status=1/FAILURE)
        CPU: 158ms

Dec 25 14:55:44 DietPi systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
Dec 25 14:55:44 DietPi librespot[11379]: [2024-12-25T14:55:44Z ERROR librespot_discovery] Avahi error: Setting up dns-sd failed: I/O error: No such file or directory (os error 2)
Dec 25 14:55:44 DietPi librespot[11379]: [2024-12-25T14:55:44Z ERROR librespot] Discovery stopped unexpectedly
Dec 25 14:55:44 DietPi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Dec 25 14:55:44 DietPi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Dec 25 14:55:44 DietPi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
root@DietPi:/etc#

Joulinar, 2024-12-25:
> ```
> I/O error: No such file or directory (os error 2)
> ```

Looks like file system corruption. Can you check for kernel error messages 

```
dmesg -l 0,1,2,3
```

gergokis, 2024-12-25:
root@DietPi:~# dmesg -l 0,1,2,3
[    0.621936] bcm2708_fb soc:fb: Unable to determine number of FBs. Disabling driver.
root@DietPi:~#

hawaltie, 2024-12-29:
I have the same problem. dmesg -l 0,1,2,3 didn't show any problem.

Joulinar, 2024-12-29:
ok it's not an I/O error of file system. It's a connection issue to `avahi-daemon` Can you try to install `Avahi`??

```
dietpi-software install 152
```

hawaltie, 2024-12-29:
Works!

gergokis, 2024-12-29:
I can also confirm installing avahi fixes the issue. Thank you very much for the very effective support.

MichaIng, 2024-12-30:
Hmm, I guess the use of Avahi-Daemon can be disabled.