MichaIng/DietPi #7324: Problem installing modules in NodeRed

TadyTheFish, 2024-12-25:
Hello
I have updated Dietpi from 9.8.0 to 9.9.0 and started to have problems with Node red.
If I try to install ANY of the available nodes I get an BADENGINE error from npm
`2024-12-25T16:17:28.748Z [err] npm error code EBADENGINE
2024-12-25T16:17:28.748Z [err] npm error engine Unsupported engine
2024-12-25T16:17:28.748Z [err] npm error engine Not compatible with your version of node/npm: @homebridge/ciao@1.3.1
2024-12-25T16:17:28.748Z [err] npm error notsup Not compatible with your version of node/npm: @homebridge/ciao@1.3.1
2024-12-25T16:17:28.748Z [err] npm error notsup Required: {"node":"^18 || ^20 || ^22"}
2024-12-25T16:17:28.748Z [err] npm error notsup Actual:   {"npm":"10.9.2","node":"v23.5.0"}
2024-12-25T16:17:28.749Z [err] npm error A complete log of this run can be found in: /mnt/dietpi_userdata/node-red/.npm/_logs/2024-12-25T16_17_27_651Z-debug-0.log
2024-12-25T16:17:28.757Z rc=1`
The installed node.js is version 23 but no modules want that, even the official recommendation from Node red is to use version 22.
I installed NodeRed via dietpi-software and it pulls version 23
can you revert back to Node.js version 22?
If I try to install Node red via their official script it thrown se an error that I am running Dietpi and should install via Dietpi-software.
Thank you

TadyTheFish, 2024-12-25:
For anyone facing the same problems, you can manually downgrade Node.js
`wget https://raw.githubusercontent.com/MichaIng/nodejs-linux-installer/master/node-install.sh
chmod +x node-install.sh
sudo ./node-install.sh -v v22.12.0
`
and reboot the system
Worked for me and now I can install modules

MichaIng, 2024-12-30:
We should switch to LTS versions by default. I checked the index we currently parse, and there is an LTS field indeed. It is a little inconsistent, sometimes containing the code name of the release, sometimes just "true", and some subversions of an LTS release have the field empty. But otherwise I think we can use it, taking the first release which has this field set to any value (other than "-"):
- https://nodejs.org/dist/index.tab
- https://unofficial-builds.nodejs.org/download/release/index.tab

I need to add support for this (an `--lts` flag) to our Node.js installer, when I find time. Will be done until next DietPi release: https://github.com/MichaIng/nodejs-linux-installer

lallafa, 2026-10-22:
Currently installing/reinstalling node-red will also install an outdated node.js 23 with 10 CVEs including Brotli RCE (CVE-2026-111497).

MichaIng, 2026-10-22:
Which architecture do you use? I just checked it, and it should be v22.23.4 from 2 days ago on `armv6l`, v26.0.0 on `riscv64` (which is an issue indeed, later v26 builds fail their end, and there would be a v24 LTS from 2 days ago), and all other architectures should be v26.12.1 from 2 days ago.

Our Node.js installer always takes the Node.js build with the highest version number available from https://nodejs.org/dist/index.tab, and https://unofficial-builds.nodejs.org/download/release/index.tab for `armv6l` and `riscv64`.

lallafa, 2026-10-22:
> 19\. Which architecture do you use? I just checked it, and it should be v22.23.4 from 2 days ago on `armv6l`, v26.0.0 on `riscv64` (which is an issue indeed, later v26 builds fail their end, and there would be a v24 LTS from 2 days ago), and all other architectures should be v26.12.1 from 2 days ago.

DietPi v10.8.2 
 - Device model : Odroid C1 (armv7l)

MichaIng, 2026-10-22:
Ohh they dropped building for `armv7l` from v24 on. Similar to `riscv64`, the last LTS version would be much newer (build date): v22.23.4 from 3 days ago.

So yeah, more arguments to implement an `--lts` flag to our Node.js installer and use that consequently in `dietpi-software`.

lallafa, 2026-10-22:
I use this workaround right now:

```shell
sudo dietpi-software reinstall nodered
sudo systemctl stop node-red
curl -sSfL 'https://raw.githubusercontent.com/MichaIng/nodejs-linux-installer/master/node-install.sh' | sudo bash -s -- -v v22.23.4
sudo -u nodered bash -c 'cd /mnt/dietpi_userdata/node-red && npm rebuild'
```

MichaIng, 2026-10-22:
Right, best workaround.

closed 2026-10-23 by TadyTheFish

MichaIng, 2026-10-23:
Done with https://github.com/MichaIng/DietPi/commit/54345a0 for next release.

While reviewing stuff, I recognized that Node.js has an own installer in their unofficial builds repo, which does almost exactly what we do, but with support for target version prefixes (`v24, v24.1`, ...), custom install path (defaulting to `/usr/local`, too), and SHA256 integrity check: https://github.com/nodejs/unofficial-builds/blob/main/www/install-node.sh

So kinda a better version of ours ... at least before we just implemented support for pulling the latest LTS ...
I just pinged the maintainer who added that script just earlier this year, and offered to implement LTS support to theirs. Then we could archive our installer, referring to theirs for good.