MichaIng/DietPi #7325: /boot/dietpi/dietpi-software: line 7328: /etc/fail2ban/filter.d/dropbear.local: No such file or directory

augustresende, 2024-12-25:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9
- Distro version | bookworm 0
- Kernel version | Linux dietpi0 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
- SBC model | RPi 3 Model B (aarch64)
- Power supply used |  N/A
- SD card used | SanDisk Ultra

#### Additional Information (if applicable)
- Software title | Fail2Ban
- Was the software title installed freshly or updated/migrated? Installed freshly
- Can this issue be replicated on a fresh installation of DietPi? Probably yes

- Bug report ID | b5ec861b-ef00-4ce4-ac17-5afd31a6bb75

#### Steps to reproduce
1. Install fail2ban after uninstalling it.
2. See the error `/boot/dietpi/dietpi-software: line 7328: /etc/fail2ban/filter.d/dropbear.local: No such file or directory`.

#### Expected behaviour
- No error and dropbear.local file creation.

#### Actual behaviour
- dropbear.local file is not created.

Joulinar, 2024-12-26:
> DietPi version | 9

What is the exact DietPi version you are on?

augustresende, 2024-12-26:
> > DietPi version | 9
> 
> What is the exact DietPi version you are on?

DietPi v9.9.0

MichaIng, 2024-12-27:
Thanks for reporting. We try to create the file before the package is installed, and hence it's directory is not assured to exist.

I'll try to create a live patch as well, since this patch is quite important.

Reminds me to check back the status of the upstream fix: https://github.com/fail2ban/fail2ban/pull/3597

MichaIng, 2024-12-27:
Fixed with: https://github.com/MichaIng/DietPi/commit/ec4c1e5

closed 2024-12-27 by MichaIng

augustresende, 2024-12-27:
> Fixed with: [ec4c1e5](https://github.com/MichaIng/DietPi/commit/ec4c1e5)

Thank you @MichaIng! That was really fast.