MichaIng/DietPi #7326: Sonarr v4 fail to start after upgrade version 9.9

doopler, 2024-12-26:
i did an upgrade of sonarr from v3 to v4 via dietpi-software reinstall 144
and also removed mono.
and now it failed to start 
here are my logs:
sonarr.service: Failed to set up mount namespacing: /run/systemd/unit-root/usr/lib/sonarr: No such file or directorysonarr.service: Failed at step NAMESPACE spawning /opt/sonarr/Sonarr: No such file or directory
thanks

doopler, 2024-12-26:
as it turns out sonarr was installed in /opt/Sonarr and not in /opt/sonarr/Sonarr
so i changed the path and it worked.
but i lost al the v3 DB and had to re configure from scrach ☹️

closed 2024-12-27 by doopler

MichaIng, 2024-12-27:
`dietpi-software` installs Sonarr v4 to `/opt/sonarr`. How did you originally install and migrate it?

On a test migration I also lost the one test series I added. Sonarr docs state that I attempts to migrate as much as possible and also firat startup logs show a database migration. So I also was confused. However one can import everything from the download directory structure, at least.

doopler, 2024-12-30:
yep thats what i eventually ended up doing.

> dietpi-software installs Sonarr v4 to /opt/sonarr. How did you originally install and migrate it?
i installed v3  a while ago via `dietpi-software` as i also did this time (v4).
but it was installed in /`opt/sonarr/Sonarr` thats why i got this error.

regarding the database i did a manual backup before the upgrade an after  i got sonarr to start in was a blank database, so i tried to restore from a backup but it hanged so i had to completely reinstall it and manually add al the shows from scratch.