MichaIng/DietPi #7327: Radxa ROCK 3B Failed Install

MidG971, 2024-12-26:
#### Details:
- Date           | 2024-12-26 16:02:16
- Program name   | DietPi-Installer
- Command        | `apt-get -y autopurge gawk original-awk`
- Exit code      | 100
- DietPi version | v9.9.0 (MichaIng/master)
- Distro version | bullseye (ID=6,RASPBIAN=0)
- Kernel version | `Linux rock-3b 5.10.160-18-rk356x #425b07550 SMP Thu Oct 12 08:48:55 UTC 2023 aarch64 GNU/Linux`
- Architecture   | `arm64`
- Hardware model |  (ID=22)
- Power supply   | (EG: RAVPower 5V 1A)
- SD card        | (EG: SanDisk Ultra 16 GB)
#### Steps to reproduce:

1. Launch the script , choose generic device
2. ...
#### Expected behaviour:

- Install DietPi
#### Actual behaviour:

- It  fails , trying to delete something that doesn't exist apparently
#### Extra details:

- cf attached log from 
#### Additional logs:
```
Reading package lists...
Building dependency tree...
Reading state information...
Package 'gawk' is not installed, so not removed
Package 'original-awk' is not installed, so not removed
The following packages will be REMOVED:
  radxa-sddm-theme
0 upgraded, 0 newly installed, 1 to remove and 438 not upgraded.
1 not fully installed or removed.
After this operation, 17.4 kB disk space will be freed.
(Reading database ... 63364 files and directories currently installed.)
Removing radxa-sddm-theme (0.1.1) ...
rm: cannot remove '/usr/share/sddm/themes/breeze/Main.qml': No such file or directory
dpkg: error processing package radxa-sddm-theme (--remove):
 installed radxa-sddm-theme package post-removal script subprocess returned error exit status 1
dpkg: too many errors, stopping
Errors were encountered while processing:
 radxa-sddm-theme
Processing was halted because there were too many errors.
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
[log_failed_install_Rock3B.txt](https://github.com/user-attachments/files/18253984/log_failed_install_Rock3B.txt)

MichaIng, 2024-12-27:
That Radxa package `postrm` script tries to remove a file which does not exist. Create a dummy file at that location, then retry to purge it:
```sh
sudo mkdir -p /usr/share/sddm/themes/breeze
sudo touch /usr/share/sddm/themes/breeze/Main.qml
sudo apt purge radxa-sddm-theme
sudo rmdir -p /usr/share/sddm/themes/breeze
```

MidG971, 2024-12-27:
I did as requested, with slight changes:

root@rock-3b:/tmp/DietPi-Installer# sudo mkdir -p /usr/share/sddm/themes/breeze
root@rock-3b:/tmp/DietPi-Installer# sudo touch /usr/share/sddm/themes/breeze/Main.qml
root@rock-3b:/tmp/DietPi-Installer# sudo apt purge radxa-sddm-theme
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  radxa-sddm-theme
0 upgraded, 0 newly installed, 1 to remove and 438 not upgraded.
1 not fully installed or removed.
After this operation, 17.4 kB disk space will be freed.
Do you want to continue? [Y/n]
(Reading database ... 63362 files and directories currently installed.)
Removing radxa-sddm-theme (0.1.1) ...
Removing 'diversion of /usr/share/sddm/themes/breeze/Main.qml to /usr/share/sddm/themes/breeze/Main.qml.bak by radxa-sddm-theme'
Processing triggers for desktop-file-utils (0.26-1) ...
Processing triggers for initramfs-tools (0.142+deb12u1) ...
update-initramfs: Generating /boot/initrd.img-5.10.160-38-rk356x
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for libglib2.0-0:arm64 (2.66.8-1+deb11u5) ...
Processing triggers for libc-bin (2.36-9+deb12u9) ...
Processing triggers for dbus (1.12.28-0+deb11u1) ...
Processing triggers for shared-mime-info (2.0-1) ...
Processing triggers for udev (247.3-7+deb11u6) ...
Processing triggers for libgdk-pixbuf-2.0-0:arm64 (2.42.2+dfsg-1+deb11u2) ...
Processing triggers for fontconfig (2.13.1-4.2) ...
root@rock-3b:/tmp/DietPi-Installer# sudo rmdir -p /usr/share/sddm/themes/breeze
rmdir: failed to remove directory '/usr/share': Directory not empty
root@rock-3b:/tmp/DietPi-Installer# sudo rm -d -f -r -v /usr/share/sddm/themes/breeze **I did this manually, then exited the shell**
root@rock-3b:/tmp/DietPi-Installer#

Afterwards the install is successful but I lose my two onboard NICs, the wlan is still ok, but I get some errors on the bluetooth side.. I've got the dmseg output linked. I tried to install the audio, but failed also.
[dmesg_OK_install_NO_NICs_Rock3B.txt](https://github.com/user-attachments/files/18260604/dmesg_OK_install_NO_NICs_Rock3B.txt)

MidG971, 2024-12-27:
I know that, the board is in the kernel [https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3568-rock-3b.dts](url) since v6.10... 
Can I fellow this method you gave me previously for my other sbc (Rock4SE) :  
[image]

If yes which binary should I use ?

MichaIng, 2024-12-30:
Why did you use a vendor image, and not our ROCK 3A image? While it can work with any pre-image, there is not guarantee, as especially these vendor images are pretty messy and kinda malformed often. The Radxa package which tries to remove a non-existing file and even fails to uninstall completely, if that file does not exist, is a prefect example.

Please try our image first, before we start to debug an issue which might be entirely related to the vendor pre-image, like faulty firmware placed into the filesystem without APT package reference and such stuff.

MidG971, 2024-12-30:
I used the vendor image because the Rock **3B** is not supported nor proposed to be downloaded, only the Rock **3A**. They do  share the same CPU but the  3B has two NICs instead of one. Also there's the fact that for instance , there's a difference of implementation of the DRAM on the Rock **4** and Rock **4SE**, I figured that I should choose the generic device road to be sure. I'll try the Rock **3A** firmware on it and will give you the feedback.

MichaIng, 2024-12-30:
Ah right, I missed the little important difference 3A and 3B 😉.

Hmm, based on your logs, I guess the Debian Bookworm firmware packages cause the issue, as the Radxa image contains `/etc/apt/apt.conf.d/no-bookworm-firmware.conf`, which is removed by our installer. And then there is a package removed `rockchip-iqfiles-rk356x`, where I do not know what it contains.

However, I'd try something different, as you recognised, mainline Linux supports this SBC already. Do you have a WiFi adapter at hand, to establish Internet access with that one temporarily? If so:
1. Flash the ROCK 3A image
1. Set up Internet via WiFi adapter (or any other USB network adapter)
1. Install a Linux 6.12 package I just built:
    ```sh
    cd /tmp
    wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-rockchip64.deb
    dpkg -i linux-{image,dtb}-current-rockchip64.deb
    ```
1. Apply the now available ROCK 3B device tree overlay
    ```sh
    G_CONFIG_INJECT 'fdtfile=' 'fdtfile=rockchip/rk3568-rock-3b.dtb' /boot/dietpiEnv.txt
    reboot
    ```

If this works, we can add some preliminary support to DietPi. If upstream U-Boot supports it as well, we can add support to our Armbian fork natively.

MidG971, 2024-12-30:
It happens 😄
I have a Wifi adapter on the M.2 B slot, I'll do as you recommended.

MichaIng, 2024-12-30:
Okay, kernel build has finished, hence this could be tried now.

MidG971, 2024-12-30:
Using the Rock 3A image is fine as far as: 

-  I added my wifi info in the config file before booting the board. It worked fine, no NICs, but Wifi OK. 

- When I pushed the new ROCK 3B device tree overlay , I lost the connection, I'm currently using serial console as I don't have access to a hdmi screen at the moment. After the first reboot, I tried to reboot numerous times, tried to access the serial console, with the default speeds  9600 / 115200 bps , I only get scrambled characters that one would get if the speed isn't correctly configured.

MichaIng, 2024-12-30:
Did you upgrade the kernel first, and does it contain the expected device tree?
```sh
ls -l /boot/dtb/rockchip/rk3568-rock-3b.dtb
```

MidG971, 2024-12-30:
Its partially ok now :
i'll give you a detailed report when I get home, but to be short, leds dont turn on when cable are plugged and only one NICs up at the time. Its a mess to test right now. I do it later. Also the dietpi-config menu detects only one NIC 

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether e2:e1:76:03:12:f8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.100/24 brd 192.168.0.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet 192.168.75.118/24 brd 192.168.75.255 scope global dynamic eth0
       valid_lft 28083sec preferred_lft 28083sec
    inet6 fe80::e0e1:76ff:fe03:12f8/64 scope link
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether e2:e1:76:03:12:f9 brd ff:ff:ff:ff:ff:ff
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether c8:21:58:b7:a7:b9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.75.215/24 brd 192.168.75.255 scope global dynamic wlan0
       valid_lft 28089sec preferred_lft 28089sec
    inet6 fe80::ca21:58ff:feb7:a7b9/64 scope link
       valid_lft forever preferred_lft forever

```

- in the dietpi-config menu I can only see one NIC and it keep showing me this parameters, despite the changes I made ( switch to dhcp ) 👍 

```

┌──────────────────────────────┤ DietPi-Config ├───────────────────────────────┐
│ Ethernet Details:                                                            │
│ Usage   : Sent = 0 MiB | Received = 0 MiB                                    │
│ Address : IP = 192.168.0.100 | Mask = 255.255.255.0 | Gateway =  | DNS =     │
│ 192.168.254.30                                                               │
│                                                                              │
│                        ●─ DHCP/STATIC IP ──────────────────────●             │
│            Change Mode : [DHCP]                                              │
│                        ●─ Additional Options ──────────────────●             │
│            Link Speed  : [auto (default)]                                    │
│            Disable     : Disable Ethernet adapter                            │
│                        ●─ Apply ───────────────────────────────●             │
│            Apply       : Save all changes and restart networking             │
│                                                                              │
│                                                                              │
│                     <Select>                     <Back>                      │
│                                                                              │
└──────────────────────────────────────────────────────────────────────────────┘

```
Thi

here's the journactl :
```
root@DietPi:~# journalctl
Dec 30 17:21:38 DietPi kernel: Booting Linux on physical CPU 0x0000000000 [0x412fd050]
Dec 30 17:21:38 DietPi kernel: Linux version 6.12.7-current-rockchip64 (build@armbian) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Fri Dec 27 13:02:20 UTC 2024
Dec 30 17:21:38 DietPi kernel: KASLR disabled due to lack of seed
Dec 30 17:21:38 DietPi kernel: Machine model: Radxa ROCK 3B
Dec 30 17:21:38 DietPi kernel: efi: UEFI not found.
Dec 30 17:21:38 DietPi kernel: NUMA: Faking a node at [mem 0x0000000000200000-0x00000001ffffffff]
Dec 30 17:21:38 DietPi kernel: NODE_DATA(0) allocated [mem 0x1fefdb640-0x1fefdddbf]
Dec 30 17:21:38 DietPi kernel: Zone ranges:
Dec 30 17:21:38 DietPi kernel:   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
Dec 30 17:21:38 DietPi kernel:   DMA32    empty
Dec 30 17:21:38 DietPi kernel:   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
Dec 30 17:21:38 DietPi kernel: Movable zone start for each node
[... 1126 lines cut ...]
Dec 30 17:31:42 DietPi kernel: wlan0: associate with f0:1d:2d:14:25:2f (try 1/3)
Dec 30 17:31:42 DietPi kernel: wlan0: RX AssocResp from f0:1d:2d:14:25:2f (capab=0x1111 status=0 aid=3)
Dec 30 17:31:42 DietPi kernel: wlan0: associated
Dec 30 17:31:42 DietPi wpa_supplicant[2564]: wlan0: Associated with f0:1d:2d:14:25:2f
Dec 30 17:31:42 DietPi wpa_supplicant[2564]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Dec 30 17:31:42 DietPi wpa_supplicant[2564]: wlan0: WPA: Key negotiation completed with f0:1d:2d:14:25:2f [PTK=CCMP GTK=CCMP]
Dec 30 17:31:42 DietPi kernel: wlan0: Limiting TX power to 17 (20 - 3) dBm as advertised by f0:1d:2d:14:25:2f
Dec 30 17:31:42 DietPi wpa_supplicant[2564]: wlan0: CTRL-EVENT-CONNECTED - Connection to f0:1d:2d:14:25:2f completed [id=0 id_str=]
Dec 30 17:34:01 DietPi kernel: wlan0: disassociated from f0:1d:2d:14:25:2f (Reason: 1=UNSPECIFIED)
Dec 30 17:34:01 DietPi wpa_supplicant[2564]: wlan0: CTRL-EVENT-DISCONNECTED bssid=f0:1d:2d:14:25:2f reason=1
Dec 30 17:34:01 DietPi wpa_supplicant[2564]: BSSID f0:1d:2d:14:25:2f ignore list count incremented to 2, ignoring for 10 seconds
[... cut ...]
Dec 30 17:34:01 DietPi wpa_supplicant[2564]: wlan0: Associated with f0:1d:2d:14:25:2f
Dec 30 17:34:01 DietPi wpa_supplicant[2564]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Dec 30 17:34:01 DietPi wpa_supplicant[2564]: wlan0: WPA: Key negotiation completed with f0:1d:2d:14:25:2f [PTK=CCMP GTK=CCMP]
Dec 30 17:34:01 DietPi wpa_supplicant[2564]: wlan0: CTRL-EVENT-CONNECTED - Connection to f0:1d:2d:14:25:2f completed [id=0 id_str=]
Dec 30 17:34:01 DietPi kernel: wlan0: Limiting TX power to 17 (20 - 3) dBm as advertised by f0:1d:2d:14:25:2f
```

Joulinar, 2024-12-30:
`dietpi-config` is not supporting 2 ethernet adapter. This is a known limitation actually. 2nd NIC adapter would need to be configured manually.

MidG971, 2024-12-31:
@Joulinar : I wasn't aware of that, my bad.

Report: 

Below I've got the two NICs plugged to my router, only eth0 aka eth2 physically is up. Even with a fresh restart eth1 aka eth1 physically stays down.
```
root@DietPi:~# ethtool eth0
Settings for eth0:
        Supported ports: [ TP    MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  10baseT/Half 10baseT/Full
                                             100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: Symmetric Receive-only
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Auto-negotiation: on
        master-slave cfg: preferred slave
        master-slave status: slave
        Port: Twisted Pair
        PHYAD: 1
        Transceiver: external
        MDI-X: Unknown
        Supports Wake-on: ug
        Wake-on: d
        Current message level: 0x0000003f (63)
                               drv probe link timer ifdown ifup
        Link detected: yes
root@DietPi:~# ethtool eth1
Settings for eth1:
        Supported ports: [ TP    AUI     MII     FIBRE   BNC     Backplane ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Half 1000baseT/Full
                                1000baseKX/Full
                                1000baseX/Full
                                100baseT1/Full
                                1000baseT1/Full
                                100baseFX/Half 100baseFX/Full
                                10baseT1L/Full
                                10baseT1S/Full
                                10baseT1S/Half
                                10baseT1S_P2MP/Half
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  Not reported
        Advertised pause frame use: No
        Advertised auto-negotiation: No
        Advertised FEC modes: Not reported
        Speed: Unknown!
        Duplex: Half
        Auto-negotiation: off
        Port: MII
        PHYAD: 0
        Transceiver: internal
        Supports Wake-on: ug
        Wake-on: d
        Current message level: 0x0000003f (63)
                               drv probe link timer ifdown ifup
        Link detected: no
 
````
I'll do a lspci on the vendor image as soon as possible.

MidG971, 2024-12-31:
Here's more info :
dmesg :
```
root@DietPi:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.12.7-current-rockchip64 (build@armbian) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Fri Dec 27 13:02:20 UTC 2024
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: Radxa ROCK 3A
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] NODE_DATA(0) allocated [mem 0x1fefd8080-0x1fefda7ff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 128 MiB at 0x00000000e8000000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 34 pages/cpu s98968 r8192 d32104 u139264
[    0.000000] pcpu-alloc: s98968 r8192 d32104 u139264 alloc=34*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: root=UUID=463087c8-1d8f-4075-b7b1-4b86c641e99e rootfstype=ext4 rootwait console=ttyS2,1500000 console=tty1 consoleblank=0 coherent_pool=2M usb-storage.quirks= net.ifnames=0 androidboot.fwver=ddr-v1.21-2d653b3476,bl31-v1.44,uboot-rmbian-201-10/09/2024
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2031104
[    0.000000] Policy zone: Normal
[    0.000000] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000e4000000-0x00000000e8000000] (64MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: MBI range [296:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: GICD_CTRL.DS=0, SCR_EL3.FIQ=1
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[... 1720 lines cut ...]
[   25.798048] rk_gmac-dwmac fe2a0000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[   25.798165] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   25.926347] EXT4-fs (mmcblk0p3): resizing filesystem from 1069920 to 7548923 blocks
[   26.276878] EXT4-fs (mmcblk0p3): resized filesystem to 7548923
[   27.719628] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
[   31.957775] Bluetooth: RFCOMM TTY layer initialized
[   31.957811] Bluetooth: RFCOMM socket layer initialized
[   31.957865] Bluetooth: RFCOMM ver 1.11
[   32.728051] Bluetooth: hci0: BCM: failed to write update baudrate (-110)
[   32.728132] Bluetooth: hci0: Failed to set baudrate
[   34.861509] Bluetooth: hci0: command 0x0c03 tx timeout
[... cut ...]
[  191.435002] wlp1s0: RX AssocResp from 3a:07:16:e3:53:80 (capab=0x1111 status=0 aid=3)
[  191.437516] wlp1s0: associated
[  191.496477] IPv6: ADDRCONF(NETDEV_CHANGE): wlp1s0: link becomes ready
[  191.534979] wlp1s0: Limiting TX power to 30 (30 - 0) dBm as advertised by 3a:07:16:e3:53:80
[  772.820998] rockchip-vop2 fe040000.vop: [drm:vop2_crtc_atomic_disable] Crtc atomic disable vp0
[ 2620.706093] rockchip-vop2 fe040000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 1920x1080p60, type: 11(if:800) for vp0 dclk: 148500000
rock@rock-3b:~$ ~

```

MichaIng, 2024-12-31:
For the second adapter, you could define in a own config, like:
```sh
cat << '_EOF_' > /etc/network/interfaces.d/eth1.conf
allow-hotplug eth1
iface eth1 inet static
address 192.168.1.100/24
_EOF_
systemctl daemon-reload
ifup eth1
```
depending on what you aim to do with it.

MidG971, 2024-12-31:
I plan to either use it as a WAN or LAN port for a firewall, or with eth0 as a part of a etherchannel / LACP port, not quite sure yet...

MichaIng, 2024-12-31:
Yeah, above just an example. `dietpi-config` will always setup `eth0` as main Interface with default gateway etc, so of course you can also define both by yourself, and comment the lines in `/etc/network/interfaces`. Aside of that, it is normal that an interface is even physically down (no carrier signal), as long as it has not been brought up if `ifup` (resp. `/etc/network/interfaces`/`*.d/` config) or any other network stack.

MidG971, 2024-12-31:
I tried your solution, the adapter won't show up, only eth0 & the wlan are available

MidG971, 2024-12-31:
after a check, I discovered that :

````
root@DietPi:/boot# cat dietpiEnv.txt
rootdev=UUID=463087c8-1d8f-4075-b7b1-4b86c641e99e
rootfstype=ext4
# The init system logs to the console defined last.
consoleargs=console=ttyS2,1500000 console=tty1
usbstoragequirks=
extraargs=net.ifnames=0
docker_optimizations=off
overlay_path=rockchip
overlay_prefix=rockchip
overlays=
user_overlays=
fdtfile=rockchip/rk3568-rock-3a.dtb

````

I manually modified the last line to fdtfile=rockchip/rk3568-rockb-3.dtb , now I've got my two NICs present.

MichaIng, 2025-01-01:
Great. So we can add support for the SBC in general, using ROCK 3A bootloader and adjusting the device tree only. Of course a ROCK 3B bootloader would be better, for network boot capabilities, but a rare use case.

MidG971, 2025-01-02:
Is there some benchmark or or test or procedure that I can do, for you to validate the support of the board ?

MichaIng, 2025-01-02:
Not really needed. If it boots and those hardware features which differ from the ROCK 3A work, then all is good.

MidG971, 2025-01-02:
So it's 99.9% OK as I can't use both NICs at the same time yet...

MichaIng, 2025-01-02:
But both do now show up in `ip l`? Then it is all perfectly as expected, and you can configure the second NIC manually, as explained [above](https://github.com/MichaIng/DietPi/issues/7327#issuecomment-2566122563), or use another network stack.

MidG971, 2025-01-02:
I tried the method you provided / tried with nmcli / mntui , they do both appear when I use `ip a` but if I unplug eth0 (the one that works perfectly)  to leave only eth1 plugged I'm only left with the wifi. 

Those are when only eth1 is plugged :

````
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether e2:e1:76:03:12:f8 brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether e2:e1:76:03:12:f9 brd ff:ff:ff:ff:ff:ff
4: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether c8:21:58:b7:a7:b9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.75.103/24 brd 192.168.75.255 scope global dynamic noprefixroute wlan0
       valid_lft 27105sec preferred_lft 27105sec
    inet6 fe80::761c:70db:16db:d051/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
root@DietPi:~# nmcli device
DEVICE         TYPE      STATE                   CONNECTION
wlan0          wifi      connected               morioh
lo             loopback  connected (externally)  lo
eth1           ethernet  disconnected            --
p2p-dev-wlan0  wifi-p2p  disconnected            --
eth0           ethernet  unavailable             --
root@DietPi:~#

````

Also when I try to activate eth1 with nmtui it fails

MichaIng, 2025-01-02:
How did you try to configure `eth1`?  And what are the error messages?

MidG971, 2025-01-02:
I used your method 
`root@DietPi:~# cat /etc/network/interfaces.d/eth1.conf
allow-hotplug eth1
iface eth1 inet dhcp
address 192.168.1.100/24
`
and also : 

```
root@DietPi:~# cat /etc/network/interfaces
auto lo
iface lo inet loopback

````
`
Below the error message in nmtui
```
Could not activate connection:      │     │
                                                                    │ │ │ Activation failed: The device could │     │
                                                                    │ │ │ not be readied for configuration  
````
**dmesg** :
```
[ 4879.022185] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 4879.023671] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 4879.052548] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 4879.053725] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 4879.083293] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 4879.084667] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 4879.110227] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 4879.111272] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5179.025176] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5179.026180] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5179.052636] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5179.053727] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5179.078169] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5179.079219] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5179.103453] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5179.104593] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5479.026269] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5479.027284] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5479.051248] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5479.052357] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5479.076448] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5479.077491] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5479.102141] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5479.103186] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5779.019962] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5779.021480] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5779.049341] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5779.050412] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5779.080915] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5779.082346] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 5779.106237] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 5779.107239] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 6079.027204] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 6079.028218] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 6079.053214] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 6079.054293] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 6079.080016] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 6079.081153] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 6079.106868] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 6079.107866] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 6379.021878] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 6379.022900] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 6379.047653] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 6379.048823] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 6379.072991] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[ 6379.074035] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[ 6379.097699] rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[... 639 lines cut ...]
Jan 02 13:26:36 DietPi NetworkManager[414]: <info>  [1735820796.6089] device (eth1): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Jan 02 13:26:36 DietPi NetworkManager[414]: <info>  [1735820796.6129] device (eth1): state change: config -> failed (reason 'config-failed', sys-iface-state: 'managed')
Jan 02 13:26:36 DietPi kernel: rk_gmac-dwmac fe2a0000.ethernet eth1: Register MEM_TYPE_PAGE_POOL RxQ-0
[... cut ...]
Jan 02 13:26:36 DietPi kernel: rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
Jan 02 13:26:36 DietPi NetworkManager[414]: <info>  [1735820796.6642] device (eth1): state change: config -> failed (reason 'config-failed', sys-iface-state: 'managed')
Jan 02 13:26:36 DietPi NetworkManager[414]: <warn>  [1735820796.6656] device (eth1): Activation: failed for connection 'Wired connection 2'
Jan 02 13:26:36 DietPi NetworkManager[414]: <info>  [1735820796.6664] device (eth1): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
root@DietPi:~#

````

MidG971, 2025-01-02:
````
root@DietPi:~# nmcli connection show Wired\ connection\ 1
connection.id:                          Wired connection 1
connection.uuid:                        416424b2-1093-3eb4-914a-89e0972c4c94
connection.stable-id:                   --
connection.type:                        802-3-ethernet
connection.interface-name:              eth0
connection.autoconnect:                 yes
connection.autoconnect-priority:        -999
connection.autoconnect-retries:         -1 (default)
connection.multi-connect:               0 (default)
connection.auth-retries:                -1
connection.timestamp:                   1735815337
connection.read-only:                   no
connection.permissions:                 --
connection.zone:                        --
connection.master:                      --
connection.slave-type:                  --
connection.autoconnect-slaves:          -1 (default)
connection.secondaries:                 --
connection.gateway-ping-timeout:        0
connection.metered:                     unknown
connection.lldp:                        default
connection.mdns:                        -1 (default)
connection.llmnr:                       -1 (default)
connection.dns-over-tls:                -1 (default)
connection.mptcp-flags:                 0x0 (default)
connection.wait-device-timeout:         -1
connection.wait-activation-delay:       -1
802-3-ethernet.port:                    --
802-3-ethernet.speed:                   0
802-3-ethernet.duplex:                  --
802-3-ethernet.auto-negotiate:          no
802-3-ethernet.mac-address:             --
802-3-ethernet.cloned-mac-address:      --
802-3-ethernet.generate-mac-address-mask:--
802-3-ethernet.mac-address-blacklist:   --
802-3-ethernet.mtu:                     auto
802-3-ethernet.s390-subchannels:        --
802-3-ethernet.s390-nettype:            --
802-3-ethernet.s390-options:            --
802-3-ethernet.wake-on-lan:             default
802-3-ethernet.wake-on-lan-password:    --
802-3-ethernet.accept-all-mac-addresses:-1 (default)
ipv4.method:                            auto
ipv4.dns:                               --
ipv4.dns-search:                        --
ipv4.dns-options:                       --
ipv4.dns-priority:                      0
ipv4.addresses:                         --
ipv4.gateway:                           --
ipv4.routes:                            --
ipv4.route-metric:                      -1
ipv4.route-table:                       0 (unspec)
ipv4.routing-rules:                     --
ipv4.replace-local-rule:                -1 (default)
ipv4.ignore-auto-routes:                no
ipv4.ignore-auto-dns:                   no
ipv4.dhcp-client-id:                    --
ipv4.dhcp-iaid:                         --
ipv4.dhcp-timeout:                      0 (default)
ipv4.dhcp-send-hostname:                yes
ipv4.dhcp-hostname:                     --
ipv4.dhcp-fqdn:                         --
ipv4.dhcp-hostname-flags:               0x0 (none)
ipv4.never-default:                     no
ipv4.may-fail:                          no
ipv4.required-timeout:                  -1 (default)
ipv4.dad-timeout:                       -1 (default)
ipv4.dhcp-vendor-class-identifier:      --
[... 119 lines cut ...]
ipv6.routing-rules:                     --
ipv6.replace-local-rule:                -1 (default)
ipv6.ignore-auto-routes:                no
ipv6.ignore-auto-dns:                   no
ipv6.never-default:                     no
ipv6.may-fail:                          yes
ipv6.required-timeout:                  -1 (default)
ipv6.ip6-privacy:                       -1 (unknown)
ipv6.addr-gen-mode:                     default
ipv6.ra-timeout:                        0 (default)
ipv6.mtu:                               auto
ipv6.dhcp-duid:                         --
ipv6.dhcp-iaid:                         --
ipv6.dhcp-timeout:                      0 (default)
ipv6.dhcp-send-hostname:                yes
ipv6.dhcp-hostname:                     --
ipv6.dhcp-hostname-flags:               0x0 (none)
ipv6.auto-route-ext-gw:                 -1 (default)
ipv6.token:                             --
proxy.method:                           none
proxy.browser-only:                     no
proxy.pac-url:                          --
proxy.pac-script:                       --

````

MidG971, 2025-01-02:
```root@DietPi:~# inxi -xxx -N -a -i
Network:
  Device-1: Intel Wireless 7265 driver: iwlwifi v: kernel bus-ID: 0000:01:00.0 chip-ID: 8086:095a
    class-ID: 0280
  IF: wlan0 state: up mac: c8:21:58:b7:a7:b9
  IP v4: 192.168.75.103/24 type: dynamic noprefixroute scope: global broadcast: 192.168.75.255
  IP v6: fe80::761c:70db:16db:d051/64 type: noprefixroute scope: link
  Device-2: rk3568-gmac driver: rk_gmac_dwmac v: N/A port: N/A bus-ID: N/A
    chip-ID: rockchip:fe010000 class-ID: ethernet
  IF: eth0 state: down mac: e2:e1:76:03:12:f8
  Device-3: rk3568-gmac driver: rk_gmac_dwmac v: N/A port: N/A bus-ID: N/A
    chip-ID: rockchip:fe2a0000 class-ID: ethernet
  IF: eth1 state: down mac: e2:e1:76:03:12:f9
```

MichaIng, 2025-01-03:
Okay, seems something wrong with the driver or device tree:
```
__stmmac_open: Cannot attach to PHY (error: -19)
```

Mainline device tree: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3568-rock-3b.dts
There are no patches done.
Interestingly, neither Armbian nor Radxa have one in their vendor kernel sources:
- https://github.com/armbian/linux-rockchip/tree/rk-6.1-rkr4.1/arch/arm64/boot/dts/rockchip
- https://github.com/radxa/kernel/tree/linux-6.1-stan-rkr1/arch/arm64/boot/dts/rockchip

... ah, Linux 5.10 only:
- https://github.com/radxa/kernel/blob/linux-5.10-gen-rkr4.1/arch/arm64/boot/dts/rockchip/rk3568-rock-3b.dts

Parts to compare:
- https://github.com/radxa/kernel/blob/linux-5.10-gen-rkr4.1/arch/arm64/boot/dts/rockchip/rk3568-rock-3b.dts#L996-L1022
- https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3568-rock-3b.dts#n200

MidG971, 2025-01-03:
so the next step would be to compile the latest dietpi firmware with the 5.10 drivers ?

MichaIng, 2025-01-04:
No, we won't provide Linux 5.10 images, much too old to consider for a new(ly supported) board.

I will try to patch in some changes derived from the ROCK 3B device tree. But if this does not help, we need to wait until someone more experienced in kernel development runs into the issue and finds a proper solution.

MidG971, 2025-01-04:
Ok thanks 😊

MichaIng, 2025-01-09:
The Orange Pi 3B v1.1 seems to suffer from the same issue: https://dietpi.com/forum/t/22129/32

MichaIng, 2025-01-09:
So this could be tested, a fresh build with all Armbian patches removed, just to rule out it is not something conflicting with a recent upstream change:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-current-rockchip64.deb
dpkg -i linux-{dtb,image}-current-rockchip64.deb
reboot
```

MidG971, 2025-01-10:
> So this could be tested, a fresh build with all Armbian patches removed, just to rule out it is not something conflicting with a recent upstream change:
> 
> ```shell
> cd /tmp
> wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-current-rockchip64.deb
> dpkg -i linux-{dtb,image}-current-rockchip64.deb
> reboot
> ```

Will test it when I get back.

MidG971, 2025-01-11:
> ```shell
> cd /tmp
> wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-rockchip64.deb
> dpkg -i linux-{image,dtb}-current-rockchip64.deb
> ```

you meant 

 `cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-rockchip64.deb
dpkg -i linux-{image,dtb}-current-rockchip64.deb`

image and dtb are inverted ;)

MidG971, 2025-01-11:
> Will test it when I get back.

So I loose the wlan this time. Only one NIC ok, always the same.
```
root@DietPi:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.12.8-current-rockchip64 (build@armbian) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT Thu Jan  2 09:34:26 UTC 2025
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: Radxa ROCK 3B
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] NODE_DATA(0) allocated [mem 0x1ff01de80-0x1ff0205ff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 128 MiB at 0x00000000e8000000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 34 pages/cpu s98968 r8192 d32104 u139264
[    0.000000] pcpu-alloc: s98968 r8192 d32104 u139264 alloc=34*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: root=UUID=463087c8-1d8f-4075-b7b1-4b86c641e99e rootfstype=ext4 rootwait console=ttyS2,1500000 console=tty1 consoleblank=0 coherent_pool=2M usb-storage.quirks= net.ifnames=0 androidboot.fwver=ddr-v1.21-2d653b3476,bl31-v1.44,uboot-rmbian-201-10/09/2024
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2031104
[    0.000000] Policy zone: Normal
[    0.000000] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000e4000000-0x00000000e8000000] (64MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: MBI range [296:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: GICD_CTRL.DS=0, SCR_EL3.FIQ=1
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[... 470 lines cut ...]
[   13.166332] rk_gmac-dwmac fe010000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   19.689549] platform fd000000.usb: deferred probe pending: platform: supplier fe830000.phy not ready
[   19.689597] platform 3c0000000.pcie: deferred probe pending: platform: supplier fe840000.phy not ready
[   19.689623] platform fcc00000.usb: deferred probe pending: platform: supplier fe820000.phy not ready
[   33.768570] vdd_npu: disabling
root@DietPi:~#

````

````
root@DietPi:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether e2:e1:76:03:12:f8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.109/24 brd 192.168.1.255 scope global dynamic eth0
       valid_lft 42778sec preferred_lft 42778sec
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether e2:e1:76:03:12:f9 brd ff:ff:ff:ff:ff:ff

````

MidG971, 2025-01-11:
journalctl : 

```
Jan 11 09:38:31 DietPi kernel: SMP: Total of 4 processors activated.
Jan 11 09:38:31 DietPi kernel: CPU: All CPU(s) started at EL2
Jan 11 09:38:31 DietPi kernel: CPU features: detected: 32-bit EL0 Support
Jan 11 09:38:31 DietPi kernel: CPU features: detected: 32-bit EL1 Support
Jan 11 09:38:31 DietPi kernel: CPU features: detected: Data cache clean to the PoU not required for I/D coherence
Jan 11 09:38:31 DietPi kernel: CPU features: detected: Common not Private translations
Jan 11 09:38:31 DietPi kernel: CPU features: detected: CRC32 instructions
Jan 11 09:38:31 DietPi kernel: CPU features: detected: RCpc load-acquire (LDAPR)
Jan 11 09:38:31 DietPi kernel: CPU features: detected: LSE atomic instructions
Jan 11 09:38:31 DietPi kernel: CPU features: detected: Privileged Access Never
Jan 11 09:38:31 DietPi kernel: CPU features: detected: RAS Extension Support
Jan 11 09:38:31 DietPi kernel: CPU features: detected: Speculative Store Bypassing Safe (SSBS)
Jan 11 09:38:31 DietPi kernel: alternatives: applying system-wide alternatives
Jan 11 09:38:31 DietPi kernel: Memory: 7714316K/8124416K available (16384K kernel code, 2910K rwdata, 11048K rodata, 5120K init, 786K bss, 273968K reserved, 131072K cma-reserved)
Jan 11 09:38:31 DietPi kernel: devtmpfs: initialized
Jan 11 09:38:31 DietPi kernel: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
Jan 11 09:38:31 DietPi kernel: futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
Jan 11 09:38:31 DietPi kernel: 23648 pages in range for non-PLT usage
Jan 11 09:38:31 DietPi kernel: 515168 pages in range for PLT usage
Jan 11 09:38:31 DietPi kernel: pinctrl core: initialized pinctrl subsystem
Jan 11 09:38:31 DietPi kernel: DMI not present or invalid.
Jan 11 09:38:31 DietPi kernel: NET: Registered PF_NETLINK/PF_ROUTE protocol family
Jan 11 09:38:31 DietPi kernel: DMA: preallocated 2048 KiB GFP_KERNEL pool for atomic allocations
Jan 11 09:38:31 DietPi kernel: DMA: preallocated 2048 KiB GFP_KERNEL|GFP_DMA pool for atomic allocations
Jan 11 09:38:31 DietPi kernel: DMA: preallocated 2048 KiB GFP_KERNEL|GFP_DMA32 pool for atomic allocations
Jan 11 09:38:31 DietPi kernel: audit: initializing netlink subsys (disabled)
Jan 11 09:38:31 DietPi kernel: audit: type=2000 audit(0.100:1): state=initialized audit_enabled=0 res=1
Jan 11 09:38:31 DietPi kernel: thermal_sys: Registered thermal governor 'fair_share'
Jan 11 09:38:31 DietPi kernel: thermal_sys: Registered thermal governor 'bang_bang'
Jan 11 09:38:31 DietPi kernel: thermal_sys: Registered thermal governor 'step_wise'
Jan 11 09:38:31 DietPi kernel: thermal_sys: Registered thermal governor 'user_space'
Jan 11 09:38:31 DietPi kernel: cpuidle: using governor menu
Jan 11 09:38:31 DietPi kernel: hw-breakpoint: found 6 breakpoint and 4 watchpoint registers.
Jan 11 09:38:31 DietPi kernel: ASID allocator initialised with 65536 entries
Jan 11 09:38:31 DietPi kernel: Serial: AMBA PL011 UART driver
Jan 11 09:38:31 DietPi kernel: /vop@fe040000: Fixed dependency cycle(s) with /hdmi@fe0a0000
Jan 11 09:38:31 DietPi kernel: /hdmi@fe0a0000: Fixed dependency cycle(s) with /vop@fe040000
Jan 11 09:38:31 DietPi kernel: /pcie@fe260000: Fixed dependency cycle(s) with /pcie@fe260000/legacy-interrupt-controller
Jan 11 09:38:31 DietPi kernel: gpio gpiochip0: Static allocation of GPIO base is deprecated, use dynamic allocation.
Jan 11 09:38:31 DietPi kernel: rockchip-gpio fdd60000.gpio: probed /pinctrl/gpio@fdd60000
Jan 11 09:38:31 DietPi kernel: gpio gpiochip1: Static allocation of GPIO base is deprecated, use dynamic allocation.
Jan 11 09:38:31 DietPi kernel: rockchip-gpio fe740000.gpio: probed /pinctrl/gpio@fe740000
Jan 11 09:38:31 DietPi kernel: gpio gpiochip2: Static allocation of GPIO base is deprecated, use dynamic allocation.
Jan 11 09:38:31 DietPi kernel: rockchip-gpio fe750000.gpio: probed /pinctrl/gpio@fe750000
Jan 11 09:38:31 DietPi kernel: gpio gpiochip3: Static allocation of GPIO base is deprecated, use dynamic allocation.
Jan 11 09:38:31 DietPi kernel: rockchip-gpio fe760000.gpio: probed /pinctrl/gpio@fe760000
Jan 11 09:38:31 DietPi kernel: gpio gpiochip4: Static allocation of GPIO base is deprecated, use dynamic allocation.
Jan 11 09:38:31 DietPi kernel: rockchip-gpio fe770000.gpio: probed /pinctrl/gpio@fe770000
Jan 11 09:38:31 DietPi kernel: /pcie@fe280000: Fixed dependency cycle(s) with /pcie@fe280000/legacy-interrupt-controller
Jan 11 09:38:31 DietPi kernel: /hdmi@fe0a0000: Fixed dependency cycle(s) with /hdmi-con
Jan 11 09:38:31 DietPi kernel: /hdmi-con: Fixed dependency cycle(s) with /hdmi@fe0a0000
Jan 11 09:38:31 DietPi kernel: HugeTLB: registered 1.00 GiB page size, pre-allocated 0 pages
Jan 11 09:38:31 DietPi kernel: HugeTLB: 0 KiB vmemmap can be freed for a 1.00 GiB page
Jan 11 09:38:31 DietPi kernel: HugeTLB: registered 32.0 MiB page size, pre-allocated 0 pages
Jan 11 09:38:31 DietPi kernel: HugeTLB: 0 KiB vmemmap can be freed for a 32.0 MiB page
Jan 11 09:38:31 DietPi kernel: HugeTLB: registered 2.00 MiB page size, pre-allocated 0 pages
Jan 11 09:38:31 DietPi kernel: HugeTLB: 0 KiB vmemmap can be freed for a 2.00 MiB page
Jan 11 09:38:31 DietPi kernel: HugeTLB: registered 64.0 KiB page size, pre-allocated 0 pages
Jan 11 09:38:31 DietPi kernel: HugeTLB: 0 KiB vmemmap can be freed for a 64.0 KiB page
Jan 11 09:38:31 DietPi kernel: cryptd: max_cpu_qlen set to 1000
Jan 11 09:38:31 DietPi kernel: raid6: neonx8   gen()  1407 MB/s
[... cut ...]
 The job identifier is 682 and the job result is dependency.
Jan 11 09:54:22 DietPi systemd[1]: ifup@wlan0.service: Job ifup@wlan0.service/start failed with result 'dependency'.
Jan 11 09:54:29 DietPi sudo[5360]: pam_unix(sudo:session): session closed for user root
Jan 11 09:54:37 DietPi sudo[6409]:   dietpi : TTY=pts/2 ; PWD=/home/dietpi ; USER=root ; COMMAND=/usr/bin/journalctl -xe
Jan 11 09:54:37 DietPi sudo[6409]: pam_unix(sudo:session): session opened for user root(uid=0) by dietpi(uid=1000)
dietpi@DietPi:~$ ```

MichaIng, 2025-01-12:
> image and dtb are inverted ;)

This doesn't matter. `123{foo,bar}456` is a syntax to add two arguments `123foo456` and `123bar456`. And the order of download and dpkg install does not matter.

Okay, does the ROCK 3B also use this `sprdwl_ng` driver for its onboard WiFi? Or maybe another 3rd party driver which I removed together with the patches. On Orange Pi 3B it did not work either (to bring up Ethernet), so the Armbian patches can be ruled out. I'll try now to patch some parts from the vendor device tree into the mainline device tree.

MichaIng, 2025-01-13:
I did a new build, with Armbian patches re-applied + another patch which changes the PHY mode and applies RX/TX delays in the MAC. This is true as well in the vendor kernel. Let's see whether it is the part which fixes the 2nd Ethernet port. Same procedure to apply:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-current-rockchip64.deb
dpkg -i linux-{dtb,image}-current-rockchip64.deb
reboot
```

MidG971, 2025-01-13:
> I did a new build, with Armbian patches re-applied + another patch which changes the PHY mode and applies RX/TX delays in the MAC. This is true as well in the vendor kernel. Let's see whether it is the part which fixes the 2nd Ethernet port. Same procedure to apply:
> 
> ```shell
> cd /tmp
> wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-current-rockchip64.deb
> dpkg -i linux-{dtb,image}-current-rockchip64.deb
> reboot
> ```

Hello, will try at lunch break 😉

MidG971, 2025-01-13:
```
root@DietPi:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.12.9-current-rockchip64 (build@armbian) (aarch64-linux-gn                                                                                                u-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 S                                                                                                MP PREEMPT Thu Jan  9 12:33:55 UTC 2025
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] Machine model: Radxa ROCK 3B
[    0.000000] efi: UEFI not found.
[    0.000000] NUMA: Faking a node at [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] NODE_DATA(0) allocated [mem 0x1fefde540-0x1fefe0cbf]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 128 MiB at 0x00000000e8000000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 34 pages/cpu s98968 r8192 d32104 u139264
[    0.000000] pcpu-alloc: s98968 r8192 d32104 u139264 alloc=34*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 24                                                                                                41009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: root=UUID=463087c8-1d8f-4075-b7b1-4b86c641e99e rootf                                                                                                stype=ext4 rootwait console=ttyS2,1500000 console=tty1 consoleblank=0 coherent_pool=2M u                                                                                                sb-storage.quirks= net.ifnames=0 androidboot.fwver=ddr-v1.21-2d653b3476,bl31-v1.44,uboot                                                                                                -rmbian-201-10/09/2024
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linea                                                                                                r)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2031104
[    0.000000] Policy zone: Normal
[    0.000000] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000e4000000-0x00000000e8000000] (64MB                                                                                                )
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] RCU Tasks: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_                                                                                                cpu_ids=4.
[    0.000000] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu                                                                                                _task_cpu_ids=4.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: MBI range [296:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: GICD_CTRL.DS=0, SCR_EL3.FIQ=1
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[... cut ...]
[   19.433586] platform fcc00000.usb: deferred probe pending: platform: supplier fe82000                                                                                                0.phy not ready
[   19.433632] platform fd000000.usb: deferred probe pending: platform: supplier fe83000                                                                                                0.phy not ready
[   19.433658] platform 3c0000000.pcie: deferred probe pending: platform: supplier fe840                                                                                                000.phy not ready
[   33.768833] vdd_npu: disabling

````

MidG971, 2025-01-13:
I forgot, wifi is still broken.

MichaIng, 2025-01-14:
Thanks for testing. Very weird with the WiFi, as Armbian patches are all back in. The 6.12.7 kernel you used before I uploaded the one with removed patches, had WiFi, right? 🤔
Thing is, on Orange Pi 3B the same happend. I however checked the logs and all patches were indeed applied. So no idea what changed between the 6.12.7 build and 6.12.9 now. Coincidence that it broke at Armbian or upstream?

No related patch changes at Armbian: https://github.com/armbian/build/commits/main/patch/kernel/archive/rockchip64-6.12
At upstream device tree has not changed either.

... ah, I see now in case of ROCK 3B it is an Intel WiFi chip with `iwlwifi` driver, hence all mainlined anyway. No idea why the module is suddenly not loading anymore (automatically). What happens if you try manualy?
```sh
modprobe iwlwifi
ip l
```

MichaIng, 2025-01-14:
Next build ready to test, this time with the clock rate explicitly defined like in vendor kernel: https://github.com/MichaIng/build/commit/59f048
 If this does not work either, I'll switch the PHY node to `phy@0` and move the resets back into the back, like in vendor kernel.

To test, as always:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-current-rockchip64.deb
dpkg -i linux-{dtb,image}-current-rockchip64.deb
reboot
```

Kwiboo, 2025-01-14:
Use mainline U-Boot on any board with rtl8211f Ethernet PHYs for mainline Linux, also make sure you use a build for your board or you will have Ethernet issue in Linux, i.e use an ROCK 3A image for 3B and you will only have one Ethernet port working, or use vendor U-Boot on Orange Pi 3B and you may have Ethernet PHY issue in Linux.

Please see https://patchwork.kernel.org/project/linux-rockchip/patch/20241220163227.1501912-1-alicja.michalska@9elements.com/#26173827 for some details on the rtl8211f phy issue.

> Currently I know of these workarounds:
> a) reset PHY before OS, i.e. use mainline U-Boot
> b) use a ethernet-phy-id001c.c916 compatible for the onboard RTL8211F
>   PHY instead of the generic ethernet-phy-ieee802.3-c22 compatible
> c) use deprecated snps,reset- props
> d) use reset props on mdiobus level

Kwiboo, 2025-01-14:
> So no idea what changed between the 6.12.7 build and 6.12.9 now. Coincidence that it broke at Armbian or upstream?

v6.6.69 and v6.12.9 picked up a patch that broke some PCIe, SATA and USB on Rockchip RK356x, see https://lore.kernel.org/all/df8e0ff8-355b-4c18-9349-0f04ae16e01b@kwiboo.se/

MichaIng, 2025-01-14:
Hey @Kwiboo thanks for chiming in.

a) Mainline U-Boot is used already, so that alone does not seem to help. Though it is a U-Boot for ROCK 3A, so probably it does not reset the 2nd adapter. In that case, it seems we need to properly add ROCK 3B support to the Armbian build system. However, we have the very same issue on an Orange Pi 3B with a very different Ethernet adapter, and a matching U-Boot build. It broke just after switching from a patched in vendor device tree to new mainline device tree, where RX/TX delays and resets have been moved to the PHY/mdio node.

b) Sounds like another good attempt, which however would not work on the Orange Pi 3B.

c) Switching to "deprecated" snps,reset is what I wanted to test as well, if other things fail.

d) This is how mainline Linux does it already: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3568-rock-3b.dts#n525
But the deassert delay is shorter. Hence aligning these reset delays with the vendor device tree, still using this "modern" method, is what I wanted to try next.

MidG971, 2025-01-14:
> Hey @Kwiboo thanks for chiming in.
> 
> a) Mainline U-Boot is used already, so that alone does not seem to help. Though it is a U-Boot for ROCK 3A, so probably it does not reset the 2nd adapter. In that case, it seems we need to properly add ROCK 3B support to the Armbian build system. However, we have the very same issue on an Orange Pi 3B with a very different Ethernet adapter, and a matching U-Boot build. It broke just after switching from a patched in vendor device tree to new mainline device tree, where RX/TX delays and resets have been moved to the PHY/mdio node.
> 
> b) Sounds like another good attempt, which however would not work on the Orange Pi 3B.
> 
> c) Switching to "deprecated" snps,reset is what I wanted to test as well, if other things fail.
> 
> d) This is how mainline Linux does it already: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3568-rock-3b.dts#n525
> But the deassert delay is shorter. Hence aligning these reset delays with the vendor device tree, still using this "modern" method, is what I wanted to try next.
> 
> > v6.6.69 and v6.12.9 picked up a patch that broke some PCIe, SATA and USB on Rockchip RK356x
> 
> Maybe this broke onboard WiFi as well then.
> 
> EDIT: I read through the comments of the patch request, and it is indeed very educating:
> 
> > Not being able to find the PHY has nothing to do with RGMII delays.
> If the RGMII delays are wrong, packets will not be
> received/transmitted, or they will be corrupt and thrown away with
> checksum errors. But the PHY will be found and you can attach to it.
> 
> So we know that the RX/TX and most likely `rgmii-id` are not the problem. Looks like only the resets are. On the Orange Pi 3B, mainline U-Boot 2024.10 is used, with the specific device tree, which contains the code to reset the PHY on the v2.1 revision: https://github.com/u-boot/u-boot/blob/master/board/xunlong/orangepi-3b-rk3566/orangepi-3b-rk3566.c
> But it still has the issue. So it seems like, the assumption that U-Boot alone must to the reset and then all is fine, seems to be wrong for this board at least, maybe also for the ROCK 3B. Apart of this, it would be good to not rely on a particular U-Boot, even if it works.
> 
> @MidG971 
> Do you have time and mood to still test the current build, even that it most likely won't fix the issue, or should I apply the raised reset delays already?
> 
> @Kwiboo 
> Just in case you understand it better: I missed the `clock_in_out` input/output thing. In Orange Pi 3B vendor device tree, it is "input" with "rgmii", RX/TX delays and reset defined in the MAC node the legacy way. In the ROCK 3B vendor kernel, it is "output" with those. I understood from the comments in the patchwork, that vendor kernel devs often do not understand themselves, how things work, resulting in such weird/contradicting stuff, and I know it from other parts of their images, where I have know knowledge. But did I get it right, that, when assigning `assigned-clock-rates` in the GMAC, `clock_in_out` needs to be output? At least the Orange Pi 3B device tree then would be prove, that the clock is not the issue, and really only the resets are to have a look at.

Always in the mood to help , do not hesitate 😎

Kwiboo, 2025-01-14:
> Though it is a U-Boot for ROCK 3A, so probably it does not reset the 2nd adapter.

Correct, because the device tree used by U-Boot in such case does not have information about the 2nd adapter, the 2nd phy is never reset and Linux will not be able to find it, unless one of the mentioned workarounds is applied.

> However, we have the very same issue on an Orange Pi 3B with a very different Ethernet adapter, and a matching U-Boot build.

This issue is probably because part of the vendor U-Boot still exists in SPI flash or eMMC and is being used to load mainline U-Boot proper, when the vendor U-Boot SPL part is used to load mainline U-Boot proper, it will load the default FIT configuration that does not have any Ethernet PHY nodes enabled, when mainline U-Boot SPL is used the full revision detection code should work and `fdtfile` and correct `Model:` line will be printed, can be observed with debug UART, `Model:` should end with a version, i.e. `v1.1` or `v2.1`, when no version is printed, it may cause issues for Linux.

> b) Sounds like another good attempt, which however would not work on the Orange Pi 3B.

As mentioned above, as long as mainline U-Boot SPL was used to load mainline U-Boot proper it should work. Output from serial when booting affected boards would be helpful to see if there is any other issue on boards in the wild.

> c) Switching to "deprecated" snps,reset is what I wanted to test as well, if other things fail.

Using the `snps,reset` or just changing the phy compatible from `ethernet-phy-ieee802.3-c22` to `ethernet-phy-id001c.c916` should work on a board with `RTL8211F` PHY board.

> But the deassert delay is shorter. Hence aligning these reset delays with the vendor device tree, still using this "modern" method, is what I wanted to try next.

I mainlined both the Orange pi 3B and ROCK 3B to Linux and U-Boot. Have both op3b (v1.1.1 and v2.1) and rock-3b (commercial and industrial) boards that I tested booting into Linux on. Ethernet was working when I tested everything during mainlining process using pure mainline U-Boot and mainline linux-next at the time.
The datasheet for the RTL8211F Ethernet PHY mention minimum held low for 10ms and wait 30-76 (depending on datasheet revision) ms until reading registers after reset, so using 15-20 ms with 50-80+ ms should be enough to reset the PHY.

> Maybe this broke onboard WiFi as well then.

Most likely, error messages should be seen in `dmesg` and `cat /sys/kernel/debug/devices_deferred` on affected boards.

> But it still has the issue. So it seems like, the assumption that U-Boot alone must to the reset and then all is fine, seems to be wrong for this board at least, maybe also for the ROCK 3B. Apart of this, it would be good to not rely on a particular U-Boot, even if it works.

For the ROCK 3B, changing the phy compatible from `ethernet-phy-ieee802.3-c22` to `ethernet-phy-id001c.c916` should probably work and be okay for mainline as we know the boards have the RTL8211F PHY.

> Just in case you understand it better: I missed the `clock_in_out` input/output thing. In Orange Pi 3B vendor device tree, it is "input" with "rgmii", RX/TX delays and reset defined in the MAC node [...]

To correctly describe the hardware the `clock_in_out` should probably be changed, I did not pay too much attention when creating the mainlined device trees. In practice it does not have any impact, the clock rate is default and input/output does not affect too much in gmac driver. I run basic iperf3 test using the mainlined DT on each ethernet port and did not see any issue on my board revisions.

MidG971, 2025-01-14:
> ```shell
> ip l
> ```

Still the same, wifi broken

MidG971, 2025-01-14:
here's some info I found, it might be : [useful](https://gitlab.com/firefly-linux/docs/-/tree/rk356x/firefly/Common/GMAC)

MichaIng, 2025-01-15:
@Kwiboo 

> This issue is probably because part of the vendor U-Boot still exists in SPI flash or eMMC and is being used to load mainline U-Boot proper, when the vendor U-Boot SPL part is used to load mainline U-Boot proper

Our images are full drive images, which includes the bootloader bits. And it is booting from eMMC, hence should not use the SPI bootloader. But I asked the user to flash the SPI bootloader, just to be sure. Downside is, that I cannot test the resets from kernel anymore then, which I want to have in place as well, to not depend booting from MMC or mainline SPI bootloader.

> Using the deprecated `snps,reset`

Does it need to be the deprecated method in the GMAC? Does it work differently when using the same reset delays in the PHY/mdio node?

> The datasheet for the RTL8211F Ethernet PHY mention minimum held low for 10ms and wait 30-76 (depending on datasheet revision) ms until reading registers after reset, so using 15-20 ms with 50-80+ ms should be enough to reset the PHY.

In mainline Linux, the dessert delay is 50ms. Maybe too short in at least some individual cases?
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3566-orangepi-3b-v1.1.dts?h=linux-6.12.y

> Most likely, error messages should be seen in dmesg and cat /sys/kernel/debug/devices_deferred on affected boards.

It is above, but I do not even see an attempt for the driver to be loader (`iwlwifi`): https://github.com/MichaIng/DietPi/issues/7327#issuecomment-2586913442

@MidG971 
Can you filter it by errors:
```sh
dmesg -l 0,1,2,3
```
and `modprobe iwlwifi` does not return any error, it is shown in `lsmod` then, but `ip l` does not show the adapter at all? It is weird, because all that changed was mainline Linux 6.12.7 to 6.12.9.

> For the ROCK 3B, changing the phy compatible from ethernet-phy-ieee802.3-c22 to ethernet-phy-id001c.c916 should probably work and be okay for mainline as we know the boards have the RTL8211F PHY.

Oh you are right
```
eth1: PHY [stmmac-0:00] driver [RTL8211F Gigabit Ethernet] (irq=POLL)
```
I was not aware of that, as a different WiFi driver is used. Then it makes even more sense that this two boards suffer from the same issue. And indeed switching to `ethernet-phy-id001c.c916` is then a better next attempt. Else raising "modern" reset delays, else reverting to legacy snps method.

> here's some info I found, it might be [useful](https://gitlab.com/firefly-linux/docs/-/tree/rk356x/firefly/Common/GMAC)

Interesting indeed. I would usually trust the values found in mainline or even vendor kernels more than test results from a particular board (which I do not even own in this case), but at least we know now how they are/can be derived 🙂.

MichaIng, 2025-01-15:
Build with the compatible PHY adjusted to `ethernet-phy-id001c.c916` is running: https://github.com/MichaIng/DietPi/actions/runs/12779430484

Once done, it can be found again in the testing dir.

MidG971, 2025-01-15:
> 
> @MidG971 Can you filter it by errors:
> 
> ```shell
> dmesg -l 0,1,2,3
> ```
> 
> and `modprobe iwlwifi` does not return any error, it is shown in `lsmod` then, but `ip l` does not show the adapter at all? It is weird, because all that changed was mainline Linux 6.12.7 to 6.12.9.
> 
That's before your latest build : 

```
root@DietPi:~# dmesg -l 0,1,2,3
[    1.007176] rockchip-naneng-combphy fe830000.phy: error -ENOENT: failed to get phy reset
[    1.007239] rockchip-naneng-combphy fe830000.phy: probe with driver rockchip-naneng-combphy failed with error -2
[    1.007762] rockchip-naneng-combphy fe840000.phy: error -ENOENT: failed to get phy reset
[    1.007824] rockchip-naneng-combphy fe840000.phy: probe with driver rockchip-naneng-combphy failed with error -2
[    1.008627] rockchip-naneng-combphy fe820000.phy: error -ENOENT: failed to get phy reset
[    1.008678] rockchip-naneng-combphy fe820000.phy: probe with driver rockchip-naneng-combphy failed with error -2
[    2.164292] rtc-hym8563 5-0051: hctosys: unable to read the hardware clock
[    3.922403] rk_gmac-dwmac fe2a0000.ethernet: Can not read property: tx_delay.
[    3.923073] rk_gmac-dwmac fe2a0000.ethernet: set tx_delay to 0x30
[    3.923731] rk_gmac-dwmac fe2a0000.ethernet: Can not read property: rx_delay.
[    3.923761] rk_gmac-dwmac fe2a0000.ethernet: set rx_delay to 0x10
[    3.945745] mdio_bus stmmac-0: MDIO device at address 1 is missing.
[    9.512191] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[   50.534341] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)

root@DietPi:~# modprobe iwlwifi
root@DietPi:~# lsmod
Module                  Size  Used by
iwlwifi               290816  0
cfg80211              815104  1 iwlwifi
rfkill                 24576  2 cfg80211
snd_soc_hdmi_codec     20480  1
snd_soc_simple_card    20480  0
crct10dif_ce           12288  1
display_connector      16384  0
snd_soc_simple_card_utils    24576  1 snd_soc_simple_card
gpio_ir_recv           12288  0
rc_core                45056  2 gpio_ir_recv
snd_soc_rk817          40960  1
snd_soc_rockchip_i2s_tdm    20480  4
snd_soc_core          204800  5 snd_soc_rockchip_i2s_tdm,snd_soc_hdmi_codec,snd_soc_rk817,snd_soc_simple_card_utils,snd_soc_simple_card
rk_crypto2             36864  0
snd_compress           24576  1 snd_soc_core
rockchip_vdec2         28672  0
sm3_generic            12288  1 rk_crypto2
rockchip_rng           12288  0
snd_pcm_dmaengine      12288  1 snd_soc_core
sm3                    20480  1 sm3_generic
hantro_vpu            249856  0
snd_pcm               106496  5 snd_soc_hdmi_codec,snd_compress,snd_soc_simple_card_utils,snd_soc_core,snd_pcm_dmaengine
v4l2_vp9               20480  1 hantro_vpu
v4l2_jpeg              12288  1 hantro_vpu
rng_core               16384  1 rockchip_rng
v4l2_h264              16384  2 hantro_vpu,rockchip_vdec2
snd_timer              36864  1 snd_pcm
snd                    77824  5 snd_soc_hdmi_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
rockchip_rga           20480  0
rk817_charger          20480  0
videobuf2_dma_contig    20480  2 hantro_vpu,rockchip_vdec2
dw_hdmi_i2s_audio      12288  0
videobuf2_dma_sg       16384  1 rockchip_rga
dw_hdmi_cec            12288  0
soundcore              12288  1 snd
v4l2_mem2mem           24576  3 hantro_vpu,rockchip_rga,rockchip_vdec2
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_dma_sg
videobuf2_v4l2         20480  4 hantro_vpu,rockchip_rga,rockchip_vdec2,v4l2_mem2mem
panfrost               73728  5
videodev              237568  5 videobuf2_v4l2,hantro_vpu,rockchip_rga,rockchip_vdec2,v4l2_mem2mem
gpu_sched              36864  1 panfrost
rk805_pwrkey           12288  0
videobuf2_common       53248  8 videobuf2_dma_contig,videobuf2_v4l2,hantro_vpu,rockchip_rga,videobuf2_dma_sg,rockchip_vdec2,v4l2_mem2mem,videobuf2_memops
mc                     53248  6 videodev,videobuf2_v4l2,hantro_vpu,videobuf2_common,rockchip_vdec2,v4l2_mem2mem
drm_shmem_helper       20480  1 panfrost
cpufreq_dt             16384  0
binfmt_misc            20480  1
dm_mod                126976  0
ip_tables              28672  0
x_tables               36864  1 ip_tables
autofs4                45056  2
realtek                36864  1
phy_rockchip_snps_pcie3    16384  1
dwmac_rk               28672  0
rockchipdrm           176128  7
spi_rockchip_sfc       16384  0
drm_dma_helper         16384  1 rockchipdrm
stmmac_platform        20480  1 dwmac_rk
dw_hdmi_qp             20480  1 rockchipdrm
stmmac                241664  3 stmmac_platform,dwmac_rk
dw_hdmi                45056  2 dw_hdmi_i2s_audio,rockchipdrm
pcs_xpcs               24576  1 stmmac
cec                    45056  2 dw_hdmi_cec,dw_hdmi
dw_mipi_dsi            16384  1 rockchipdrm
rtc_rk808              16384  0
analogix_dp            40960  1 rockchipdrm
drm_display_helper    131072  4 rockchipdrm,dw_hdmi,analogix_dp,dw_hdmi_qp
drm_kms_helper        122880  9 dw_mipi_dsi,drm_dma_helper,display_connector,drm_shmem_helper,drm_display_helper,rockchipdrm,dw_hdmi,analogix_dp,dw_hdmi_qp
drm                   442368  21 gpu_sched,drm_kms_helper,dw_mipi_dsi,drm_dma_helper,display_connector,drm_shmem_helper,drm_display_helper,rockchipdrm,dw_hdmi,panfrost,analogix_dp,dw_hdmi_qp
drm_panel_orientation_quirks    28672  1 drm
root@DietPi:~# ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether e2:e1:76:03:12:f8 brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether e2:e1:76:03:12:f9 brd ff:ff:ff:ff:ff:ff
root@DietPi:~#

```

MichaIng, 2025-01-15:
Update: On the Orange Pi 3B, indeed the SPI bootloader was used, even that the eMMC contained one. Erasing the SPI flash fixed Ethernet there. However, if the changed compatible PHY fixes it on ROCK 3B even without proper mainline U-Boot, I'll keep that change as well.

@MidG971 
Now that the WiFi driver was loaded, does `dmesg -l 0,1,2,3` show any new errors?

MidG971, 2025-01-15:
> Update: On the Orange Pi 3B, indeed the SPI bootloader was used, even that the eMMC contained one. Erasing the SPI flash fixed Ethernet there. However, if the changed compatible PHY fixes it on ROCK [...]

`
root@DietPi:~# dmesg -l 0,1,2,3
[    1.007176] rockchip-naneng-combphy fe830000.phy: error -ENOENT: failed to get phy reset
[    1.007239] rockchip-naneng-combphy fe830000.phy: probe with driver rockchip-naneng-combphy failed with error -2
[    1.007762] rockchip-naneng-combphy fe840000.phy: error -ENOENT: failed to get phy reset
[    1.007824] rockchip-naneng-combphy fe840000.phy: probe with driver rockchip-naneng-combphy failed with error -2
[    1.008627] rockchip-naneng-combphy fe820000.phy: error -ENOENT: failed to get phy reset
[    1.008678] rockchip-naneng-combphy fe820000.phy: probe with driver rockchip-naneng-combphy failed with error -2
[    2.164292] rtc-hym8563 5-0051: hctosys: unable to read the hardware clock
[    3.922403] rk_gmac-dwmac fe2a0000.ethernet: Can not read property: tx_delay.
[    3.923073] rk_gmac-dwmac fe2a0000.ethernet: set tx_delay to 0x30
[    3.923731] rk_gmac-dwmac fe2a0000.ethernet: Can not read property: rx_delay.
[    3.923761] rk_gmac-dwmac fe2a0000.ethernet: set rx_delay to 0x10
[    3.945745] mdio_bus stmmac-0: MDIO device at address 1 is missing.
[    9.512191] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)
[   50.534341] rk_gmac-dwmac fe2a0000.ethernet eth1: __stmmac_open: Cannot attach to PHY (error: -19)

`

in a minute i'll test the build + plug my serial cable

MichaIng, 2025-01-15:
Weird. As if it was the wrong driver. But it was definitely used before: https://github.com/MichaIng/DietPi/issues/7327#issuecomment-2565712090

Are there other messages from this driver?
```sh
dmesg | grep iwlwifi
```

> in a minute i'll test the build + plug my serial cable

Wait before https://github.com/MichaIng/DietPi/actions/runs/12779430484 is done, to test with the changed compatible PHY directly.

MichaIng, 2025-01-15:
Okay, build is done now.

[9 more replies]