MichaIng/DietPi #7329: Home Assistant install fails on maturin

jankkhvej, 2024-12-27:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.9.0`
- Distro version | `bookworm 1`
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- SBC model | `RPi B (armv6l)`
- Power supply used | IKEA 5v 3 × USB-A charger SMÅHAGEL 705.440.79
- SD card used | Kingston 128 Gb

#### Additional Information (if applicable)
- Software title | Home Assitant
- Fresh install
- Yes, bug is repeatable on fresh install of DietPi
- Bug report ID | `9937c3ff-27ae-402c-bbe2-d28fb843a5c4`

#### Steps to reproduce

1. Install fresh DietPi without any packages
2. Upgrade everything
3. reboot
4. Attempt to install Home Assistant via dietpi-launcher -> DietPi-Software

#### Expected behaviour
- Home Assistant installed

#### Actual behaviour
- Got an error on install:
```
[...]
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error

  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [5 lines of output]
      💥 maturin failed
        Caused by: `project.version` field is required in pyproject.toml unless it is present in the `project.dynamic` list
      Error running maturin: Command '['maturin', 'pep517', 'write-dist-info', '--metadata-directory', '/tmp/pip-modern-metadata-r1yj3ba7', '--interpreter', '/home/homeassistant/.pyenv/versions/3.13.1/bin/python3.13']' returned non-zero exit status 1.
      Checking for Rust toolchain....
      Running `maturin pep517 write-dist-info --metadata-directory /tmp/pip-modern-metadata-r1yj3ba7 --interpreter /home/homeassistant/.pyenv/versions/3.13.1/bin/python3.13`
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.

error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
[FAILED] DietPi-Software | Installing Python with Home Assistant module into pyenv
 - Command: runuser -u homeassistant -- dash -c 
. /home/homeassistant/pyenv-activate.sh
pyenv install 3.13.1
pyenv local 3.13.1
pip3 install -U pip setuptools wheel
[ -z '' ] || pip3 install 
exec pip3 install homeassistant
[ INFO ] DietPi-BugReport | Generating informative command outputs, please wait...
[ INFO ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT, please wait...
[  OK  ] DietPi-BugReport | cp /tmp/G_EXEC_ERROR_REPORT G_EXEC_ERROR_REPORT
[  OK  ] DietPi-BugReport | Packing upload archive
[  OK  ] DietPi-BugReport | Sending bug report
[  OK  ] DietPi-BugReport | Your bug report has been successfully uploaded.
- Reference code: 9937c3ff-27ae-402c-bbe2-d28fb843a5c4
```

MichaIng, 2024-12-28:
Can you please share the full log of the HA install attempt? I do not see which package build actually failed.

Might be an ARMv6-related issue, but as I am traveling, I have no access to my RPi Zero W right now.

jankkhvej, 2024-12-28:
#### Details:
- Date           | 2024-12-28 14:28:25
- Program name   | DietPi-Software
- Command        | `runuser -u homeassistant -- dash -c 
. /home/homeassistant/pyenv-activate.sh
pyenv install 3.13.1
pyenv local 3.13.1
pip3 install -U pip setuptools wheel
[ -z '' ] || pip3 install 
exec pip3 install homeassistant`
- Exit code      | 1
- DietPi version | v9.9.0 (MichaIng/master)
- Distro version | bookworm (ID=7,RASPBIAN=1)
- Kernel version | `Linux DietPi 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`
- Architecture	 | `armhf`
- Hardware model | RPi B (armv6l) (ID=1)

#### Additional logs:
```
Downloading Python-3.13.1.tar.gz...
-> https://www.python.org/ftp/python/3.13.1/Python-3.13.1.tgz
Installing Python-3.13.1...
Installed Python-3.13.1 to /home/homeassistant/.pyenv/versions/3.13.1
Requirement already satisfied: pip in ./.pyenv/versions/3.13.1/lib/python3.13/site-packages (24.3.1)
Collecting setuptools
  Downloading setuptools-75.6.0-py3-none-any.whl.metadata (6.7 kB)
Collecting wheel
  Downloading wheel-0.45.1-py3-none-any.whl.metadata (2.3 kB)
Downloading setuptools-75.6.0-py3-none-any.whl (1.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.2/1.2 MB 2.1 MB/s eta 0:00:00
Downloading wheel-0.45.1-py3-none-any.whl (72 kB)
Installing collected packages: wheel, setuptools
Successfully installed setuptools-75.6.0 wheel-0.45.1
Collecting homeassistant
  Downloading homeassistant-2024.12.5-py3-none-any.whl.metadata (4.8 kB)
Collecting aiodns==3.2.0 (from homeassistant)
  Downloading aiodns-3.2.0-py3-none-any.whl.metadata (4.0 kB)
Collecting aiohasupervisor==0.2.1 (from homeassistant)
  Downloading aiohasupervisor-0.2.1-py3-none-any.whl.metadata (5.1 kB)
Collecting aiohttp==3.11.11 (from homeassistant)
  Downloading aiohttp-3.11.11.tar.gz (7.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 7.7/7.7 MB 4.1 MB/s eta 0:00:00
  Installing build dependencies: started
  Installing build dependencies: still running...
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting aiohttp-cors==0.7.0 (from homeassistant)
  Downloading aiohttp_cors-0.7.0-py3-none-any.whl.metadata (20 kB)
Collecting aiohttp-fast-zlib==0.2.0 (from homeassistant)
  Downloading aiohttp_fast_zlib-0.2.0-py3-none-any.whl.metadata (5.6 kB)
Collecting aiozoneinfo==0.2.1 (from homeassistant)
  Downloading aiozoneinfo-0.2.1-py3-none-any.whl.metadata (5.5 kB)
Collecting astral==2.2 (from homeassistant)
  Downloading astral-2.2-py2.py3-none-any.whl.metadata (1.5 kB)
Collecting async-interrupt==1.2.0 (from homeassistant)
  Downloading async_interrupt-1.2.0-py3-none-any.whl.metadata (4.9 kB)
Collecting attrs==24.2.0 (from homeassistant)
  Downloading attrs-24.2.0-py3-none-any.whl.metadata (11 kB)
Collecting atomicwrites-homeassistant==1.4.1 (from homeassistant)
  Downloading atomicwrites_homeassistant-1.4.1-py2.py3-none-any.whl.metadata (6.0 kB)
Collecting awesomeversion==24.6.0 (from homeassistant)
  Downloading awesomeversion-24.6.0-py3-none-any.whl.metadata (9.0 kB)
Collecting bcrypt==4.2.0 (from homeassistant)
  Downloading bcrypt-4.2.0.tar.gz (24 kB)
  Installing build dependencies: started
  Installing build dependencies: still running...
  Installing build dependencies: finished with status 'done'
[... 78 lines cut ...]
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'error'
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [5 lines of output]
      💥 maturin failed
        Caused by: `project.version` field is required in pyproject.toml unless it is present in the `project.dynamic` list
      Error running maturin: Command '['maturin', 'pep517', 'write-dist-info', '--metadata-directory', '/tmp/pip-modern-metadata-htk6cb0j', '--interpreter', '/home/homeassistant/.pyenv/versions/3.13.1/bin/python3.13']' returned non-zero exit status 1.
      Checking for Rust toolchain....
      Running `maturin pep517 write-dist-info --metadata-directory /tmp/pip-modern-metadata-htk6cb0j --interpreter /home/homeassistant/.pyenv/versions/3.13.1/bin/python3.13`
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

frozen905, 2024-12-29:
I'm here for the same issue, project.version field error trying to build maturin

Joulinar, 2024-12-29:
This is an issue with the python module `orjson` and outside our control. It seems you are not alone https://github.com/ijl/orjson/issues/541

MichaIng, 2024-12-30:
Seems to be solved upstream. Can you all retry?

steals, 2024-12-30:
Upgrading to the lates version will not help, as HA strictly specifies 3.10.12 version https://github.com/home-assistant/core/blob/2024.12.5/homeassistant/package_constraints.txt
but this issue could be solved by manually downloading sources `https://github.com/ijl/orjson/archive/refs/tags/3.10.12.tar.gz` manually changing the `project.toml` fie and adding `version = "3.10.12"` and building it locally by `pip install .`

MichaIng, 2024-12-30:
Hmm, much easier and less intrusive to adjust `package_constraints.txt` instead. v3.10.13 is fully compatible, only ships with this one little change.

I recognised that the issue affects ARMv6 and RISC-V only, as for all other architectures, pre-compiled binaries are shipped by PyPI: https://pypi.org/project/orjson/3.10.12/#files

jankkhvej, 2024-12-30:
> Seems to be solved upstream. Can you all retry?
> 
> `maturin` introduced a breaking change here: https://github.com/PyO3/maturin/pull/2391/files Latest `orjson` explicitly declares the latest functional version before this breaking change as dependency. No long-term solution, I hope it does not imply a dependency version conflict at some point.

Simply retrying, without any changes, results in the same error.
Will try with manual patch and pip install.

MichaIng, 2024-12-30:
Yeah, as steals mentions, the orjson version dependency in HA is hardcoded. I am currently trying another way, defining the `maturing` version with an own `constraints.txt`: #7335

Testing it for RISC-V: https://github.com/MichaIng/DietPi/actions/runs/12546201627

MichaIng, 2024-12-31:
Actually also this change in latest maturin from yesterday should have solved it: https://github.com/PyO3/maturin/pull/2417/files
But pinning a version that is assured to be compatible with orjson is still not a bad idea.

jankkhvej, 2024-12-31:
> It seems to work. Can you try on ARMv6?
> 
> ```shell
> G_DEV_BRANCH homeassistant
> dietpi-software install 157
> ```

I didn't know about G_DEV_BRANCH, and applied a patch manually. Home Assistant installed successfully, but haven't time to test it yet – RPi Model B is SLOW 🤷🏻‍♂️.

Will retry now according to your recommendations.

MichaIng, 2024-12-31:
If you applied the patch manually, then it is all good. No need to redo the lengthy build. But can you check the installed `maturin` version? The logs hide it, as it is a dependency of a dependency only:
```sh
sudo -u homeassistant bash
. /home/homeassistant/pyenv-activate.sh
pip3 freeze | grep maturin
```

jankkhvej, 2025-01-01:
> If you applied the patch manually, then it is all good. No need to redo the lengthy build. But can you check the installed `maturin` version? The logs hide it, as it is a dependency of a dependency only:

After successful installation of HA maturin does not exist in the system:

```homeassistant@DietPi:~$ . /home/homeassistant/pyenv-activate.sh
homeassistant@DietPi:~$ python --version
Python 3.13.1
homeassistant@DietPi:~$ pip3 freeze | grep maturin
homeassistant@DietPi:~$ pip3 freeze | grep json
orjson==3.10.12
homeassistant@DietPi:~$ 
```

MichaIng, 2025-01-01:
Ah, maybe those build dependencies do not stay in the environment.

closed 2025-02-09 by MichaIng