MichaIng/DietPi #7330: MD5 checksum not matching

amilopowers, 2024-12-28:
Hello
I downloaded the image for the Raspberry Pi 3: https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bookworm.img.xz
MD5 checksum should be: https://dietpi.com/downloads/images/DietPi_RPi-ARMv8-Bookworm.img.xz.sha256

What I get on my mac and in an online tool: 3d608286995f5b3e1e8055c2865630a5

Joulinar, 2024-12-28:
Hmm, I downloaded the image as a test.

```
PS C:\Users\Joulinar\Downloads> ls DietPi_RPi-ARMv8-Bookworm.img.xz

    Verzeichnis: C:\Users\Joulinar\Downloads

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a----        28.12.2024     11:23      147181400 DietPi_RPi-ARMv8-Bookworm.img.xz

PS C:\Users\Joulinar\Downloads>
```

and the hash is fine for me

```
PS C:\Users\Joulinar\Downloads> Get-FileHash -Algorithm SHA256 .\DietPi_RPi-ARMv8-Bookworm.img.xz

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
SHA256          B959C8314CFB5F568FBFC98590EC43D149EA6CCD568FAB50A7257AEADAE421EE       C:\Users\Joulinar\Downloads\DietPi_RPi-ARMv8-Bookworm.img.xz

PS C:\Users\Joulinar\Downloads>
```

amilopowers, 2024-12-28:
I am sorry. My bad I simply typed "MD5 <path to file>" into the terminal of my mac. The correct command is apparently: shasum -a 256.

amilopowers, 2024-12-28:
User's error.

closed 2024-12-28 by amilopowers