MichaIng/DietPi #7331: DietPi Imager: Prompt for Number of Cores to Allocate for `xz`

mal1k-me, 2024-12-28:
Currently, `xz` compression in DietPi Imager uses a single CPU core by default. It would be helpful to add an option prompting the user to choose the number of CPU cores to allocate for the `xz` compression process (`-T` flag), allowing users to speed up image creation by utilizing more resources.

Joulinar, 2024-12-28:
this might be related/similar to your question https://github.com/MichaIng/DietPi/issues/7234

MichaIng, 2024-12-30:
Yes, LZMA2 compression takes quite some memory, and it automatically limits the amount of used cores to avoid running into OOM errors. We did already lift the default limit of 25% physical RAM size to 75% physical RAM size recently (see linked issue above), more would risk to cause issues, or leads to memory swapping, which slows down the whole process massively.

Hence check the physical RAM size of your machine, and the image size that is to be compressed, and compare it with the limits we ran into in the liked issue.

closed 2025-01-06 by mal1k-me