MichaIng/DietPi #7338: OrangePi 5 Max Wifi is not working with DietPi 9.9

diego-hsp, 2024-12-30:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.9
- Distro version | bookworm`
- Kernel version | `Linux orangepi 6.1.75-vendor-rk35xx #1 SMP`
- SBC model | `Orange Pi 5 Max (aarch64)
- Power supply used | 5V 3A 
- SD card used | SanDisk ultra 32Gb - Tested with H2testw

#### Actual behaviour

- I can´t find or use the onboard wifi with dietpi, but it is working with armbian and the official distro from orangepi
I downloaded the dietpi image again and reflashed the SD card, but it didn't change anything. I didn't find if there is a HASH test for the dietpi image.
I did not tested the bluetooth.

diego-hsp, 2024-12-30:
I found the hash files, and the values matched.
I also found images of versions bullseye and trixie for the orange pi 5 max, I don't know if it makes any difference with the wifi drivers, but I tested them without sucess...

MichaIng, 2024-12-31:
Thanks for reporting. The adapter does not show up at all?
```sh
ip l
```
Which Armbian image in particular did you test? Because kernel and firmware should be the same than ours.

MichaIng, 2024-12-31:
Found it, we need an additional driver for this WiFi chip: https://github.com/armbian/bcmdhd-dkms

MichaIng, 2024-12-31:
Actually the driver is there already: https://github.com/armbian/linux-rockchip/tree/rk-6.1-rkr4.1/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd
Only it is built without SDIO interface. I'll do a build with this enabled. If this works, we can avoid the DKMS hassle.

MichaIng, 2024-12-31:
Build running: https://github.com/MichaIng/DietPi/actions/runs/12558156119
With this change applied: https://github.com/MichaIng/build/commit/adedb02
I did not compare it 1-to-1 to the one from the [bcmdhd-dkms](https://github.com/armbian/bcmdhd-dkms) repo, but the few samples I checked were identical.

Packages with "vendor-rk35xx" can be found here, once done: https://dietpi.com/downloads/binaries/testing/

MichaIng, 2025-01-02:
Please test this:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{dtb,image}-vendor-rk35xx.deb
dpkg -i linux-{dtb,image}-vendor-rk35xx.deb
reboot
```
After reboot, if the WiFi adapter does still not show up, assure this kernel module is loaded:
```sh
modprobe bcmdhd
```
And in case check for kernel errors:
```sh
dmesg -l 0,1,2,3
dmesg | grep bcmdhd
```

diego-hsp, 2025-01-04:
Hi MichaIng!
Happy new year! 
Sorry for the delay in answering you and thanks for getting back to me so quickly!

I use linux for some time, but I'm not really good with hardware, kernel, dkms... yet...
I tried to answer and follow your commands, reportting below... If something is not undestandable, please point it to me, english is not my first language (Greetings from Brazil)!

> Thanks for reporting. The adapter does not show up at all?

When I ran "ip l" I did not find the wifi, and this was the result:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether c0:74:2b:fd:ce:c7 brd ff:ff:ff:ff:ff:ff
3: br-6bd6f99774a5: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN mode DEFAULT group default 
    link/ether 02:42:15:32:d6:10 brd ff:ff:ff:ff:ff:ff
4: br-9cfcf63f8113: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default 
    link/ether 02:42:09:e4:74:0c brd ff:ff:ff:ff:ff:ff
5: docker0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default 
    link/ether 02:42:23:a3:88:a2 brd ff:ff:ff:ff:ff:ff
7: vethb18ccc0@if6: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-9cfcf63f8113 state UP mode DEFAULT group default 
    link/ether 72:c2:a9:33:08:f4 brd ff:ff:ff:ff:ff:ff link-netnsid 1
9: veth5c75dd3@if8: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master docker0 state UP mode DEFAULT group default 
    link/ether 7a:96:0e:4c:0d:b3 brd ff:ff:ff:ff:ff:ff link-netnsid 0
10: vethac348d7@veth1ec5e20: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default 
    link/ether 9a:36:2a:bd:6b:13 brd ff:ff:ff:ff:ff:ff
11: veth1ec5e20@vethac348d7: <NO-CARRIER,BROADCAST,MULTICAST,UP,M-DOWN> mtu 1500 qdisc noqueue master br-6bd6f99774a5 state LOWERLAYERDOWN mode DEFAULT group default 
    link/ether 2a:a7:fd:10:8b:e5 brd ff:ff:ff:ff:ff:ff

``` 

> Which Armbian image in particular did you test? Because kernel and firmware should be the same than ours.

I tested Armbian with the image "Armbian_community_25.2.0-trunk.195_Orangepi5-max_noble_vendor_6.1.84_gnome_desktop.img"

>To be sure, this driver is really not part of the kernel yet, right?

modinfo bcmdhd returned this (import_ns was repetteaded a LOT, I deleted the most...) 
```
filename:       /lib/modules/6.1.75-vendor-rk35xx/kernel/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/bcmdhd.ko
import_ns:      VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver
license:        GPL and additional rights
import_ns:      VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver
import_ns:      VFS_internal_I_am_really_a_filesystem_and_am_NOT_a_driver
srcversion:     73EA1DE2D2719B239531B94
alias:          pci:v00001E01d*sv*sd*bc02sc80i*
alias:          pci:v000014E4d*sv*sd*bc02sc80i*
depends:        
intree:         Y
name:           bcmdhd
vermagic:       6.1.75-vendor-rk35xx SMP mod_unload modversions aarch64
parm:           fw_ap_select:uint
parm:           wl_reassoc_support:uint
parm:           mscs_offload:uint
parm:           dhd_rxf_prio:int
parm:           g_assert_type:int
parm:           clm_path:string
parm:           info_string:string
parm:           op_mode:int
parm:           dhd_msg_level:int
parm:           iw_msg_level:int
parm:           wl_dbg_level:int
parm:           android_msg_level:int
parm:           config_msg_level:int
parm:           dhd_arp_mode:uint
parm:           disable_proptx:int
parm:           firmware_path:string
parm:           nvram_path:string
parm:           config_path:string
[... 179 lines cut ...]
[   15.506280] rk-pcie fe170000.pcie: PCIe Link Fail, LTSSM is 0x3, hw_retries=0
[   15.506310] rk-pcie fe190000.pcie: PCIe Link Fail, LTSSM is 0x3, hw_retries=0
[   15.515251] rk-pcie fe150000.pcie: PCIe Link Fail, LTSSM is 0x0, hw_retries=0
[   15.515286] rk-pcie fe150000.pcie: failed to initialize host
[   16.245638] Goodix-TS 7-0014: Error reading 1 bytes from 0x8140: -6
[   16.272281] Goodix-TS 7-0014: Error reading 1 bytes from 0x8140: -6
[   16.298704] Goodix-TS 7-0014: I2C communication failure: -6
[   18.821636] rk-pcie fe190000.pcie: PCIe Link Fail, LTSSM is 0x3, hw_retries=1
[   18.821835] rk-pcie fe170000.pcie: PCIe Link Fail, LTSSM is 0x3, hw_retries=1
[   19.848768] rk-pcie fe170000.pcie: failed to initialize host
[   19.848867] rk-pcie fe190000.pcie: failed to initialize host
[   39.395401] [GTP-ERR][__do_register_ext_module:79] Module [goodix-fwu] timeout
[ 1143.379350] [dhd] _dhd_module_init: in Dongle Host Driver, version 101.10.591.52.27 (20240409-1)(20240411-2)(d83d8d7)
               drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd compiled on Dec 24 2024 at 13:36:04
[ 1143.379452] [dhd] STATIC-MSG) dhd_static_buf_init : 101.10.361.36 (wlan=r892223-20231107-1)
[ 1143.379911] [dhd] STATIC-MSG) dhd_init_wlan_mem : prealloc ok for index 0: 2371584(2316K)
[ 1143.380268] [dhd] STATIC-MSG) dhd_static_buf_exit : Enter
``` 
and 'dmesg | grep bcmdhd' returned: 
```
root@orangepi:~# dmesg | grep bcmdhd
               drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd compiled on Dec 24 2024 at 13:36:04
``` 

After all this dietpi showed me there was a update in "armbian-firmware", that I installed, and the wifi still did not work (I don't know what I was expecting... )

MichaIng, 2025-01-04:
Thanks for testing. So something is still missing. The firmware package upgrade was unrelated, just general maintenance.

Can you check whether those exist:
```sh
ls -l /lib/firmware/ap6275p/fw_bcmdhd.bin /lib/firmware/ap6275p/nvram.txt /lib/firmware/ap6275p
```
However, I think the problem is that the driver cannot be compiled with both (or multiple of) USB, PCIe and SDIO bus support. amazingfate who implemented the DKMS extension, said "it is hard to both have sdio and pcie in vendor kernel". It is a choice menu, not multi-select, IIUC.

... and indeed, I checked the build logs, which say:
```
 .config:3174:warning: override: BCMDHD_PCIE changes choice state
```
So I guess, the `CONFIG_BCMDHD_PCIE=y` did override the `CONFIG_BCMDHD_SDIO=y` from the previous line. Okay, so either we implement an additional module via DKMS as well, or, what I prefer, we do a dedicated kernel build for above choice swapped for the Orange Pi 5 Max. Not great, but better than the hassle with DKMS, and we can easily provide different builds for different SBCs, thanks to our separated APT server components. But I need maybe more than a day to implement this into our kernel build workflow, so that the package has a different filename and is merged separately from the regular `vendor-rk3588` build.

Another approach would be to test mainline kernel. I wanted to build it for all RK3588 boards anyway, so it can be tested easily then.

diego-hsp, 2025-01-04:
Using the command ls 
```
ls -l /lib/firmware/ap6275p/fw_bcmdhd.bin /lib/firmware/ap6275p/nvram.txt /lib/firmware/ap6275p
```
returned this
```
root@orangepi:~# ls -l /lib/firmware/ap6275p/fw_bcmdhd.bin /lib/firmware/ap6275p/nvram.txt /lib/firmware/ap6275p
ls: cannot access '/lib/firmware/ap6275p/fw_bcmdhd.bin': No such file or directory
ls: cannot access '/lib/firmware/ap6275p/nvram.txt': No such file or directory
/lib/firmware/ap6275p:
total 1804
-rw-rw-r-- 1 root root  80602 Dec 29 03:20 BCM4362A2.hcd
-rw-rw-r-- 1 root root  30993 Dec 29 03:20 clm_bcm43752a2_pcie_ag.blob
-rw-rw-r-- 1 root root  20121 Dec 29 03:20 clm_syn43711a0.blob
-rw-rw-r-- 1 root root    240 Dec 29 03:20 config.txt
-rw-rw-r-- 1 root root 936074 Dec 29 03:20 fw_bcm43752a2_pcie_ag.bin
-rw-rw-r-- 1 root root 752242 Dec 29 03:20 fw_syn43711a0_sdio.bin
-rwxrwxr-x 1 root root   7458 Dec 29 03:20 nvram_AP6275P.txt
lrwxrwxrwx 1 root root     17 Dec 29 03:20 nvram_ap6275p.txt -> nvram_AP6275P.txt
-rw-rw-r-- 1 root root   6871 Dec 29 03:20 nvram_ap6611s.txt
```
In the orangepi 5 max page, there is  information that the board have the ap6611s chip...

MichaIng, 2025-01-04:
The chip has a lot of aliases and rebrands and the ap6611s seems compatible, also based on the `nvram_ap6611s.txt` in this dir. It is strange that the Armbian kernel config explicitly declares default firmware files which are not present in `armbian-firmware`, and we could work around it with some symlinks, but it is not the issue AFAIK: The build logs and other info indicate that the driver was indeed built without SDIO support, and a missing firmware file usually leads to a related kernel log, when the driver is loaded. So likely it chose another matching fallback automatically.

Joulinar, 2025-01-05:
For reference, some user within forum having similar issues https://dietpi.com/forum/t/orange-pi-5-max-test-image-feedback/22083/3?u=joulinar

MichaIng, 2025-01-06:
I just pushed a new kernel to our APT server. Can you give this one a try?
```sh
apt update
apt upgrade
```

diego-hsp, 2025-01-08:
> I just pushed a new kernel to our APT server. Can you give this one a try?

IT WORKED!!!
Just had to reboot after updated and It's there!!!
Really thank you!

One last question about the orange pi 5 max... 
The others  dietpi that I have running in the raspberrys 3b+ and zero show no errors with this command, but the orangepi shows a LOT!

```
[   11.432308] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@0': base 0x0000000000000000, size 0 MiB
[   11.432327] OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@0': base 0x0000000000000000, size 0 MiB
[   12.841758] fiq_debugger fiq_debugger.0: error -ENXIO: IRQ fiq not found
[   12.841773] fiq_debugger fiq_debugger.0: error -ENXIO: IRQ wakeup not found
[   12.841783] fiq_debugger_probe: could not install nmi irq handler
[   12.899152] debugfs: Directory 'vcc_mipicsi1' with parent 'regulator' already present!
[   12.899172] debugfs: Directory 'vcc_mipicsi1' with parent 'vcc_mipicsi1' already present!
[   12.969114] dw-pcie fe180000.pcie: invalid resource
[   12.969128] dw-pcie fe180000.pcie: Failed to initialize host
[   12.969207] dw-pcie fe190000.pcie: invalid resource
[   12.969215] dw-pcie fe190000.pcie: Failed to initialize host
[   12.969334] dw-pcie fe150000.pcie: invalid resource
[   12.969343] dw-pcie fe150000.pcie: Failed to initialize host
[   12.969425] dw-pcie fe170000.pcie: invalid resource
[   12.969433] dw-pcie fe170000.pcie: Failed to initialize host
[   13.034489] rockchip-vop2 fdd90000.vop: Failed to get leakage
[   13.034556] rockchip-vop2 fdd90000.vop: error -ENODEV: _opp_set_regulators: no regulator (vop) found
[   13.034588] rockchip-vop2 fdd90000.vop: failed to set opp config
[   13.034599] rockchip-vop2 fdd90000.vop: failed to init opp info
[   13.064443] rk806 spi2.0: no sleep-setting state
[   13.064459] rk806 spi2.0: no reset-setting pinctrl state
[   13.064469] rk806 spi2.0: no dvs-setting pinctrl state
[   13.123649] cpu cpu0: Failed to get specification_serial_number
[   13.123668] cpu cpu0: Failed to get specification_serial_number
[   13.123687] cpu cpu0: Failed to get leakage
[   13.127266] cpu cpu4: Failed to get specification_serial_number
[   13.127284] cpu cpu4: Failed to get specification_serial_number
[   13.127301] cpu cpu4: Failed to get leakage
[   13.140760] cpu cpu6: Failed to get specification_serial_number
[   13.140779] cpu cpu6: Failed to get specification_serial_number
[   13.140798] cpu cpu6: Failed to get leakage
[   13.188205] arm-scmi firmware:scmi: Failed. SCMI protocol 17 not active.
[   13.366659] mpp_rkvdec2 fdc38100.rkvdec-core: shared_niu_a is not found!
[   13.366690] rkvdec2_init:1198: No niu aclk reset resource define
[   13.366717] mpp_rkvdec2 fdc38100.rkvdec-core: shared_niu_h is not found!
[   13.366739] rkvdec2_init:1201: No niu hclk reset resource define
[   13.374635] mpp_rkvdec2 fdc48100.rkvdec-core: shared_niu_a is not found!
[   13.374663] rkvdec2_init:1198: No niu aclk reset resource define
[   13.374690] mpp_rkvdec2 fdc48100.rkvdec-core: shared_niu_h is not found!
[   13.374712] rkvdec2_init:1201: No niu hclk reset resource define
[   13.379479] mpp_rkvenc2 fdbd0000.rkvenc-core: rockchip_init_read_margin: no regulator (venc) found: -19
[   13.379537] mpp_rkvenc2 fdbd0000.rkvenc-core: Failed to get leakage
[   13.379581] mpp_rkvenc2 fdbd0000.rkvenc-core: error -ENODEV: _opp_set_regulators: no regulator (venc) found
[   13.379607] mpp_rkvenc2 fdbd0000.rkvenc-core: failed to set opp config
[   13.379614] mpp_rkvenc2 fdbd0000.rkvenc-core: failed to init opp info
[   13.379619] mpp_rkvenc2 fdbd0000.rkvenc-core: failed to init_opp_table
[   13.379625] rkvenc_init:2147: failed to add venc devfreq
[   13.380172] mpp_rkvenc2 fdbe0000.rkvenc-core: rockchip_init_read_margin: no regulator (venc) found: -19
[   13.380193] mpp_rkvenc2 fdbe0000.rkvenc-core: Failed to get leakage
[   13.380230] mpp_rkvenc2 fdbe0000.rkvenc-core: error -ENODEV: _opp_set_regulators: no regulator (venc) found
[   13.380252] mpp_rkvenc2 fdbe0000.rkvenc-core: failed to set opp config
[   13.380258] mpp_rkvenc2 fdbe0000.rkvenc-core: failed to init opp info
[   13.380264] mpp_rkvenc2 fdbe0000.rkvenc-core: failed to init_opp_table
[   13.380269] rkvenc_init:2147: failed to add venc devfreq
[   13.380858] rockchip-dmc dmc: Failed to get specification_serial_number
[   13.380900] rockchip-dmc dmc: Failed to get leakage
[   13.389776] mali fb000000.gpu: Failed to get specification_serial_number
[   13.389800] mali fb000000.gpu: Failed to get specification_serial_number
[   13.389826] mali fb000000.gpu: Failed to get leakage
[   13.389876] debugfs: Directory 'fb000000.gpu-mali' with parent 'vdd_gpu_s0' already present!
[... 18 lines cut ...]
[   16.193196] [dhd] STATIC-MSG) dhd_static_buf_init : 101.10.361.36 (wlan=r892223-20231107-1)
[   16.193276] [dhd] STATIC-MSG) dhd_init_wlan_mem : prealloc ok for index 0: 1102848(1077K)
[   16.282086] Goodix-TS 7-0014: Error reading 1 bytes from 0x8140: -6
[   16.312441] Goodix-TS 7-0014: Error reading 1 bytes from 0x8140: -6
[   16.341872] Goodix-TS 7-0014: I2C communication failure: -6
[   16.533634] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 3, size 139264
[   16.533789] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 4, size 0
[   16.548877] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 7, size 42288
[   16.548973] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : section 0, size 10320
[... cut ...]
[   18.593250] rk-pcie fe190000.pcie: PCIe Link Fail, LTSSM is 0x3, hw_retries=1
[   18.619564] rk-pcie fe170000.pcie: PCIe Link Fail, LTSSM is 0x3, hw_retries=1
[   19.618452] rk-pcie fe190000.pcie: failed to initialize host
[   19.645047] rk-pcie fe170000.pcie: failed to initialize host
[   39.165057] [GTP-ERR][__do_register_ext_module:79] Module [goodix-fwu] timeout

``` 

Do i have to worry about this? 
(Sorry for changing the subject, Is it better if I post this in another issue or in the forum??)

Again: Thank you!

MichaIng, 2025-01-08:
This is (sadly) normal for those vendor kernels. Mainline Linux usually does not show errors, but those based on Rockchip Linux sources throw a lot of errors, which can/must be ignored. Some of them is due to too high message severity, others due to loaded drivers without an actually available device. E.g. in your case, maybe there is no screen attached, leading to additional video related errors.