MichaIng/DietPi #7341: New install on odroid X4U (HC2) fails to boot after first start

lebochequirit, 2025-01-03:
### Creating a bug report/issue
- [ ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | Arm7-Bookworm
- Kernel version | 
- SBC model | odroid HC2 (hardware compatible to XU4)
- Power supply used | (EG: 12V Power)
- SD card used | ACME

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Steps to reproduce
1. Flash image to SD card
2. boot up system
3. login via SSH as root, watch initial system update
4. connection closes, on reboot no new connection possible
5. no further methods to troubleshoot, the device has no HDMI out, /var/log is emtpy on the SD card.

#### Expected behaviour

- after first boot, the system should update and reboot into a running system

#### Actual behaviour

- system doesn't come up after reboot and is neither pingable nor loginable

#### Extra details

- there's a HDD drive attached, but it doesn't show any action. Obviously no resize filesystem or so running

Joulinar, 2025-01-03:
similar to https://dietpi.com/forum/t/odroid-xu4-kernel-panic-after-update/22405

MichaIng, 2025-01-04:
I took the faulty kernel off the APT repo, now it should work again (with a new image). I'll close this and further discuss on the forum, where more info is collected.

closed 2025-01-04 by MichaIng

lebochequirit, 2025-01-04:
Can confirm it works again. Thanks a lot for fixing this so quickly!

MichaIng, 2025-05-31:
It seems to have been a buggy kernel config, fixed with: https://github.com/armbian/build/commit/5115cdf
I'll push a new tested kernel build to our APT repository.