MichaIng/DietPi #7343: Navidrome Music Server not starting anymore after updating to latest version 0.54.3

bartolus39, 2025-01-04:
#### Required Information
- DietPi version | v9.9.0
- Distro version | bookworm
- Kernel version | 6.6.62+rpt-rpi-v8
- SBC model | RPi 4 Model B
- Power supply used | 5V 3A official RAVpower)
- SD card used | SanDisk SSD via USB

#### Additional Information (if applicable)
- Software title | navidrome
- An update was performed from v0.53.3 to v0.54.3
- Can this issue be replicated on a fresh installation of DietPi? I do not know

#### Steps to reproduce
1. Having installed navidrome v0.53.3
2. Update to v0.54.3

#### Expected behaviour
- Navidrome page should be available on x.x.x.x:4533

#### Actual behaviour
- Navidrome is obviously not running

#### Extra details
- journalctl -u navidrome.service says:
Jan 04 11:23:52 DietPi systemd[1]: Started navidrome.service - navidrome (DietPi).
Jan 04 11:23:52 DietPi (avidrome)[546]: navidrome.service: Failed to locate executable /opt/navidrome/navidrome: Permission denied
Jan 04 11:23:52 DietPi (avidrome)[546]: navidrome.service: Failed at step EXEC spawning /opt/navidrome/navidrome: Permission denied
Jan 04 11:23:52 DietPi systemd[1]: navidrome.service: Main process exited, code=exited, status=203/EXEC
Jan 04 11:23:52 DietPi systemd[1]: navidrome.service: Failed with result 'exit-code'.
Jan 04 11:23:52 DietPi systemd[1]: navidrome.service: Scheduled restart job, restart counter is at 1.
Jan 04 11:23:52 DietPi systemd[1]: Stopped navidrome.service - navidrome (DietPi).

So probably it is somehow related to permissions. I have also a navidrome server running in a LXC container on proxmox -> same problem there. I also could not fix it there, so I used a backup version of the container.

Happy New Year from Munich!

BR, Bernhard

Joulinar, 2025-01-04:
Why not using search function? https://github.com/MichaIng/DietPi/issues/7336

closed 2025-01-04 by Joulinar

bartolus39, 2025-01-04:
I searched for it, but I was obviously unable to find it .... Sorry. 
Thanks for the link!

BR
Bernhard

MichaIng, 2025-01-04:
I pinned it now. Since it is a multi-line step, not so easy to ship a live patch for that. Luckily it is easy to fix afterwards. I added the fix to the top of the thread as well.