MichaIng/DietPi #7344: Orange Pi 5 Max spl module (zfs-dkms package) load error

enx1998, 2025-01-04:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `9.9.0`
- Distro version | `bookwarm`
- Kernel version | `Linux leonard 6.1.84-vendor-rk35xx #1 SMP Tue Dec 24 13:36:04 UTC 2024 aarch64 GNU/Linux`
- SBC model | `Orange Pi 5 Max (aarch64)`
- Power supply used | `27W 5.1V 5A`
- SD card used | none

#### Additional Information (if applicable)
- Software title | zfs-dkms

#### Steps to reproduce

1. update `linux-image-vendor-rk35xx`  to `25.02.0-trunk-dietpi1` version
2. install `zfs-dkms` package any version available (`2.2.7-1~bpo12+1` for example)
3. `modprobe zfs` or `modprobe spl`

#### Expected behaviour
Correctly load the modules

#### Actual behaviour
the modules do not load and with dmesg I see this errors
```
[    8.530596] Internal error: Oops: 0000000096000004 [#1] SMP
[    8.531095] Modules linked in: spl(O+) pwm_fan rk805_pwrkey goodix_ts nvmem_rockchip_otp btsdio panfrost drm_shmem_helper gpu_sched rockchip_cpuinfo uio_pdrv_genirq uio fuse dm_mod nfsd ip_tables ipv6 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx raid1 raid0 multipath linear md_mod r8169 adc_keys pwm_bl
[    8.533738] CPU: 2 PID: 837 Comm: modprobe Tainted: G           O       6.1.84-vendor-rk35xx #1
[    8.534513] Hardware name: RK3588 OPi 5 Max (DT)
[    8.534924] pstate: 80400009 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[    8.535545] pc : mod_sysfs_setup+0x230/0x518
[    8.535942] lr : mod_sysfs_setup+0x17c/0x518
[    8.536332] sp : ffff80000bfbbb30
[    8.536633] x29: ffff80000bfbbb30 x28: ffff80000ab1a910 x27: 0000000000000000
[    8.537270] x26: ffff800001491da8 x25: ffff80000ab1a8a8 x24: ffff800001491ad0
[    8.537906] x23: ffff80000bfbbd30 x22: ffff800001491a98 x21: ffff80000148d390
[    8.538542] x20: ffff800001491a80 x19: 0000000000000000 x18: 0000000000000000
[    8.539178] x17: ffff000103ffd5f0 x16: ffff000103ffd5b0 x15: ffff000103ffd570
[    8.539814] x14: ffff000103ffd530 x13: 00646e69625f6461 x12: 657268745f716b73
[    8.540450] x11: ffff000103ffd770 x10: 0000000000000000 x9 : ffff8000092316d0
[    8.541086] x8 : 0101010101010101 x7 : 7f7f7f7f7f7f7f7f x6 : ffff0001088e2d19
[    8.541722] x5 : 0000000000000000 x4 : 0000000000000000 x3 : 0000000000000000
[    8.542358] x2 : ffff800001491a98 x1 : ffff800001491ad0 x0 : 97ffe3ec97ffed00
[    8.542994] Call trace:
[    8.543219]  mod_sysfs_setup+0x230/0x518
[    8.543578]  load_module+0x1870/0x1990
[    8.543918]  __do_sys_finit_module+0xf8/0x118
[    8.544311]  __arm64_sys_finit_module+0x24/0x30
[    8.544725]  invoke_syscall+0x8c/0x128
[    8.545063]  el0_svc_common.constprop.0+0xd8/0x128
[    8.545498]  do_el0_svc+0xac/0xbc
[    8.545801]  el0_svc+0x2c/0x54
[    8.546087]  el0t_64_sync_handler+0xac/0x13c
[    8.546478]  el0t_64_sync+0x19c/0x1a0
```

#### Extra details
- with the previous kernel version I had no problem

Thanks, ENx

MichaIng, 2025-01-05:
So the kernel update solved some problem, and caused one as well 😄. I guess we need to wait for mainline kernel support, to have a more stable experience. But of course you can downgrade to the previous kernel at any time, and set the packages on hold:
```sh
apt install linux-{image,dtb,headers}-vendor-rk35xx=24.11.0-trunk-dietpi1
apt-mark hold linux-{image,dtb,headers}-vendor-rk35xx
```
Were there any errors during module compilation via DKMS?

enx1998, 2025-01-06:
Hi @MichaIng, thanks for your reply.
Yes downgrading kernel and related packages everything is working great, thanks!
> Were there any errors during module compilation via DKMS?

With the new kernel, there was no error during compilation via DKMS, only when loading modules.

MichaIng, 2025-01-07:
Okay. For now we need to live with that workaround. The solved issues are more general than this particular one with a specific DKMS-build module. The call trace does not tell me much, other than that it failed here:
- https://github.com/armbian/linux-rockchip/blob/rk-6.1-rkr4.1/kernel/module/internal.h#L262
- https://github.com/armbian/linux-rockchip/blob/rk-6.1-rkr4.1/kernel/module/sysfs.c#L370

Maybe an `strace` can reveal more. Since Armbian has ZFS support more integrated, there is some hope one of their devs with more experience faces it.

MichaIng, 2025-02-21:
Btw the issue is a general one with the recent RK3588 kernel. See here: #7412
As we collected some more info over there, I'll close this issue.

closed 2025-02-21 by MichaIng