MichaIng/DietPi #7345: Mympd no longer available?

hackslikeus, 2025-01-05:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version` DietPi v9.9.0
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN` bookworm 1
- Kernel version | `uname -a` Linux crapbeate 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3) RPi Zero W (armv6l)
- Power supply used | (EG: 5V 1A RAVpower) - 5v USB connector to monitor
- SD card used | (EG: SanDisk ultra) - Microcenter 16g class 10

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud) - MyMPD
- Was the software title installed freshly or updated/migrated? - No, no option to install, that's the bug.
- Can this issue be replicated on a fresh installation of DietPi? - Last image (9.8 Download 12/30/2024) - no attempted downloads since, but have upgraded to 9.9 via dietpi-update
 b33c1376-d2d3-4e6a-b579-bdc567633f9c
- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...Dietpi-Config/Software
2. ...Browse Software - no option for MyMPD available in full list
3. ...Search Software - Search "mypmd" - results in: "We couldn't find any available software title for the search term: "mympd"
4. ...

#### Expected behaviour
Mympd should be available under media systems, searchable, and installable
- ...

#### Actual behaviour
 no sign that Mympd is part of the image offering as an option (was in the past and is still listed under software options)
- ...

#### Extra details

- ...would prefer not to re-image, if an option to reinstall by #id is a hot fix, please provide mympd software id#.

Joulinar, 2025-01-05:
> Mympd should be available

It is available still

> no sign that Mympd is part of the image offering

Software options are not related to a dedicated image. It depends on SBC architecture and Debian version 

> would prefer not to re-image

Flashing a new image might not change anything as it depends on SBC architecture and Debian version 

> please provide mympd software id#.

https://github.com/MichaIng/DietPi/wiki/DietPi-Software-list

Simply check availability on your system as follows 

```
dietpi-software list | grep myMP
```

hackslikeus, 2025-01-05:
so mympd was disabled:
dietpi-software list | grep myMPD
/boot/dietpi/dietpi-software: line 1463: modprobe: command not found
ID 148 | =0 | myMPD: fork of ympd with improved features | +MPD DISABLED for Debian bookworm | https://dietpi.com/docs/software/media/#mympd

Joulinar, 2025-01-05:
best to my knowledge, there is no myMPD package for armv6l/bookworm. Therefore, it's disables on your SBC.

correct @MichaIng ?

MichaIng, 2025-01-06:
At least it as like that, but a Raspbian_12 suite has been added in the meantime: https://download.opensuse.org/repositories/home:/jcorporation/Raspbian_12/

I'll enable it with next release.

hackslikeus, 2025-01-07:
Much obliged Michalng.  FWIW, I was able to add it in the hard way, using the steps I found on the LinuxAudioFoundation website:   
   -    $ sudo git clone https://github.com/jcorporation/myMPD.git
   -    $ cd myMPD
   -    $ sudo ./build.sh installdeps
   -    $ sudo ./build.sh release
   -       sudo ./build.sh installI then followed the configuration options found at Jcorporation
   -    systemd-run -p DynamicUser=yes -p StateDirectory=mympd -p CacheDirectory=mympd -E MYMPD_LOGLEVEL=4 -E MYMPD_HTTP=false -E MYMPD_SSL_PORT=1333 mympd -c 
   -    and a little port & SSL certification re-configuration:   sudo nano /var/lib/mympd/config/http_portAnd finally some enablement as I am an enabler:
   -    sudo systemctl enable mympd
And it works right out of the box...Took forever, especially steps 3-5, but a really nice mpd client

closed 2025-01-07 by hackslikeus