MichaIng/DietPi #7349: PaperMC installation not possible

onkeld, 2025-01-06:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues
#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`

 bullseye 0

- Kernel version | `uname -a`

Linux DietPi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)

RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)

irrelevant 

- SD card used | (EG: SanDisk ultra)

irrelevant

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)

PaperMC Minecraft Server

Update via dietpi-software reinstall fails with a 403 error when trying to fetch paperMC from papermc.io.

- Bug report ID 
486cd0df-8498-4516-8092-e60506831654

#### Steps to reproduce
1. run sudo dietpi-software reinstall 181

#### Expected behaviour
- paperMC minecraft server should be reinstalled in the most recent publicly available stable version

#### Actual behaviour

Command: curl -sSfL https://papermc.io/api/v2/projects/paper/versions//builds//downloads/paper--.jar -o  paper--.jar
 - Exit code: 22
 - DietPi version: v9.9.0 (MichaIng/master) | HW_MODEL: 4 | HW_ARCH: 3 | DISTRO: 6
 - Error log: curl: (22) The requested URL returned error: 403

MichaIng, 2025-01-06:
Many thanks for reporting. The API URL changed from a sub path to a sub domain 😄. Fixed with: https://github.com/MichaIng/DietPi/commit/8fd0327

I wonder whether it is worth a live patch. As of the small change, it would be an easy one 🤔. So people do not need to next DietPi release.

MichaIng, 2025-01-06:
Okay, while the live patch works, there is a dedicated issue, which cannot be patched that easily: Minecraft 1.20.5 and above requires Java 21, available on Debian Trixie only. So we need to install v1.20.4 in that case. So the live patch does not make sense: We fix the download, but the service won't start successfully. I fixed it for upcoming DietPi release: https://github.com/MichaIng/DietPi/commit/e0430a7

MichaIng, 2025-01-07:
Hmm, pinning older Minecraft version is now awesome either. But installing new Java versions on older Debian is as well not that trivial, and can conflict with other software titles. Though we have a request for this very reason: #4505

... wait, it has become much easier now to install other Java versions: https://github.com/MichaIng/DietPi/issues/4505#issuecomment-2574240256

MichaIng, 2025-01-08:
Yeah, for APIs this is especially nasty 😄. Would have been trivial to add a redirect, and inform people through other channels about the deprecated old URL. Though I cannot rule out this has happened, and we just missed it. There is no real way to add deprecation messages to the API itself, without distorting the output either.

If you actually stored the empty response, then I suggest to update your script. E.g. in case of `curl`, use the `-f` flag to print the HTTP error to STDERR, and use `-o` to store the actual response body, if there is any and no error, instead of shell redirect `>`. This way you see such errors in the output and store content only on successfull 2xx responses.

Jakesta13, 2025-01-10:
> Yeah, for APIs this is especially nasty 😄. Would have been trivial to add a redirect, and inform people through other channels about the deprecated old URL. Though I cannot rule out this has [...]

Appreciate this recommendation, I never even thought about this.

closed 2025-02-09 by MichaIng