MichaIng/DietPi #7361: One single image for Pi Zero W2, Pi3, Pi4 and Pi5

DrCWO, 2025-01-16:
### Creating an image request

I currently use the same ARMv8 image for Pi Zero W2, Pi3 and Pi4.
Is there a Bullseye image supporting Pi Zero W2, Pi3, Pi4 and Pi5 using the same image?
This will help me a lot as I only have to distribute and administrate one image for all platforms.

Best DrCWO

Joulinar, 2025-01-16:
I don't think there will be a single image for all. But you can use our latest https://dietpi.com/downloads/images/testing/DietPi_RPi234-ARMv8-Bookworm.img.xz image supporting W2, Pi2 Pi3, Pi4 and install the Pi5 kernel using `dietpi-config`. This way you could create an own image and move it between all device.

MichaIng, 2025-01-16:
That image supports the RPi 5 as well. It ships with the 4k page size kernel, that supports all (64-bit capable) models. The RPi5 image ships with the 16k page size kernel, which might have some performance benefits on RPi 5, but is not supported by the older ARMv8 arch of the older RPi models.

Did you btw mean Bullseye for some reason, or is Bookworm fine? While those kernel packages are generally distro version agnostic, they are available on the RPi Bookworm repo only. So they would need to be pulled/provided separately, to generate a Bullseye image with them.

DrCWO, 2025-01-29:
Well, I did not get it completely I believe :-(

With my two productive releases I'm still on Bullseye showing this:
Raspberry Pi4 `Linux rooExtend 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux`
Friendly ARM R2S: `Linux rooExtend 5.15.52-rockchip64 #22.05.4 SMP PREEMPT Mon Jul 4 09:40:29 CEST 2022 aarch64 GNU/Linux`

Some time ago I tried this with a Pi5 but no luck.

Did I understand it right that there is a chance to update the kernel of the Pi (and some other files) in /boot so it boots on a Pi5? This would be the perfect solution for me at the moment saving my time upgrading to Bookworm.

MichaIng, 2025-01-29:
@DrCWO even on Bullseye, your kernel should be Linux 6.1. And you need the new Bookworm image image linked by Joulinar or upgrade your system to Bookworm, then migrate to the new kernel stack, to support RPi 5.

But let's find out why you are running an outdated kernel:
```sh
dpkg -l rapberrypi-kernel
apt policy rapberrypi-kernel
ls -l /lib/modules/
dpkg -S "/lib/modules/$(uname -r)"
```

DrCWO, 2025-01-30:
Hi, here the requested output:

```
root@rooExtend:/home/pi# dpkg -l rapberrypi-kernel
dpkg-query: no packages found matching rapberrypi-kernel
root@rooExtend:/home/pi# apt policy rapberrypi-kernel
N: Unable to locate package rapberrypi-kernel
root@rooExtend:/home/pi# ls -l /lib/modules/
total 4
drwxr-xr-x 3 root root 4096 Oct 23  2022 5.15.61-v8+
root@rooExtend:/home/pi# dpkg -S "/lib/modules/$(uname -r)"
raspberrypi-kernel: /lib/modules/5.15.61-v8+
root@rooExtend:/home/pi#
```

Main reason that I'm still on this release is that there are several hundred unattended Raspberry PIs with my software out in the field. These boxes run an unattended nightly update of my own software. But updating the system without touching it from remote is not so easy. I did a lot of modifications and a simple Dietpi update breaks my setup.

On the system I configured only very few processes run as it is just a host for my own software. Chromium runs in Kiosk-Mode via HDMI.
`systemd-journald, systemd-udevd, dbus-daemon, rngd, dhcpd, dropbear, agetty, avahi-daemon, cron, hciattach, bluetoothd, openbox, chromium, node`

I plan to build a new image some day on a more modern Linux but if I do this I like to have a system that supports all ARMv8 PIs PI5 included.

In the meantime it would be great if I might to upgrade the existing image so it can be used on Pi5.

MichaIng, 2025-01-30:
Oh `rapberrypi-kernel` was missing and `s`, my bad. But as far as I understood, you just did never run `apt upgrade` for a long time, and doing so would install Linux 6.1, right?

As said, RPi 5 strictly requires Bookworm and latest kernel, so indeed you would need to migrate your software then to support the new packages.

But I wonder, which APT upgrades break your software  in particular, and how? Because Debian does actually not push any breaking changes to their stable repository, so it should be very rare and in case easy to fix any issues caused by `apt upgrade`. That least, at least as long as you do not use 3rd party repositories which follow their own upgrade policy, or update to a new Debian version.

DrCWO, 2025-01-30:
Thank you for clarification. 
Yes, I use some 3rd party repositorys and [had to go back in Bluetooth as there was a reconnect delay](https://github.com/MichaIng/DietPi/issues/6127#event-14328740893) in the latest release I tested. 

So as soon as I can  I will set up a new image and try to automate my installation so in future it should be easier for me to migrate.
Thanks' for your work and your excellent support 👍

MichaIng, 2025-01-30:
Ah right, now I remember. An `apt-mark hold bluez` should allow you to update DietPi and all other APT packages to the latest version, keeping `bluez` at what it is. But in any case, it makes sense to test with Bookworm and the new RPi kernel.

MichaIng, 2025-07-18:
The `dev` branch version of the build script btw allows to generate 64-bit as well as 32-bit (userland) images for all RPi models (64-bit for RPi 2 v1.2 up only, of course) with new kernel stack, using model `0`. I do not aim to distribute them, but needed to implement this properly for testing rustup on 32-bit userland with 64-bit kernel. Use the `--arch` option to choose between ARMv6 (Raspbian), ARMv7 and ARM64 userland, and the installer will install all kernel packages for the image to run on all RPi models, which can work with that userland architecture. In the particular case with 64-bit, it does not make so much difference, since our RPi 2-4 image supports the RPi 5 as well, but the following will include the 16k page size kernel as well, to be used automatically on RPi 5:
```sh
dietpi-build --model 0 --arch 3
```
or
```sh
dietpi-build -m 0 -a 3
```
For 32-bit, it really adds more flexibility, adding 32-bit and 64-bit kernel images, to support all RPi models.