MichaIng/DietPi #7364: Vaultwarden version not displayed in admin panel (or anywhere?)

gh3597, 2025-01-19:
### Creating a bug report/issue
- [Tick ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | DietPi v9.8.0 (although this issue has been apparent in multiple versions previous to this going back for a couple of years)
- Distro version | bullseye
- Kernel version | Linux vaultwarden 5.10.0-33-amd64 #1 SMP Debian 5.10.226-1 (2024-10-03) x86_64 GNU/Linux
- SBC model | Virtual Machine (x86_64)
- Power supply used | Not relevant but redundant 960W power supplies!
- SD card used | Again not relevant but installed on RAID5 array on 4x 1TB WD RED drives

#### Additional Information (if applicable)
- Software title | vaultwarden
- Was the software title installed freshly or updated/migrated? Installed with DietPi using the extra software options.
- Can this issue be replicated on a fresh installation of DietPi? Yes.

#### Steps to reproduce
Log in to the admin panel of vaultwarden.
Go to diagnostics
View the blank space where current version should be
[image]

#### Expected behaviour
Vaultwarden version should be displayed. 

#### Actual behaviour
Version is not being populated. 

#### Extra details
I posted this as an issue with vaultwarden but they have passed it back to the packager.

Some info from that discussion. https://github.com/dani-garcia/vaultwarden/discussions/4936

> if packagers provide the right env variable during compile time it should also work
> For example, DietPi could use ```VW_VERSION="1.32.7 - DietPi Edition"```

MichaIng, 2025-01-25:
I just pushed a new build to our APT servers, which should show the version in the web UI.

closed 2025-02-09 by MichaIng

gh3597, 2025-02-15:
Just tested this and it is better but seems that the build is displayed in the current version and the DietPi build differs slightly so it shows there's an update available,but there isn't. 

[image]

MichaIng, 2025-02-15:
It shows the very latest commit of the master branch: https://github.com/dani-garcia/vaultwarden/commit/3baffee
That is however not released: https://github.com/dani-garcia/vaultwarden/releases
The latest release tag `1.33.2` has the commit `1f868b8`. So IMO this information is somewhat misleading, at least if the aim is to show the latest release.

gh3597, 2025-02-15:
> IMO this information is somewhat misleading, at least if the aim is to show the latest release

That was my thinking too, both bits of info are superfluous here. Just 1.33.2 is all that is required. Defintiely "workable" as it is but a minor annoyance nonetheless. Appreciate this is an issue with the package and not DietPi.

MichaIng, 2025-02-15:
Based on the info [here](https://github.com/dani-garcia/vaultwarden/discussions/4936#discussioncomment-12211345), I replaced the `-` with a `+`. But actually you are right, and we could just remove the suffix completely 🤔. It is important for the package version, as we may do a rebuild with same vaultwarden version but different web vault, or something around packaging or config changed. But the vaultwarden source version/binary is not affected by this, so I guess indeed we can just remove the suffix entirely.