MichaIng/DietPi #7365: Problem iso orangepi 3 lts

Seb-tech77, 2025-01-20:
Hello, I have a problem when writing the image of orange pi 3 lts on my sd card, balena etcher puts me an error while with other images it works perfectly. Could you help me?

Cordially 

[image]

MichaIng, 2025-01-20:
Thanks for reporting. Does it work with [rufus](https://rufus.ie/)?

Seb-tech77, 2025-01-20:
Indeed, I had not tested with rufus, it seems to work. I would test to see if the boot card correctly 

[image]

MichaIng, 2025-01-20:
I remember such was reported before. I guess we need to check which problem extractly BelanaEtcher has with our `.img.xz` files. Btw did you extract it beforehand or selected the xz archive as is?

Seb-tech77, 2025-01-21:
Hello, I extracted the disk image from the archive and copied it with balenaetcher

Seb-tech77, 2025-01-22:
okay, no worries. I'll do this with rufus then if it works. Thanks for your help

MichaIng, 2025-02-09:
And did it work with Rufus?

Seb-tech77, 2025-02-10:
Hi, sorry I totally forgot to test.

I tested starting up with rufus and it boots fine and I get no errors.

MichaIng, 2025-02-16:
I just tried it with a NanoPi M6 image, and it worked well. I first tried it selecting the compressed `xz` file, which it extracts automatically internally. Then I extracted it manually, and selected that one, and it worked as well.

Then I tried with the Orange Pi 3 LTS image, and it failed with the same error you were showing. This is weird, since the image has been updated in the meantime, has a different size etc. I was expected that balenaEtcher might expect images to be e.g. sized to round 4 MiB sizes or something like that.

I tried a bunch of other images then, also with same kernel, and all succeeded. Then I tried the Orange Pi 3 LTS image again, which previously failed, and for whatever reason it succeeded 🤔. It failed 2 times previously. No idea what to think of this. I also downloaded it again, and it still works. I guess if anyone faces this issue again, we need to ask at Etcher GitHub repo how to better debug it: https://github.com/balena-io/etcher/issues

Actually there is a matching report already: https://github.com/balena-io/etcher/issues/4396

MichaIng, 2026-11-09:
Closing this for now. I assume it has been fixed in the meantime. If there is any future report or similar flashing issues, we'll ban it from our docs, and recommend Rufus on Windows, and `dd` for Linux and macOS (any fancy flashing/GUI tool is overkill for our small images).

closed 2026-11-09 by MichaIng