MichaIng/DietPi #7367: Latest Bookworm Image for the Neo 2 Black seems to be broken

peedy2495, 2025-01-22:
Writing on different SD-Cards doesn't matter.

I tried the "normal" neo2 image, too.
This one is working like a charm.

Therefore it seems that the neo2 black image has an issue.

MichaIng, 2025-01-22:
Many thanks for your report.

Hmm, the images are identical despite the bootloader. Do you have a USB-UART adapter to check serial console output? I also just see it seems to support a serial console via USB, but it would need to be enabled. Can you access the ext4 partition from another system, in case?

peedy2495, 2025-01-22:
Yes, I'm able to access the rootfs partition using a card adapter for my laptop.

With the black edition image the firstboot is alwasy stopping on the same place.
First, i had to solder the serial debug pins ... ;-))
Here is the full debug output:
```
U-Boot SPL 2024.01-armbian-2024.01-S866c-Pa395-Ha9af-Vbe06-Bda0a-R448a (Oct 02 2024 - 03:34:11 +0000)
DRAM: 1024 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.9(debug):armbian
NOTICE:  BL31: Built : 03:33:22, Oct  2 2024
NOTICE:  BL31: Detected Allwinner H5 SoC (1718)
NOTICE:  BL31: Found U-Boot DTB at 0x20a2340, model: FriendlyARM NanoPi NEO Core 2
INFO:    ARM GICv2 driver initialized
INFO:    Configuring SPC Controller
NOTICE:  PMIC: Assuming H5 reference regulator design
INFO:    BL31: Platform setup done
INFO:    BL31: Initializing runtime services
INFO:    BL31: cortex_a53: CPU workaround for 855873 was applied
INFO:    BL31: cortex_a53: CPU workaround for 1530924 was applied
SCP/INF: Crust v0.6.10000
INFO:    PSCI: Suspend is available via SCPI
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x4a000000
INFO:    SPSR = 0x3c9
ns16550_serial serial@1c28000: pinctrl_select_state_full: uclass_get_device_by_phandle_id: err=-19

U-Boot 2024.01-armbian-2024.01-S866c-Pa395-Ha9af-Vbe06-Bda0a-R448a (Oct 02 2024 - 03:34:11 +0000) Allwinner Technology

CPU:   Allwinner H5 (SUN50I)
Model: FriendlyARM NanoPi NEO Core 2
DRAM:  1 GiB
Core:  61 devices, 19 uclasses, devicetree: separate
WDT:   Not starting watchdog@1c20ca0
MMC:   mmc@1c0f000: 0, mmc@1c11000: 1
Loading Environment from FAT... Unable to use mmc 0:1...
In:    serial,usbkbd
Out:   serial,vidconsole
Err:   serial,vidconsole
Net:   No ethernet found.
starting USB...
Bus usb@1c1b000: USB EHCI 1.00
scanning bus usb@1c1b000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Autoboot in 1 seconds, press <Space> to stop
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
2765 bytes read in 2 ms (1.3 MiB/s)
## Executing script at 4fc00000
308 bytes read in 1 ms (300.8 KiB/s)
23443464 bytes read in 971 ms (23 MiB/s)
9957636 bytes read in 442 ms (21.5 MiB/s)
34618 bytes read in 6 ms (5.5 MiB/s)
Working FDT set to 4fa00000
504 bytes read in 5 ms (97.7 KiB/s)
Applying kernel provided DT overlay sun50i-h5-usbhost1.dtbo
504 bytes read in 5 ms (97.7 KiB/s)
Applying kernel provided DT overlay sun50i-h5-usbhost2.dtbo
4191 bytes read in 5 ms (818.4 KiB/s)
Applying kernel provided DT fixup script sun50i-h5-fixup.scr
## Executing script at 45000000
Moving Image from 0x40080000 to 0x40200000, end=418f0000
## Loading init Ramdisk from Legacy Image at 4ff00000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    9957572 Bytes = 9.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
[... 628 lines cut ...]
504 bytes read in 5 ms (97.7 KiB/s)
Applying kernel provided DT overlay sun50i-h5-usbhost2.dtbo
4191 bytes read in 5 ms (818.4 KiB/s)
Applying kernel provided DT fixup script sun50i-h5-fixup.scr
## Executing script at 45000000
Moving Image from 0x40080000 to 0x40200000, end=418f0000
## Loading init Ramdisk from Legacy Image at 4ff00000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    9957460 Bytes = 9.5 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
Working FDT set to 4fa00000
   Loading Ramdisk to 49680000, end 49fff054 ... OK
   Loading Device Tree to 000000004960f000, end 000000004967ffff ... OK
Working FDT set to 4960f000

Starting kernel ...
```

Take a look at the ethernet device, too!
In the black edition it hasn't been recognized.

MichaIng, 2025-01-23:
Okay, while the bootloader does not detect the Ethernet, it detects the eMMC directly and loads kernel etc successfully. So you say at the end of the output, it does not do anything anymore? This is from the filesystem expansion script, which did expand partition successfully at this point. The next commands are `partprobe` and `partx -u` to tell the kernel about the new partition size, and then `resize2fs` runs to max the filesystem size as well. The first two commands do not produce any output, but `resize2fs` does, at least once it finished. It actually indeed takes a little time sometimes, on slower SD cards, or very large devices. But it should finish eventually within a minute max. It any of those 2 commands is failing, a reboot is triggered.

What caught my eyes is that the bootloader is actually not for the NEO2 Black, but the NEO Core2. This is intentional, but not sure whether correct. At least it is two dedicated SBCs:
- https://github.com/armbian/build/blob/main/config/boards/nanopineo2black.csc
- https://wiki.friendlyelec.com/wiki/index.php/NanoPi_NEO_Core2
- https://wiki.friendlyelec.com/wiki/index.php/NanoPi_NEO2_Black

Mainline U-Boot does not support any of those two, but they are patched inside by the Armbian build system:
- https://github.com/armbian/build/tree/main/patch/u-boot/u-boot-sunxi/board_nanopineo2black
- https://github.com/armbian/build/blob/main/patch/u-boot/u-boot-sunxi/board_nanopineocore2/add-xx-nanopineocore2.patch

While FriendlyELEC's U-Boot fork does not provide a dedicated defconf for that board, the board detection code seems to indicate the the NEO2 Black is closer to the NEO2, than the NEO Core2: https://github.com/friendlyarm/u-boot/blob/sunxi-v2017.x/common/board_r.c#L842-L871

But it is probably a particular one particular hardware feature which lead to taking the Core2 config and device tree. Does the NEO2 image probably limit available RAM to 512 MiB on the 1 GiB RAM variant of the NEO2 Black? The U-Boot output however shows 1 GiB in both cases.

peedy2495, 2025-01-27:
The image of the Neo2 takes only a few minutes when resizing the partitions with resize2fs.
I gave the Black image a bit more than half an hour ... it really seems to get stuck here.
For the Neo2 Black, the only additional requirement should be the eMMC kernel driver.
I think that the memory is allocated dynamically.

MichaIng, 2025-01-27:
Ah right, the eMMC which the NEO Core2 has as well will e the reason why its bootloader is used. And since there is no other/dedicated one for the NEO2 Black in mainline U-Boot or compatible vendor branch, that one was seen and likely just worked well enough.

So the problem is not in the bootloader, but actually in the kernel/device tree. Let me check how it looks like:
- Patched inside here: https://github.com/armbian/build/blob/main/patch/kernel/archive/sunxi-6.12/patches.armbian/arm64-dts-Add-sun50i-h5-nanopi-neo-core2-device.patch
- Versus mainline NEO2: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/arch/arm64/boot/dts/allwinner/sun50i-h5-nanopi-neo2.dts

Difference in the SD card node: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit?id=aee2eca
This is for disabling write-protection. I mean, before the filesystem is expended, the SD card has already been written to, but probably it somehow has connected. Let me add this to the patch and see whether it makes a difference.

MichaIng, 2025-01-27:
U-Boot is ready for testing: https://github.com/MichaIng/build/commit/03b248f
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-u-boot-nanopineo2black-current.deb
dpkg -i linux-u-boot-nanopineo2black-current.deb
/boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
```
Then check whether it detects the Ethernet device now. As far as I can see, the kernel drive is enabled with the existing patch already.

MichaIng, 2025-01-27:
Can you test the image from here? It shops with the patched kernel and bootloader: https://dietpi.com/downloads/images/testing/

peedy2495, 2025-01-30:
Ok, I'll do it. But for now I have an hardware issue. I'll be back, soon.

MichaIng, 2025-05-31:
Cycling back to this issue. I regenerated new kernel and bootloader, as well as image, with above patches, so can be again tested:
- Image: https://dietpi.com/downloads/images/testing/
- Or install kernel+bootloader on running system, and test whether this allows the resize script to run without issues:
```sh
cd /tmp
wget https://dietpi.com/downloads/binaries/testing/linux-{image,dtb}-current-sunxi64.deb https://dietpi.com/downloads/binaries/testing/linux-u-boot-nanopineo2black-current.deb
dpkg -i linux-{image,dtb}-current-sunxi64.deb linux-u-boot-nanopineo2black-current.deb
/boot/dietpi/func/dietpi-set_hardware flash-uboot-mmc
systemctl enable dietpi-fs_partition_resize
reboot
```

MichaIng, 2025-07-18:
@peedy2495 
Do you find time to test the patched kernel and bootlaoder builds? I would love to tick this off the list.