MichaIng/DietPi #7368: ZFS problem

Seb-tech77, 2025-01-22:
Hello, I use openmediavault to manage my disks connected to my orange pi 5 pro. I wanted to switch my disk to zfs to benefit from data deduplication but when I want to use the OMV extension I get this error: 

OMV\ExecException: Failed to execute command 'export PATH=/bin:/sbin:/usr/bin:/usr/sbin:/usr/local/bin:/usr/local/sbin ; export LC_ALL=C.UTF-8 ; export LANGUAGE= ; zfs list -p -H -t all -o name,type 2>&1' with exit code '1': ZFS modules are not loaded.
Try running '/sbin/modprobe zfs' as root to load them. in /usr/share/php/openmediavault/system/process.inc:247
Stack trace :
#0 /usr/share/omvzfs/Utils.php(450) : OMV\System\Process->execute()
#1 /usr/share/omvzfs/Utils.php(262) : OMVModuleZFSUtil::exec()
#2 /usr/share/openmediavault/engined/rpc/zfs.inc(297) : OMVModuleZFSUtil::getZFSFlatArray()
#3 [internal function]: OMVRpcServiceZFS->listPools()
#4 /usr/share/php/openmediavault/rpc/serviceabstract.inc(124) : call_user_func_array()
#5 /usr/share/php/openmediavault/rpc/serviceabstract.inc(155) : OMV\Rpc\ServiceAbstract->callMethod()
#6 /usr/share/php/openmediavault/rpc/serviceabstract.inc(628) : OMV\Rpc\ServiceAbstract->OMV\Rpc\{closure}()
#7 /usr/share/php/openmediavault/rpc/serviceabstract.inc(152) : OMV\Rpc\ServiceAbstract->execBgProc()
#8 /usr/share/openmediavault/engined/rpc/zfs.inc(304) : OMV\Rpc\ServiceAbstract->callMethodBg()
#9 [internal function] : OMVRpcServiceZFS->listPoolsBg()
#10 /usr/share/php/openmediavault/rpc/serviceabstract.inc(124): call_user_func_array()
#11 /usr/share/php/openmediavault/rpc/rpc.inc(86): OMV\Rpc\ServiceAbstract->callMethod()
#12 /usr/sbin/omv-engined(544): OMV\Rpc\Rpc::call()
#13 {main}

Could you tell me if this error comes directly from openmediavault or I also saw that it could be an incompatibility between the linux kernel and the ZFS module?

Sincerely

MichaIng, 2025-01-22:
Did you install/compile the ZFS kernel module? This is not part of the kernel by default, but needs to be added via DKMS, which also requires kernel headers. Which kernel do you have installed?
```sh
dpkg -l | grep linux-image
```

Seb-tech77, 2025-01-22:
No, I haven't tried to compile the module for the kernel, I don't really know how to do it. 

`Sebastien@DietPi:~$ dpkg -l | grep linux-image
ii  linux-image-vendor-rk35xx        25.02.0-trunk-dietpi1               arm64        Armbian Linux vendor kernel image 6.1.84-vendor-rk35xx
Sebastien@DietPi:~$`

MichaIng, 2025-01-22:
Try this:
```sh
apt install linux-headers-vendor-rk35xx
apt install zfs-dkms
```

Seb-tech77, 2025-01-22:
It's done, both orders have been executed successfully.

`Armbian 'linux-headers-vendor-rk35xx' for '6.1.84-vendor-rk35xx': 'postinst' finishing.
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait
zfs-dkms est déjà la version la plus récente (2.1.11-1+deb12u1).
zfs-dkms passé en « installé manuellement ».
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
Sebastien@DietPi:~$`

MichaIng, 2025-01-22:
So `zfs-dkms` was installed already, but the headers were likely missing before. Please try to reconfigure it to compile the kernel module:
```sh
dpkg-reconfigure zfs-dkms
```

Seb-tech77, 2025-01-22:
the command is finished but I get a segmentation error when I try to launch the ZFS module

Sebastien@DietPi:~$ sudo modprobe zfs
Erreur de segmentation
Sebastien@DietPi:~$ zpool status
The ZFS modules are not loaded.
Try running '/sbin/modprobe zfs' as root to load them.
Sebastien@DietPi:~$

MichaIng, 2025-01-22:
Can you paste the output of the module build? And:
```sh
modinfo zfs
```

Seb-tech77, 2025-01-22:
No, it doesn't boot anymore, I can't even restore my backup now.

MichaIng, 2025-01-23:
The `netfilter-persistent` service hangs forever. But before that, `ifupdown-pre` fails. It basically waits for udev to settle, before network connections can be reliably established, else the network adapters might not be ready yet. But `ifup@eth0` etc do not depend on `ifupdown-pre`, so even if the latter fails, the former will run. But `netfilter-persistent` seems to aim for setting up firewall rules before networks are brought up, which makes sense, but hangs. This does not seem to be related to ZFS, as long as nothing it uses is on a ZFS filesystem, which hangs to mount of so.

Likely unrelated: You have `cpufrequtils` installed, which is not needed on DietPi, as `dietpi-preboot` applies the CPU governor and settings already, as can be seen some lines above. It does not hurt, but doubles the job. Also, the `loadcpufreq` service from this package fails for whatever reason, so that failure would at least be gone. You can change the CPU governor in `dietpi-config` performance options then, though there is almost never a reason to switch away from `schedutil`, IMO.

Why udev does not settle is another question, but the main problem is the netfilter service. You could disable the service by attacking the card/drive to another Linux system, then boot, and test `netfilter-persistent start` from console to see why it hangs.

Seb-tech77, 2025-01-24:
Hello, it's strange because net-filter has been installed for a while, it's only when I compiled the ZFS module that I had the problem. I know I don't see the connection either but apart from that I haven't done anything else. I've reset my system and I've managed to restore my backup so everything's fine, the system's working properly again. I'm going to make an image directly from my SD card and retest the commands. 

For the cpu module, it must have been installed by openmediavault by default, but I use dietpi-config to adjust my cpu performance.

MichaIng, 2025-01-24:
I'll try it on my Orange Pi 5.

Seb-tech77, 2025-01-24:
Okay, thanks for testing on your side, if I understand correctly there's just waiting for an update of the rockchip kernel? Hoping it will be solved

MichaIng, 2025-01-24:
Looks like it is the kernel headers, not the module/source itself:
```sh
root@DietPi:~# modprobe r8168
Segmentation fault
```
I'm triggering a new kernel build, just in case it has been fixed in the meantime.

MichaIng, 2025-06-22:
Yes they are. Tracking the issue here: #7412

closed 2025-06-22 by MichaIng