MichaIng/DietPi #7369: Java 21 LTS support

miloit, 2025-01-22:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

OpenHAB 5.0.0 is scheduled for the summer release, and it is essential to align with modern Java versions to leverage the latest language enhancements, performance improvements, and security updates. Currently, OpenHAB 4.x is not optimized for Java 21, which is the latest LTS (Long-Term Support) version of Java but it's running under Java 21. This limitation may hinder developers from utilizing Java 21 features and maintaining compatibility with the ecosystem.

#### Describe the solution you'd like:

Ensure OpenHAB 5.0.0 is fully compatible with Java 21 by updating dependencies, libraries, and configurations to support the latest features and runtime.

Conduct extensive testing to ensure stability and reliability when running OpenHAB 5.0.0 on Java 21.

Provide clear documentation and guidelines to migrate from Java 17 to Java 21.

#### Describe alternatives you've considered:

Maintain dual compatibility with Java 17 and Java 21, but this increases maintenance overhead and could lead to inconsistencies in supported features.

#### Additional context

OpenHAB’s move to Java 21 aligns with the industry trend of utilizing the latest LTS versions for modern application development.

MichaIng, 2025-01-22:
We currently install Java from the Debian repository, where Trixie ships Java 21: https://packages.debian.org/openjdk-21-jdk-headless

We do provide Trixie images, but since it is Debian testing, packages can receive major version upgrades, i.e. breaking changes: https://dietpi.com/downloads/images/

We are thinking about installing Java from another source. But it has not been implemented yet. See my last post, and follow the instructions to install any Java version via Adoptium APT repository: #4505

miloit, 2025-02-22:
So how do I install java 21 now the best way?

MichaIng, 2025-02-28:
A "best way" does not exist. But if you are not on Debian Trixie, the solution I aim to implement into DietPi is this: https://adoptium.net/de/installation/linux/#_deb_installation_on_debian_or_ubuntu

... the instructions are a bit outdated the `apt-transport-https` package is not required since ages in Debian, and `gpg` is not required anymore to dearmor the key, instead it can be stored to `/etc/apt/trusted.gpg.d/adoptium.asc` (the `asc` file extension is important) in ASCII armored format directly. However, it works.

closed 2025-07-18 by MichaIng