MichaIng/DietPi #7370: DietPi v9.9.0 breaks PCIE / NVMe on Pine SOQUARTZ compute modules (perhaps more), fixed in kernel 6.13.

m00dawg, 2025-01-22:
### Creating a bug report/issue
- [X] I have searched the existing open and closed issues (May relate https://github.com/MichaIng/DietPi/issues/7344)

#### Required Information
- DietPi version | 9.9.0, master, MichaIng
- Distro version | bookworm
- Kernel version | 6.12.8
- SBC model | Quartz64 (aarch64)
- Power supply used | SOQUARTZ Blade / PoE (D-Link 8-port PoE+ switch)
- SD card used | None, using MMC

#### Steps to reproduce
1. Ran dietpi-update which appeared to swap out the kernel and/or may have misconfigured the CM
2. After the upgrade, my NVMe would not be recognized and `lspci` didn't show any devices.
3. dmesg contained some potential PCIe errors (https://termbin.com/sv4p)
4. Fix: Upgrading the kernel to 6.13 seems to have the patch, but 6.13 isn't available within the repos and I have trouble rolling to a previous kernel (see below).

#### Expected behaviour
1. NVMe should be recognized with LVM volumes mounting per my fstab

#### Actual behaviour
1. Initially system didn't boot since I didn't have `noerrors` setup on the fstab for the volume. This was due to the NVMe not being there so LVM had nothing to do.

#### Extra details
1. I talked about this on the Quartz64 Matrix and it seems to be related to [this](https://lore.kernel.org/linux-rockchip/20250106100001.1344418-1-amadeus@jmu.edu.cn/T/#u) bug which has been fixed in Kernel 6.1.3
2. I tried down-grading the kernel to 6.12.9+bpo though installing it and changing /boot/extlinux/extlinux.conf caused it not to boot.
3. Additionally though, extlinux.conf had the wrong board type. It was `/boot/rk3566-soquartz-cm4.dtb` but I updated it to `/boot/rk3566-soquartz-blade.dtb` as that is what I am using. As I recall before the front USB didn't work but now it does so that's something to note.
4. Finally, I tried building and installing an updated u-boot via [these](https://pine64.org/documentation/Quartz64/Software/Building_U-Boot/) instructions and placing it on the boot sector (via dd) and in /boot. This did work but didn't fix the issue since it appears as though it's related to 6.12 specifically.

m00dawg, 2025-01-22:
I was able to get PCIe working again by manually adding the line noted in [this](https://lore.kernel.org/linux-rockchip/20241122073006.99309-1-amadeus@jmu.edu.cn/T/#u) patch to my /boot/rk3566-soquartz-blade.dtb file by doing:

```
dtc -I dts -O dtb ~/out > rk3566-soquartz-blade.dtb
vi out (add patch)
dtc -I dts -O dtb ~/out > rk3566-soquartz-blade.dtb
```

Credit to one of the kind folks on the PINE64 Matrix channel (not sharing their name just in case they don't want me to, privacy and what not).

MichaIng, 2025-01-23:
Many thanks for the report. I'll do a Linux 6.13 package build once the LTS tag as been rolled over. Should not take long.

And right, we include the CM4 device tree for the SOQuartz only, I remember it was most compatible. But makes sense that we add as well others: there is as well a SOQuartz Model A device tree, the one I actually have.

We could try to auto-detect the carrier board on first boot, and adjust the device tree automatically. Do you have an idea how to identify the Blade? I'll try to find one for the Model A carrier board.

AnonymousMonkeyNo12, 2025-02-06:
@MichaIng 

I am running a soquartz cm4 on a blade.

Also having an issue with the NVME not working.

If it helps somehow, I can get into a call and we can try to see how to detect it.

yukijoou, 2025-02-08:
hii,

i just ran into the same issue with my PINE64 QUARTZ64 A, and applying the DTB patch fixed it for me too!

here are detailed instructions of the steps i followed, in case somebody else has the same issue
1. make a copy of the currently used dtb `cp /boot/rk3566-quartz64-a.dtb .`
2. convert it to an editable dts file `dtc -I dtb -O dts rk3566-quartz64-a.dtb > rk3566-quartz64-a.dts` (you may need to install the `device-tree-compiler` package first!)
3. patch the dts file: i opened `rk3566-quartz64-a.dts` in vim and searched for `compatible = "rockchip,rk3568-naneng`. then, in the 2 sections with matching text, i inserted `reset-names = "phy";` under the `resets =…` instruction.
4. compile the patched dts into a dtb file that can be loaded by the kernel: `dtc -I dts -O dtb rk3566-quartz64-a.dts > rk3566-quartz64-a-patched.dts`
5. put it in the right place: `sudo cp rk3566-quartz64-a-patched.dts /boot/`
6. edit `/boot/extlinux/extlinux.conf` to use this file: on the `fdt /boot/…` line, replace the filepath by `/boot/rk3566-quartz64-a-patched.dts`
7. reboot your system :)

i hope this gets fixed soon!

yukijoou, 2025-02-08:
also, for SEO reasons and to make this fix more findable:

i would like to point out this applies to **all arm linux distros running older kernels on rockchip-based devices**!

i had the same issue on plebian linux before trying out diet pi..! quite bothersome such a major issue got into a production kernel.

closed 2025-02-23 by MichaIng