MichaIng/DietPi #7374: DietPi + Home Assistant = SQLite Version

HebeROF, 2025-01-26:
Hey there,

for now I have no real bug or software feature request. I am running DietPi 9.9.0.

In addition to Adguard, Unbound and other things, I mainly use it for Home Assistant.

Home Assistant version 2025.1.4 also runs smoothly for me. However, I have had an entry in the notifications for a few months now. 

>Support for SQLite version 3.34.1 will end soon; the minimum supported version is 3.40.1. Please update your database software. This no longer works in version 2025.2.0. Please fix this before upgrading.

I have not installed SQLite via the DietPi software menu, so I suspect some kind of basic version in DietPi itself. 

At least when I call it up via the command line, it confirms that SQLite 3.34.1 is installed on my DietPi Version. 

Will there be an update from DietPi itself, or do I have to install an SQLite version manually? The 2025.2 version of Home Assistant is coming soon and I don't want to fall into the trap of suddenly not having a functioning Home Assistant instance. 

Many thanks for the answer.

MichaIng, 2025-01-28:
Thanks for the info. SQLite 3.40.1 is available from Debian Bookworm on, so this limitation applies to Bullseye systems only: https://packages.debian.org/sqlite3

Upgrading to Bookworm is quite easy on DietPi, as we created a script for that: https://dietpi.com/blog/?p=3128

Probably we need to hardcode the HA version on Bullseye systems then 🤔. I hope the web UI prevents an update if no compatible SQLite is available.

HebeROF, 2025-01-28:
Thank you very much for the quick response and feedback. 
After I had some problems with my backup, the script ran cleanly without a single error.

Many thanks for that as well. 

The note is now no longer displayed in HA. It also feels like the whole system is running faster.

MichaIng, 2025-02-09:
Fixed with: https://github.com/MichaIng/DietPi/commit/23bb6af

closed 2025-02-09 by MichaIng