MichaIng/DietPi #7375: Domoticz is not installing. Invalid path.

IgrekLg, 2025-01-28:
### Creating a bug report/issue
- [+] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye
- Kernel version | `uname -a`
Linux DietPi 6.6.63-current-rockchip64 #2 SMP PREEMPT Fri Nov 22 14:38:37 UTC 2024 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
ROCK Pi S (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
5V 3A
- SD card used | (EG: SanDisk ultra)
micro SDXC 64Gb Kingston Canvas Go Plus UHS-I U3 A2 + ADP

#### Additional Information (if applicable)
- Software title | domoticz

Domoticz will not install.
Error in box: 
Checking URL: https://releases.domoticz.com/releases/release/domoticz_linux_aarch64.tgz
                   │  - Command: curl -ILfvm 10 https://releases.domoticz.com/releases/release/domoticz_linux_aarch64.tgz
                   │  - Exit code: 22
                   │  - DietPi version: v9.9.0 (MichaIng/master) | HW_MODEL: 73 | HW_ARCH: 3 | DISTRO: 6

Indeed 
The path 'https://releases.domoticz.com/releases/release/domoticz_linux_aarch64.tgz' is vrong!

True path is 'https://releases.domoticz.com/release/domoticz_linux_aarch64.tgz' 
One of the '/releases/ ' is obviously redundant!

P.S..
I have already downloaded domoticz_linux_aarch64.tgz file to my PC, how can I skip the step of downloading it with wrong path?

MichaIng, 2025-01-28:
Many thanks for your report. Indeed the URL has changed recently, fixed for next release: https://github.com/MichaIng/DietPi/commit/4c5457c

To get it working in your case, easiest is to select "Change command" from the error prompt, and remove the `/releases` there. It is downloaded to and expected at a temporary directory `/tmp/DietPi-Software`, so your pre-downloaded archive is at the wrong location.

MichaIng, 2025-01-28:
I just wanted to check back the situation with libssl: https://github.com/domoticz/domoticz/issues/5233

However, that should be actually an issue on Bookworm and above with ARM architecture only. But you are on Bullseye, which should have libssl1.1 OOTB. Can you share this:
```sh
dpkg -l | grep libssl
```

IgrekLg, 2025-01-29:
> dpkg -l | grep libssl

ii  libssl3:arm64                  3.0.15-1~deb12u1               arm64        Secure Sockets Layer toolkit - shared libraries

MichaIng, 2025-01-30:
> I've already completely upgraded to the latest system from the website.

What do you mean by this? `dietpi-update` won't upgrade the underlying Debian version. In the bug report template it says "bullseye", but the `libssl3` package is from Bookworm. So something is mixed. Did you upgrade to Bookworm in the meantime, using this guide (or manually)? https://dietpi.com/blog/?p=3128

Else, can you show this:
```sh
cat /etc/debian_version /etc/os-release /boot/dietpi/.version
```

IgrekLg, 2025-01-30:
Hello!
I updated manually. Having failed to update the system with updates, I downloaded the distribution from the site and unmatted the CD and loaded the latest distribution. Thus repeating all the steps. 
The result of the command **cat /etc/debian_version /etc/os-release /boot/dietpi/.version**

> PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
> NAME="Debian GNU/Linux"
> VERSION_ID="12"
> VERSION="12 (bookworm)"
> VERSION_CODENAME=bookworm
> ID=debian
> HOME_URL="https://www.debian.org/"
> SUPPORT_URL="https://www.debian.org/support"
> BUG_REPORT_URL="https://bugs.debian.org/"
> G_DIETPI_VERSION_CORE=9
> G_DIETPI_VERSION_SUB=9
> G_DIETPI_VERSION_RC=0
> G_GITBRANCH='master'
> G_GITOWNER='MichaIng'
> G_LIVE_PATCH_STATUS[0]='applied'
> G_LIVE_PATCH_STATUS[1]='applied'
> G_LIVE_PATCH_STATUS[2]='not applicable'
> G_LIVE_PATCH_STATUS[3]='not applicable'
>

MichaIng, 2025-01-31:
Sorry one essential file was missing:
```sh
cat /boot/dietpi/.hw_model
```
So you are definitely using a Bookworm system, where the Domoticz software option should not be available, because of this very issue: For ARM, Domoticz provides only binaries compiled against libssl1.1, which has been replaced with libssl3 on Bookworm.

Only solution is to compile own Domoticz binaries, with I probably find time for with DietPi v9.11 release (if no one else can pick it up in the meaning).

IgrekLg, 2025-01-31:
```
G_HW_MODEL=73
G_HW_MODEL_NAME='ROCK Pi S (aarch64)'
G_HW_ARCH=3
G_HW_ARCH_NAME='aarch64'
G_HW_CPUID=0
G_HW_CPU_CORES=4
G_DISTRO=7
G_DISTRO_NAME='bookworm'
G_ROOTFS_DEV='/dev/mmcblk1p1'
G_HW_UUID='0459aebd-a830-40f6-aadb-ea696e59460b'
```

MichaIng, 2025-01-31:
Okay, that contains now the correct values: `G_DISTRO=7` compared to `G_DISTRO=6` in your original post. Hence Domoticz does not show up as option anymore now, right?

IgrekLg, 2025-02-04:
> Hence Domoticz does not show up as option anymore now, right?

Yes. (

closed 2025-02-09 by MichaIng