MichaIng/DietPi #7376: E: Command line option 'e' [from -eany] is not understood in combination with the other options.

keesma, 2025-01-30:
#### Details:
- Date           | Thu Jan 30 19:41:21 CET 2025
- DietPi version | v8.25.3 (MichaIng/8)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | `Linux werkkamer 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux`
- Distro         | bullseye (ID=6,RASPBIAN=1)
- Command        | `apt-get -y -eany update`
- Exit code      | 100
- Software title | DietPi-Update
- 
#### Steps to reproduce:

1. upgrade to bull's eye from buster
2. dietpi-udate
 
#### Expected behaviour:
The device should update

#### Actual behaviour:
Error message:
- ...
#### Extra details:

- ...
#### Additional logs:
```

E: Command line option 'e' [from -eany] is not understood in combination with the other options.

Full log:

dietpi-update
[  OK  ] DietPi-Update | Setting in /boot/dietpi.txt adjusted: DEV_GITBRANCH=master

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[  OK  ] DietPi-Update | Checking IPv4 network connectivity
[  OK  ] DietPi-Update | Checking DNS resolver
[ INFO ] DietPi-Update | Getting latest version from: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid latest version: 9.9.0
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v8.25.3 (branch: MichaIng/8)
[ INFO ] DietPi-Update | Latest version  : v9.9.0 (branch: MichaIng/master)

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=1979 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop 
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : docker
[  OK  ] DietPi-Services | stop : squeezelite
[  OK  ] DietPi-Services | stop : shairport-sync
[  OK  ] DietPi-Services | stop : avahi-daemon

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[ INFO ] DietPi-Pre-patches | Migrating from raspberrypi.org to raspberrypi.com
[  OK  ] DietPi-Pre-patches | sed --follow-symlinks -i s/raspberrypi.org/raspberrypi.com/ /etc/apt/sources.list /boot/dietpi.txt
[  OK  ] DietPi-Pre-patches | sed --follow-symlinks -i s/raspberrypi.org/raspberrypi.com/ /etc/apt/sources.list.d/docker.list /etc/apt/sources.list.d/raspi.list
[  OK  ] DietPi-Pre-patches | mkdir -p ../DietPi-Update/DietPi-master/dietpi
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
E: Command line option 'e' [from -eany] is not understood in combination with the other options.
[FAILED] DietPi-Update | APT update
 - Command: apt-get -y -eany update
[ INFO ] DietPi-Update | APT update, please wait...
E: Command line option 'e' [from -eany] is not understood in combination with the other options.
[FAILED] DietPi-Update | APT update
 - Command: apt-get -y -eany update
```

MichaIng, 2025-01-30:
Hmm, as it the `apt` version is still old. Can you show the output of this:
```sh
dpkg -l apt
apt -v
```

keesma, 2025-01-31:
~# dpkg -l apt
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  apt            1.8.2.3      armhf        commandline package manager

~# apt -v
apt 1.8.2.3 (armhf)

MichaIng, 2025-01-31:
This is still the version from Buster, not Bullseye. Please repeat the package upgrades, and be sure that they finish without error and do not abort in the middle:
```sh
apt update
apt full-upgrade
```

keesma, 2025-01-31:
I did :
apt install apt to update to the newer apt version

then
apt update
apt full-upgrade

and then dietpi-update. 
Everything seems to work again.

@MichaIng Thank you for your support!

MichaIng, 2025-01-31:
Okay great. If all your software and services are working well, consider to do the next step right away and upgrade to the current stable Debian Bookworm: https://dietpi.com/blog/?p=3128
It is somewhat easier, since we put all these manual steps into a script 🙂.

closed 2025-01-31 by MichaIng