MichaIng/DietPi #7380: Follow up on issue #4855 from 2021 re DietPi hanging on shutdown in PXE boot environment

digitalformula, 2025-02-04:
I'm running 9.9.0 on a Raspberry Pi 5 and everything works great except for shutdown.  This hangs the entire system (forever) although the following issue seems to describe the problem perfectly: https://github.com/MichaIng/DietPi/issues/4855

I haven't tested the steps there yet (will do so tonight) but the issue mentions a PR to change the ordering of various things to get around this issue.  Did that PR happen/get merged at some point?  I assume not since the issue is still there.  I can confirm shutdown when booted from SD card works as expected.

MichaIng, 2025-04-19:
Hmm it sounds like this issue, which has been however fixed 🤔: #7104

MichaIng, 2025-04-19:
Okay so generally it makes sense: deconfiguring the network breaks the PXE NFS root mount, which makes every rootfs access hang.

Search machines provide a lot of results discussing the very same issues. One contained a somewhat important hint: The system has Internet access already on early boot. Hence it does not make any sense or can even break things when trying to configure the same interface again. It hence makes sense to disable the Internet facing network adapter. `dietpi-config` calls `ifdown` explicitly, resulting in the same issue at least once, in a less optimal state. But `allow-hotplug eth0` (or `wlan0` respectively) can be commented in `/etc/network/interfaces` to achieve the same from next boot on, probably even for current session already. The `networking.service` deconfigures only interfaces defined like `auto eth0` or `allow-hotplug eth0` in `/etc/network/interfaces`.

We could detect an NFS root mount in `dietpi-config` and in that case skip the problematic line for the network adapter attached to the default route. When trying to configure it, we could show a notice instead that this network must be configured via initramfs instead.

MichaIng, 2026-09-12:
@numbersixnz 
Does it work to comment out all `allow-hotplug` lines in `/etc/network/interfaces` (and reboot)? It seems logic to me that this is correct when using PXE boot, to leave network connectivity, established by the bootloader, untouched. But not sure whether this practically works, e.g. whether/how e.g. routes are taken over, and sure sure there is an issue when DHCP was used, since no DHCP client would renew the lease.

If something needs to setup network again, then a hack would need to be added manually, to prevent `network.service` from attempting to bring down the interface at shutdown.

numbersixnz, 2026-09-12:
> @numbersixnz
> Does it work to comment out all `allow-hotplug` lines in `/etc/network/interfaces` (and reboot)? It seems logic to me that this is correct when using PXE boot, to leave network [...]

No idea. I posted that 2 years ago and have long since moved on.

MichaIng, 2026-09-12:
Okay, I see unsetting the `ExecStop=` in `network.service` was already the workaround years ago. This however is default `ifupdown` behavior, nothing DietPi specific. We need someone with PXE boot on a different OS where this issue does not exist, to check and see how the setup is done there: no userspace interface configuration at all, all done in bootloader and/or initramfs, or just another network stack which does not bring down interfaces at shutdown, or which has an explicit option to skip that for a particular interface.

Not sure why I kept the milestone attached, as I see nothing reasonable we can do our end: Hacking the service or auto-detecting NFS on rootfs is a workaround that should not be needed with correct setup, which most likely just must not configure the interface, that is obviously active for the NFS rootfs mount already, via `ifupdown` or in userspace at all: `AUTO_SETUP_NET_ETHERNET_ENABLED=0`

If anyone knows a proper functioning setups on a Linux distro, with additional userspace network configuration for the same interface, let me know. Otherwise, I assume `AUTO_SETUP_NET_ETHERNET_ENABLED=0` is the correct starting point.

closed 2026-09-12 by MichaIng