MichaIng/DietPi #7381: soquartz image dist-upgrade no space left

AnonymousMonkeyNo12, 2025-02-05:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'
```

```
G_HW_MODEL=49
G_HW_MODEL_NAME='Quartz64 (aarch64)'
G_HW_ARCH=3
G_HW_ARCH_NAME='aarch64'
G_HW_CPUID=9
G_HW_CPU_CORES=4
G_DISTRO=7
G_DISTRO_NAME='bookworm'
G_ROOTFS_DEV='/dev/mmcblk0p1'
G_HW_UUID='cc220d78-6406-40cb-bf76-75f341608686'
```

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
> soquartz 4GB on soquartz Blade

- Power supply used | (EG: 5V 1A RAVpower)
> OS6250 PoE

- SD card used | (EG: SanDisk ultra)
> Transcend Silver 32GB

#### Steps to reproduce

1. Pretty default dietpi.txt, no major changes except for locale, password and ssh key
2. First boot, wanted to let it autosetup, but it seemed to fail. So I have connected a display and keyboard and went through the steps it shows after logging in (had to use default pw, as it didn't set up my password yet)
3. During the dist-upgrade step it ran out of space, I assume because it didn't expand the partition to fill the microSD card yet

#### Expected behaviour

- Either don't dist-upgrade on initial setup or do it after space expansion

#### Actual behaviour

- It ran a dist-upgrade and failed with the message 
`cannot copy extracted data for '...' to '...' : failed to write (No space left on device)`

Joulinar, 2025-02-05:
Can you share `df -h`. There we should see current file system sizes

AnonymousMonkeyNo12, 2025-02-05:
I did `df -h` and the root partition size was like on the .img (nearly no free space)

I noticed one more strange thing, the 1M DIETPISETUP partition is directly after the root partition.

Then I did the following:
1. Flashed the SD card
2. Manually moved the DIETPISETUP partition to the end of the sdcard
3. booted it

Then the root partition expanded successfully.

Joulinar, 2025-02-05:
Something @MichaIng would need to have a look into

AnonymousMonkeyNo12, 2025-02-06:
Possibly related:

https://github.com/MichaIng/DietPi/issues/6838

https://github.com/MichaIng/DietPi/issues/6822

MichaIng, 2025-02-06:
Please show the log of the exansion script:
```sh
cat /var/tmp/dietpi/logs/fs_partition_resize.log
```

AnonymousMonkeyNo12, 2025-02-07:
To properly collect all the data, I re-did everything from scratch today:

1. Re-flashed currently available img
2. Mount the root partition and disable ipv6 (because otherwise update fails to ping)
```
cd /path/to/sdcard/root
cat << _EOF_ > ./etc/sysctl.d/disable-ipv6.conf
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
net.ipv6.conf.eth0.disable_ipv6 = 1
_EOF_
```
3. Copied this dietpi.txt to the DIETPISETUP partition [https://pastebin.com/sASqPBMF](https://pastebin.com/sASqPBMF)
4. This is how the sdcard looks before booting for the first time:
```
fdisk -l /dev/sdb
Disk /dev/sdb: 28.33 GiB, 30416044032 bytes, 59406336 sectors
Disk model: STORAGE DEVICE  
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xcc5a2dfa

Device     Boot   Start     End Sectors  Size Id Type
/dev/sdb1  *      32768 1572863 1540096  752M 83 Linux
/dev/sdb2       1572864 1574911    2048    1M  c W95 FAT32 (LBA)
```
```
df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/sdb1       723M  574M   98M  86% /run/media/user/7081b5aa-02f8-4c97-b66e-dcd93a504081
/dev/sdb2       988K   54K  934K   6% /run/media/user/DIETPISETUP
```
5. Put it into the soquartz blade & boot
6. Still get the dietpi-update fail
```
┌────────────────────────────────────────────────┤ DietPi-Update ├─────────────────────────────────────────────────┐
│ Checking IPv6 network connectivity                                                                               │
│  - Command: ping -6nc 1 -W 10 2620:fe::fe                                                                        │
│  - Exit code: 1                                                                                                  │
│  - DietPi version: v9.9.0 (MichaIng/master) | HW_MODEL: 49 | HW_ARCH: 3 | DISTRO: 7                              │
│  - Error log:                                                                                                    │
│ PING 2620:fe::fe(2620:fe::fe) 56 data bytes                                                                      │
│ From 2003:d2:770e:8200:7e77:16ff:fee5:7f00 icmp_seq=1 Destination unreachable: No route                          │
│                                                                                                                  │
│ --- 2620:fe::fe ping statistics ---                                                                              │
│ 1 packets transmitted, 0 received, +1 errors, 100% packet loss, time 0ms                                         │
│                                                                                                                  │
│                  Retry               : Re-run the last command that failed                                       │
│                  Double timeout      : Retry with doubled timeout for ping to wait for a reply                   │
│                  Change IPv6 address : Change the IPv6 address used for this test                                │
│                  Disable IPv6        : Disable IPv6 on your system and continue                                  │
│                  Network settings    : Enter dietpi-config network options                                       │
│                  DietPi-Config       : Edit network, APT/NTP mirror settings etc                                 │
│                  Open subshell       : Open a subshell to investigate or solve the issue                         │
│                  Send report         : Upload bug report including system info to DietPi                         │
│                  Print report        : Print bug report template for GitHub or forum                             │
│                                      ●─ Devs only ───────────────────────────────────────────●                   │
│                  Change command      : Adjust and rerun the command                                              │
│                                                                                                                  │
│                                                                                                                  │
│                                 <Select>                                 <Exit>                                  │
│                                                                                                                  │
└──────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘
```
7. Disable IPv6 and continue
8. Error on dist-upgrade log:[https://pastebin.com/AgNrT61E](https://pastebin.com/AgNrT61E)
9. Current state of the microsd card
[... 42 lines cut ...]
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xcc5a2dfa

Old situation:

Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk0p1 *      32768 1572863 1540096  752M 83 Linux
/dev/mmcblk0p2      1572864 1574911    2048    1M  c W95 FAT32 (LBA)

/dev/mmcblk0p1: 
New situation:
Disklabel type: dos
Disk identifier: 0xcc5a2dfa

Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk0p1 *      32768 1572863 1540096  752M 83 Linux
/dev/mmcblk0p2      1572864 1574911    2048    1M  c W95 FAT32 (LBA)

The partition table has been altered.
resize2fs 1.47.0 (5-Feb-2023)
The filesystem is already 192512 (4k) blocks long.  Nothing to do!
```

closed 2025-02-07 by MichaIng