MichaIng/DietPi #7382: dietpi-backup cronjob is not executed / created

Boebbele, 2025-02-06:
My daily backup hasn't been running for a few days now. I can run it manually under "dietpi-backup" but the cronjob is no longer executed. The reason is unknown to me. When I display all cron jobs on the console (crontab -l) there is not a single entry.
What can I do to get the daily backup running again?

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm
- Kernel version | `uname -a`
Linux homeassistant.lan 6.1.0-30-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.124-1 (2025-01-12) x86_64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Native PC (x86_64)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
dietpi-backup

- Can this issue be replicated on a fresh installation of DietPi? --> no

- Bug report ID | `echo $G_HW_UUID`
- 
b9e4afa2-3bcb-42f7-8ec4-4dcb8471ea32

MichaIng, 2025-02-06:
The ceon job is at `/etc/cron.daily/dietpi`. Can you check whether this file exists and has proper permissions?
```sh
ls -l /etc/cron.daily/dietpi
```
Also, check the daily cron job execution times via `dietpi-cron` and see whether you find the related log in `journalctl`.

Boebbele, 2025-02-06:
ls -l /etc/cron.daily/dietpi
-rwxrwxrwx 1 root root 2614 27. Dez 08:50 /etc/cron.daily/dietpi

/etc/cron.daily# ls
apt-compat  dietpi  dpkg  lighttpd  man-db  samba

#!/bin/bash
{
        #////////////////////////////////////
        # DietPi Cron.Daily script
        #
        #////////////////////////////////////
        #
        # Info:
        # - Location: /etc/cron.daily/dietpi
        #
        #////////////////////////////////////

        # Assure that whiptail dialogues are skipped, even when the script is called manually, since STDOUT is redirected
        export G_INTERACTIVE=0

        #----------------------------------------------------------------
        # Main Loop
        #----------------------------------------------------------------
        # Sync system time if mode 2 (daily) is detected
        grep -q '^[[:blank:]]*CONFIG_NTP_MODE=2' /boot/dietpi.txt && /boot/dietpi/func/run_ntpd 1
        #----------------------------------------------------------------
        # Check for DietPi and/or APT updates
        APT_UPDATE_MODE=$(sed -n '/^[[:blank:]]*CONFIG_CHECK_APT_UPDATES=/{s/^[^=]*=//p;q}' /boot/dietpi.txt)
        if grep -q '^[[:blank:]]*CONFIG_CHECK_DIETPI_UPDATES=1' /boot/dietpi.txt
        then
                /boot/dietpi/dietpi-update 2
                [[ $APT_UPDATE_MODE == 2 && -f '/run/dietpi/.update_available' ]] && { . /boot/dietpi/func/dietpi-globals; G_AGUP -f; }

        elif [[ $APT_UPDATE_MODE != 0 ]]
        then
                . /boot/dietpi/func/dietpi-globals
                G_AGUP -f
        fi
        # Apply APT upgrades if chosen and available
        [[ $APT_UPDATE_MODE == 2 && -f '/run/dietpi/.apt_updates' ]] && apt-get -qq upgrade &> /var/tmp/dietpi/logs/dietpi-upgrade_apt.log
        #----------------------------------------------------------------
        # Sync DietPi-RAMlog to disk: https://github.com/MichaIng/DietPi/issues/4884
        systemctl -q is-enabled dietpi-ramlog 2> /dev/null && /boot/dietpi/func/dietpi-ramlog 1
        #----------------------------------------------------------------
        # Refresh DietPi MOTD
        if [[ ! -f '/boot/dietpi/.dietpi-banner' ]] || grep -q '^[[:blank:]]*aENABLED\[12\]=1' /boot/dietpi/.dietpi-banner
        then
                [[ -f '/run/dietpi/.dietpi_motd' ]] && rm /run/dietpi/.dietpi_motd
                curl -sSfL 'https://dietpi.com/motd' -o /run/dietpi/.dietpi_motd
        fi
        #----------------------------------------------------------------
        # DietPi-Backup daily
        if [[ -f '/boot/dietpi/.dietpi-backup_settings' ]] && grep -q '^[[:blank:]]*DAILY_BACKUP=1' /boot/dietpi/.dietpi-backup_settings
        then
                /boot/dietpi/dietpi-backup 1
        fi
        #----------------------------------------------------------------
        # DietPi-Sync daily
        if [[ -f '/boot/dietpi/.dietpi-sync_settings' ]] && grep -q '^[[:blank:]]*SYNC_CRONDAILY=1' /boot/dietpi/.dietpi-sync_settings
        then
                /boot/dietpi/dietpi-sync 1
        fi
        #----------------------------------------------------------------
        exit
        #----------------------------------------------------------------
} > /dev/null

MichaIng, 2025-02-06:
Looks good. The 777 mode if the script is not good. Did you change it in attempt to fix the issue? Please adjust it so that not every 3rd party software an user can edit a script that is executed as root user:
```sh
chmod +x 755 /etc/cron.daily/dietpi
```
Ah, maybe this is even the reason cron denies to execute it. Do the other scripts in this dir have 777 (`rwxrwxrwx`) mode as well?

Boebbele, 2025-02-06:
ls -l /etc/passwd
-rw-rw-r-- 1 root root 1627 27. Jan 17:06 /etc/passwd

ls -l fstab
-rwxrwxrwx 1 root root 1276 30. Jan 09:55 fstab

ls -l host.conf
-rwxrwxrwx 1 root root 9  7. Aug 2006  host.conf

ls -l /etc/network/interfaces
-rwxrwxrwx 1 root root 680 30. Jan 09:59 /etc/network/interfaces

I think it is all rwxrwxrwx .... only passwd is rwrwr

MichaIng, 2025-02-06:
Yeah that does not look good. Hmm, that is very difficult or time consuming to repair. Let me think ... so for those file coming from packages, reinstalling said packages restores the intended permissions. But much `/etc` and `/var` content is not part of the package lists, but created by package maintainer scripts and similar.

One quite step that should be mostly safe and addresses the biggest security issue:
```sh
chmod -R 'o-w' /etc
```
This removes global write permissions from `/etc` content.

There are some ideas, but the only better solution there requires a functional backup, and the other answers are bullshit: https://superuser.com/questions/1252600/fix-permissions-of-server-after-accidental-chmod-debian

You said you clone the drive with Clonezilla. Do you still have the original one to check whether permissions are still correct there, and to potentially restore permissions from there? It certainly makes sense to find out where this is coming from.

Boebbele, 2025-02-06:
Can you please help me? Do I have to change the file permissions of every system folder now?

MichaIng, 2025-02-06:
> Do I have to change the file permissions of every system folder now?

Theoretically yes, if there is not backup/the original drive does not exist anymore. Though at least it can be limited to `/etc` `/var` `/usr/local` and `/mnt`, if you reinstall _all_ APT packages. But see my questions above, let's try to find out some more first.

Another probably safer option is to go with a fresh system, and move over only the needed configs and data from software you use/configured. For those, in most cases, even `dietpi-software` adjusts the permissions on installs, taking over whichever configs and data is present.

Boebbele, 2025-02-06:
No, I don't have the old disk anymore. But I still have a dietpi backup on an external disk.
Can I check whether it was like this before?

How could this even happen?

I remember that a few days before mirroring with clonezilla I no longer had access via SSH. So I uninstalled openssh and reinstalled it. After that everything worked again. 

the computer only runs on the local LAN. No access from outside. except with tailscale

Boebbele, 2025-02-06:
Okay...on my old Backup ther is /etc/cron.daily/dietpi:

ingo@voidlinux /r/m/i/8/d/d/e/cron.daily> ls -l dietpi
-rwxr-xr-x 1 root root 2614 27. Dez 08:50 dietpi*

What can I do with this backup-files?

MichaIng, 2025-02-06:
> It actually seems to have happened when mirroring with Clonezilla. I can't think of any other explanation.

It seems so, but strange as Clonezilla is a Linux tool, and AFAIK does a bit to bit clone. Surprising if it really restored files that way. And also weird that not all permissions are identical.

I would then indeed restore the backup. Do you have `/mnt/dietpi_userdata` on an external drive? If not, exclude it in `dietpi-backup` with the Filter option: Comment or remove the `+ /mnt/dietpi_userdata/` line. And was Docker installed via `dietpi-software`, respectively do you know where the configs and volumes for the Docker images are located? So you can backup them as well, of they are not in `/mnt/dietpi_userdata`. Easiest is probably to create another `dietpi-backup` to another drive first, then restore the old one, and then restore the needed data and configs from the newer backup.

Boebbele, 2025-02-07:
Thank you very much for your help. I did it the way you wrote it. Fortunately, I still had the backup on an external plate .... Puhhh ...

It actually worked. The file rights should now be correct:

ls -l /etc/cron.daily/dietpi
-rwxr-xr-x 1 root root 2614 27. Dez 08:50 /etc/cron.daily/dietpi

/etc/fstab
-rw-r--r-- 1 root root 1283 26. Nov 2023  /etc/fstab

ls -l /etc/sudo.conf
-rw-r--r-- 1 root root 4343 27. Jun 2023  /etc/sudo.conf

/etc/host.conf
-rw-r--r-- 1 root root 9  7. Aug 2006  /etc/host.conf

Is this okay?

I found that /etc/network/interfaces is stil rwxrwxrwx:
/etc/network/interfaces
-rwxrwxrwx 1 root root 680 30. Jan 09:59 /etc/network/interfaces

Do I have to change that?

Can you give me some important folders or files that I should still test?

Tonight I will see whether the daily Dietpi-Backup works again.

MichaIng, 2025-02-07:
> I found that /etc/network/interfaces is stil rwxrwxrwx:

Yeah, please change that to 644.

> Oder vielleicht hast du ne Idee wie ich mir per script alle Dateien aus / zeigen lassen kann die nicht koscher sind und die ich mit den alten backup Dateien vergleichen sollte?

Use this command, which searches for all files and directories within the root filesystem (it does not iterate through mounts) which are writable by "others":
```sh
find / -mount ! -type l -perm -o=w
```
It returns this on my system:
```console
root@micha:~# find / -mount ! -type l -perm -o=w
/var/lib/php/sessions
/var/tmp
/var/tmp/systemd-private-7f2c30b9a55b480980d408d5edcf35cf-redis-server.service-46XG0R/tmp
/var/tmp/systemd-private-7f2c30b9a55b480980d408d5edcf35cf-coturn.service-teXjhW/tmp
/var/tmp/systemd-private-7f2c30b9a55b480980d408d5edcf35cf-apache2.service-JbZixG/tmp
/tmp
```

> Die Datei aus dem backup Ordner hat -rw-r--r-- 1 root root. Sie hat sich beim kopieren also verändert. Aber warum?

Okay, that is very weird. Can you check the log file `dietpi-backup.log` in the backup directory? Since it contains a list of all files, it is probably too sensitive to share here. You can send it to me via email, if you want me to have a look. Which filesystem type is the backup?

Boebbele, 2025-02-07:
OK. I thank you! How can I exclude the /home directory and /mnt/userdata from the search?
Otherwise I get a long list of docker containers and .Cache folder from home assistant etc.

Boebbele, 2025-02-08:
Problem solved. The file permissions  Were not korrekt. I restored a backup. Now it works again. Many Thanks to Micha!

closed 2025-02-08 by Boebbele