MichaIng/DietPi #7383: Machine won't reboot

nicolasestrem, 2025-02-07:
### Creating a bug report/issue
- [ X] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`

G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='applied'
G_LIVE_PATCH_STATUS[2]='not applicable'
G_LIVE_PATCH_STATUS[3]='not applicable'

- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bookworm

- Kernel version | `uname -a`
Linux PCBOX 6.1.0-30-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.124-1 (2025-01-12) x86_64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Native PC (x86_64)

#### Additional Information (if applicable)
Bug Report ID
e29009ea-9295-4af5-8b99-774d9389e160

#### Steps to reproduce
run command `sudo reboot` or `sudo systemctl reboot`

#### Expected behaviour
Reboot

#### Actual behaviour
System not rebooting, machine remains on and cannot be reached

#### Extra details
I've got this issue for the first time yesterday. I've performed a new clean install this morning because of this issue as well as caddy not being able to run as a service. 
After reinstalling my usual dietpi-software + caddy the problem reappeared. I suspect JellyFin to be the root of the problem, just gut feeling.

MichaIng, 2025-02-07:
Most likely a hanging service indeed, though normally they should be SIGKILLed by systemd after 90 seconds. Any last output on the screen?

The CIFS mount throws errors constantly:
```
CIFS: __readahead_batch() returned 4/16
```

- https://bbs.archlinux.org/viewtopic.php?id=284974
- https://www.truenas.com/community/threads/risk-of-boot-pool-overrun-samba-spamming-kernel-log.114996/

Does this show up only while you are transferring data on the CIFS share? Maybe adding `cache=none` to the mount options in `/etc/fstab` mutes this.

Also Jellyfin throws regular errors, not being able to send notifications:
```
Feb 07 07:50:21 PCBOX jellyfin[675]: System.InvalidOperationException: An invalid request URI was provided. Either the request URI must be an absolute URI or BaseAddress must be set.
Feb 07 07:50:21 PCBOX jellyfin[675]:    at System.Net.Http.HttpClient.PrepareRequestMessage(HttpRequestMessage request)
Feb 07 07:50:21 PCBOX jellyfin[675]:    at Jellyfin.Plugin.Webhook.Destinations.Generic.GenericClient.SendAsync(GenericOption option, Dictionary`2 data)
Feb 07 07:50:21 PCBOX jellyfin[675]:    at Jellyfin.Plugin.Webhook.WebhookSender.SendNotification[T](IWebhookClient`1 webhookClient, T option, Dictionary`2 itemData, Type itemType)
Feb 07 07:50:22 PCBOX jellyfin[23645]: [139B blob data]
Feb 07 07:50:22 PCBOX jellyfin[675]: [07:50:22] [ERR] Unable to send notification
Feb 07 07:50:22 PCBOX jellyfin[675]: System.InvalidOperationException: An invalid request URI was provided. Either the request URI must be an absolute URI or BaseAddress must be set.
Feb 07 07:50:22 PCBOX jellyfin[675]:    at System.Net.Http.HttpClient.PrepareRequestMessage(HttpRequestMessage request)
Feb 07 07:50:22 PCBOX jellyfin[675]:    at Jellyfin.Plugin.Webhook.Destinations.Generic.GenericClient.SendAsync(GenericOption option, Dictionary`2 data)
Feb 07 07:50:22 PCBOX jellyfin[675]:    at Jellyfin.Plugin.Webhook.WebhookSender.SendNotification[T](IWebhookClient`1 webhookClient, T option, Dictionary`2 itemData, Type itemType)
```

So yeah, either Jellyfin or the CIFS share. Or is Jellyfin even using the CIFS share for something?

nicolasestrem, 2025-02-07:
Jellyfin even using the CIFS, indeed! I'll try your fix and will let you know. Thank you!

nicolasestrem, 2025-02-07:
Adding cache=none DID address this issue. Thanks!

closed 2025-02-07 by nicolasestrem