MichaIng/DietPi #7384: Dietpi services behavior

lupa18, 2025-02-07:
### Creating a bug report/issue
- [x ] I have searched the existing open and closed issues

#### Required Information
- DietPi version | 9.9
- Distro version | bullseye 0
- Kernel version | 6.1.21-v7+
- SBC model | RPi 3 Model B

#### Additional Information (if applicable)
- Software title | dietpi-services
- Can this issue be replicated on a fresh installation of DietPi? I guess yes

#### Steps to reproduce
**1.** If I understand correctly, is possible to disable a service preventing it runs on boot. Then, you could start and stop manually. 
 
**2.** So I disabled 4 services (not masked). In the screenshot you'll see one of those services:

[image]

**3.** Then, when trying to start, I got an error:

[image]

#### Expected behaviour
It must start when I called `sudo dietpi-services start jackett`

#### Actual behaviour
It doesn't start

Joulinar, 2025-02-07:
Can you try

`systemctl start jackett`

lupa18, 2025-02-07:
It appear it works:
```
dietpi@:~$ sudo systemctl start jackett
dietpi@:~$ sudo systemctl status jackett
● jackett.service - Jackett (DietPi)
     Loaded: loaded (/etc/systemd/system/jackett.service; disabled; vendor preset: enabled)
     Active: active (running) since Fri 2025-02-07 14:50:43 -03; 10s ago
   Main PID: 8289 (dash)
      Tasks: 10 (limit: 2176)
        CPU: 9.570s
     CGroup: /system.slice/jackett.service
             ├─8289 /bin/dash -c /opt/jackett/jackett --NoRestart; while pgrep -u jackett JackettUpdater > /dev/null; do sleep 1; done
             └─8290 /opt/jackett/jackett --NoRestart
```

Joulinar, 2025-02-08:
It might be ignored by our tooling as the service has been disabled using `dietpi-services`. Therefore you need to use regular `systemctl` commands to control the service.

lupa18, 2025-02-08:
Is it the desired behavior? In that case from DP Tooling point of view, disable and mask will be the same?

Joulinar, 2025-02-09:
No, there is a slight difference

If you set a service with `dietpi-services` to `disabled`, the service will no longer be started by our tools or during boot. However, it is still possible to use the standard Debian commands to control it. I would recommend to `exclude` the service as well, otherwise it might be stopped during a backup.

`masking` a service prevents its usage entirely

MichaIng, 2025-02-09:
Indeed we changed the behavior recently: #7302

But hmm, actually the intention was to skip them only if no service was explicitly given as input. Let me see whether we use it for individual services anywhere in the code, else I'll change that so that:
```sh
dietpi-services start
```
won't start disabled services, but
```sh
dietpi-services start jackett
```
strats Jackett even if disabled.

However, it remains true stat using `dietpi-services` to start/stop/enable/disable/mask/unmask individual services is obsolete and `systemctl` should be used instead.

MichaIng, 2025-02-09:
Done with: https://github.com/MichaIng/DietPi/commit/3667d23

closed 2025-02-09 by MichaIng