MichaIng/DietPi #7386: Pi5, original active cooler is not run after startup

cankirca, 2025-02-08:
Hello,

Got RP 5, original case and active cooller.
1. downloaded DietPi from the website for RP5 and imaged a new SD card.
2. Restored my previous RP4 DietPi backup from USB drive.
*All apps running as expected, however temprature is about 55°C right now, Fan is not running, system says warm but safe.
*Fan runs at startup for few seconds and stops.
*I didn't do for my previous Pi400 before, it was working best all the time.
How can I set my fan settings to keep my RP5 cooller?
Thank you.

Joulinar, 2025-02-09:
Unfortunately, you have failed to fill in the troubleshooting template. Therefore we do not know which kernel is currently active. I assume that the restore of an RPi4 backup does not use the 2712-rpi-5 kernel.

Please go to `dietpi-config` > `4 : Advanced Options` > `RPi kernel choice` and switch to the dedicated RPi5 kernel

cankirca, 2025-02-09:
Hi @Joulinar, I thought that kernel is correct, here the output when I choose Rpi Kernel from the advanced menu:
"DietPi-Set_software
                             Based on your currently used kernel variant, the following package will be kept/installed:                              
                             - linux-image-rpi-2712 for Raspberry Pi 5 only                                                                          
                             Do you want to keep/install the following additional kernel package, to boot this image from other Raspberry Pi models?                                                                                                                 
                             - linux-image-rpi-v8 for Raspberry Pi 2-5                                                                               
                                                               <Yes>                                      <No"
I choose no.
Is that correct?
Also advanced/Update RPi EEPROM firmware : [Bootloader: 1737505011 | VL805: N/A]

MichaIng, 2025-02-09:
So it is the RPi 5 kernel already. Yes you could uninstall the v8 kernel.

To enable the fan:
```sh
G_CONFIG_INJECT 'dtparam=cooling_fan=' 'dtparam=cooling_fan=on' /boot/config.txt
dtparam cooling_fan=on
```
Not sure whether the 2nd command enables it directly, or whether a reboot is requried.

Joulinar, 2025-02-09:
If I am not mistaken, the fan control should already be active via the default settings in the firmware.

MichaIng, 2025-02-09:
Correct: `/boot/firmware/overlays/README`
```
        cooling_fan             Enables the Pi 5 cooling fan (enabled
                                automatically by the firmware)
```
Hence nothing is required. It should turn on at 50°C IIRC.

Here is how to change the trip points permanently: https://github.com/MichaIng/DietPi/issues/6676#issuecomment-1878554679

closed 2025-07-17 by MichaIng