MichaIng/DietPi #7387: baikal installation reporting php version !>=8.1.0

BeardedWidget, 2025-02-09:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=9
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
G_LIVE_PATCH_STATUS[1]='not applicable'
G_LIVE_PATCH_STATUS[2]='applied'
G_LIVE_PATCH_STATUS[3]='not applicable'
- Distro version | `echo $G_DISTRO_NAME $G_RASPBIAN`
bullseye
- Kernel version | `uname -a`
Linux dietpi 6.6.56-current-rockchip64 #1 SMP PREEMPT Thu Oct 10 10:50:06 UTC 2024 aarch64 GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
ROCKPro64 (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
MFR supplied
- SD card used | (EG: SanDisk ultra)
SD SD32G

#### Additional Information (if applicable)
- Software title | baikal-0.10.1
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`
46483304-4ec7-4390-bd1c-1b2f62450c2d

#### Steps to reproduce

1. install Baikal using `dietpi-software`
2. edit `/etc/nginx/sites-dietpi/dietpi-baikal.conf` to reflect correct listen settings (e.g. [::]:8111)
3. access baikal page in accordance with installation instruction at:
    http://<server>:8111/baikal/html/index.php
4. encounter the following message:
    Composer detected issues in your platform: Your Composer dependencies require a PHP version ">= 8.1.0".

#### Expected behaviour

- `dietpi-software` should be installing a PHP version compatible with the package being installed

#### Actual behaviour

- `dietpi-software` seems to be installing php7.4 which is not ">= 8.1.0"

#### Extra details

- attempted to update OS PHP to 8.4 but baikal seems to still be using the older version
- attempted to use `composer install` using the `composer.json` supplied with the package but install fails

Joulinar, 2025-02-09:
Yes because you are running old Debian Bullseye. PHP8 will be installed on Debian Bookworm.

BeardedWidget, 2025-02-09:
Thankyethankye!  Didn't realize there was a DietPi upgrade available for [bullseye to bookworm](https://dietpi.com/blog/?p=3128#upgrade); running it now. 🤞

I'll report back with success status once it's complete.

MichaIng, 2025-02-09:
But good to know that BaiKal v10 requires PHP 8.1 or higher. So on Bullseye we need to install v9: https://github.com/sabre-io/Baikal/releases/tag/0.10.0

MichaIng, 2025-02-09:
Fixed with: https://github.com/MichaIng/DietPi/commit/7d81f38

closed 2025-02-09 by MichaIng