MichaIng/DietPi #7390: DietPi-Display | New tool to control console display modes and rotation (beta)

MichaIng, 2025-02-09:
To allow controlling console display modes, rotation and more in a generic way on every modern hardware, we added a new script which uses the KMS/DRM sysfs and `video` kernel command-line parameter to do so. Both is not perfectly documented, and there are open questions and issues, hence for now it is in beta phase.

General `video` parameter documentation: https://docs.kernel.org/fb/modedb.html
We can further add reflection and refresh rate, probably other options.

Also, we plan to shift most options from the `dietpi-config` display menu into this new script. The RPi camera module options will be probably moved elsewhere, some others are obsolete with the new kernel and will be removed. However, for now the `dietpi-config` menu remains untouched.

Issues/Questions:
- `/sys/class/drm/*/modes` theoretically contains a list of supported modes. However, at least with my HDMI screen, if I select a smaller resolution, all larger resolutions (respectively all above in the list) are hidden. So changing back to a larger resolution only works when either resetting the settings (removing the kernel command-line parameter) in between, or setting it manually.
- At first I though `/sys/class/drm/*/modes` is always shows the current mode at the top, but it does not. In above mentioned case, it just cut away all above modes. So the question is how to get the current mode without using further tools like `fbset` and similar.
- At least on my NanoPi M6, Linux provides only a single framebuffer, even of the DSI display + HDMI screen are attached. Hence both screens shot the same console output, attached to the same console device `tty`. This seems to have the implication that displays cannot be rotated independently either. As fast as I use both screens, rotating the DSI LCD does not work anymore. This is restored once the HDMI display is disabled. Would be interesting to find out whether/how to enable multi-display functionality, with a dedicated framebuffer for a different `tty` on each screen. On Raspberry Pi 4 and 5 there should be two framebuffers possible, but I did not test this yet. In case, mapping console devices to DRM devices might be a nice feature to add.

InnovoDeveloper, 2025-02-10:
Just tried the dietpi-display (Just and FYI, I had to reboot after installing the update to get dietpi-display to work).  Howerver i am getting this error.  Not sure if I missed a step.

` [FAILED] No DRM device has been found                                                                                │
                                           │                                                                                                                      │
                                           │ A DRM device could not be found on your system. Please assure the display is properly attached and working, and your │
                                           │ kernel supports DRM.                                                                                                 │
                                           │                                                                                                                      │
                                           │                                                        <Ok>     `

MichaIng, 2025-02-15:
@InnovoDeveloper which SBC is it? And you say attacked panel, so I guess no HDMI screen, but a DSI LCD or similar? It is possible that at least for this display, no KMS/DRM is used, and hence it cannot be controlled via `dietpi-display` and the respective APIs.

InnovoDeveloper, 2025-02-15:
It's the NanoPi M6 with the case built in display.  https://wiki.friendlyelec.com/wiki/index.php/NanoPi_M6

MichaIng, 2025-02-16:
Oh that is interesting. On my NanoPi M6 with official case and LCD it works perfectly fine 🤔. I'll retry with a freshly flashed image, just to be sure.

MichaIng, 2025-02-16:
Okay so I flashed our current NanoPi M6 Bookworm image. It has the DSI1 display enabled OOTB, which is the one used by the case, mine was delivered with. When running `dietpi-display`, it works OOTB:
```
┌──────────────────────────┤ DietPi-Display ├───────────────────────────┐
│ DietPi-Display is currently in beta status. Please report             │
│ issues/suggestions:                                                   │
│ - https://github.com/MichaIng/DietPi/issues/7390                      │
│                                                                       │
│ Select a display to change its mode and rotation                      │
│                                                                       │
│           LCDs     Toggle LCD drivers                                 │
│                    ●─ Configure detected displays ────────●           │
│           Display  [DSI-1] Select display                             │
│           Mode     [480x800] Change screen resolution mode            │
│           Rotation [0] Change screen rotation                         │
│           Reset    Reset settings for selected display                │
│           Toggle   [enabled] Enable or disable this display           │
│                                                                       │
│                                                                       │
│                  <Select>                  <Exit>                     │
│                                                                       │
└───────────────────────────────────────────────────────────────────────┘
```
But I found the issue now if you use a prior image without the DSI1 port enabled. The respective menu to toggle LCD displays was not shown in this case, as planned. I just fixed it: https://github.com/MichaIng/DietPi/commit/48e061a

Can you try this version, please?
```sh
curl -sSfo /boot/dietpi/dietpi-display 'https://raw.githubusercontent.com/MichaIng/DietPi/48e061a/dietpi/dietpi-display'
```

InnovoDeveloper, 2025-02-16:
this worked great.  This system did start and 9.9 Master, then Dev, then back to Master then upgrade to 9.10.  The above solved it.

sweetmeats83, 2025-02-25:
I'm using the Raspberry Pi Touch Display 2. I can not get this to rotate to landscape.

MichaIng, 2025-02-25:
@sweetmeats83 
The display appears in `dietpi-display`, and you can set the rotation there, but after reboot it does not apply? Do you have another display attached and configured? I faced this issue when using multiple screens on the NanoPi M6 (DSI LCD + HDMI screen): both share the same framebuffer, hence cannot be configured separately. Not sure whether there is a way to have an individual framebuffer for each, attach them to Individual TTYs etc.

sweetmeats83, 2025-03-01:
yes this was the issue

rmscode, 2025-03-06:
> yes this was the issue

Was the issue that it didn’t apply after a reboot, that you have another display attached, or both?

sweetmeats83, 2025-03-06:
It was because I had an hdmi display attached. When I removed the hdmi the DSI display would rotate as expected.

MichaIng, 2025-03-06:
> It was because I had an hdmi display attached. When I removed the hdmi the DSI display would rotate as expected.

Okay, then it is as expected, sadly. I need to dig further into this, also whether there is a good way to find out whether two displays share the same framebuffer, and hence cannot be configured separately, and also which one of the two (if any) can be configured in this case.

Optimal would be of course to allow a HDMI with main console and a dedicated LCD for some monitoring dashboard or similar. On RPi this works, but no idea right now how to achieve it on e.g. the NanoPi M6.

adshrc, 2025-03-11:
I'm using the Native PC Image and I just tried (out of curiosity) to disable the screen. It is indeed disabled now, but I cannot enable it anymore as i get "[FAILED] No DRM device has been found".

@MichaIng is there a quick fix to get the screen enabled again?

MichaIng, 2025-03-12:
Oh that is a bug indeed, thanks for reporting. I tested it with two screens for switching between them. I will add all displays with existing configs to the detected list, even if their DRM devices do not exist. So they can be re-enabled.

adshrc, 2025-03-21:
@MichaIng I can't use `dietpi-display -r 90 apply` in an automation, because it asks me to reboot. Can we skip the reboot modal if using this command with args? Or add another argument , like --quiet? That would be great!

rmscode, 2025-05-15:
Hey there. I have a RPi 5 running the latest DietPi image (9.12.1) and `dietpi-display` does not work to set the resolution. It is attached to a cheap Amazon Fire TV (TCL 55Q650F). The TV is 4K and `dietpi-display` utility _does_ detect the appropriate resolution, but it continues to output 1080p after a reboot. I have actually [posted](https://dietpi.com/forum/t/raspberry-pi-5-will-not-ignore-edid/22556) about my issues with this TV on the forums. Anyway, I have resorted to adding `export DISPLAY=:0` and `xrandr --output HDMI-1 --mode 3840x2160` to `.bashrc` in order to get the correct resolution.

MichaIng, 2025-05-15:
I guess you mean the resolution in X11 apps/desktop? That is normal, as X11 handles resolutions on its own. Use the desktop settings to adjust it. `dietpi-display` sets only the plain console resolution, for before the desktop/X11 app shows up, or when you do not have any GUI installed.

rmscode, 2025-05-15:
_Ohhh_...heh.

MichaIng, 2025-05-25:
Manually disabled displays, or such which are disconnected, but settings changed, do now show up in `dietpi-display` as well: https://github.com/MichaIng/DietPi/commit/a1a140d

Since disabled displays usually show up as disconnected and without any ability to read parameters from sysfs (which totally makes sense), they were also not found by `dietpi-display` anymore after a reboot. This is of course annoying when one wants to re-enable it. Hence cmdline configs are now scanned as well for video (display) entries.

adshrc, 2025-06-27:
@MichaIng on my custom arm64 VM, I'm getting "[FAILED] Boot configuration not found" when using `dietpi-display`. After looking through the code, i found this:

https://github.com/MichaIng/DietPi/blob/f9b4fb07d2328780acba1cc54164a346283095f6/dietpi/dietpi-display#L192-L196

I can't see a reason why grub should only be available on x86_64 arch. After adding `|| $G_HW_ARCH == 3` to the condition it works perfectly, all settings are getting applied. Maybe we should use a condition like `if ls /boot/grub* 1>/dev/null 2>&1`?

MichaIng, 2025-06-27:
Thanks for the hint, makes sense. I adjusted it a way that it checks for GRUB configs regardless which architecture, if other configs are not found. And when applying settings, this is done to all existing (and supported) config files, and to GRUB if the `update-grub` command exists, which should be a good measure, also since we use this command: https://github.com/MichaIng/DietPi/commit/68860c4

closed 2025-09-20 by MichaIng