MichaIng/DietPi #7396: Problem CPU GOVERNOR

Seb-tech77, 2025-02-10:
#### Required Information
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
bookworm
Linux DietPi 6.1.84-vendor-rk35xx
Orange Pi 5 Pro (aarch64)

#### Steps to reproduce
launch dietpi-config
go to the performance options tab
Select governor in performance mode
restart oranpi 5 pro 

#### Expected behaviour
on reboot the governor mode should remain in performance mode

#### Actual behaviour
on restart, the governor mode reverts to shedutil

MichaIng, 2025-02-16:
Can you please show the output of these commands:
```sh
grep 'CONFIG_CPU_GOVERNOR' /boot/dietpi.txt
journalctl -u dietpi-preboot
```

Seb-tech77, 2025-02-16:
Hello, here's the command output, I've added the cpu command 

```
Sebastien@DietPi:~$ sudo grep 'CONFIG_CPU_GOVERNOR' /boot/dietpi.txt
[sudo] Mot de passe de Sebastien :
CONFIG_CPU_GOVERNOR=performance
Sebastien@DietPi:~$ cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     aarch64
 Temperature  |     24 °C / 75 °F : Who put me in the freezer!
 Governor     |     schedutil

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1608 MHz      1200 MHz    1800 MHz
 CPU1         |      1608 MHz      1200 MHz    1800 MHz
 CPU2         |      1608 MHz      1200 MHz    1800 MHz
 CPU3         |      1608 MHz      1200 MHz    1800 MHz
 CPU4         |      408 MHz      408 MHz    2256 MHz
 CPU5         |      408 MHz      408 MHz    2256 MHz
 CPU6         |      1608 MHz      408 MHz    2304 MHz
 CPU7         |      1608 MHz      408 MHz    2304 MHz

[ INFO ] DietPi-CPU_info | The current CPU frequency may be affected by processing this script itself.

Sebastien@DietPi:~$ sudo journalctl -u dietpi-preboot
-- No entries --
Sebastien@DietPi:~$
```

MichaIng, 2025-02-16:
Is `dietpi-preboot` actually enabled?
```sh
sudo systemctl status dietpi-preboot
```
Else please enable it:
```sh
sudo systemctl enable dietpi-preboot
```
And either way, do a reboot, to check again. It should then show up in the journal output as well:
```sh
sudo journalctl -u dietpi-preboot
```

Seb-tech77, 2025-02-16:
here's the result of the commands, strange we'd say dietpi-preboot runs normally 

```
Sebastien@DietPi:~$ sudo systemctl status dietpi-preboot
[sudo] Mot de passe de Sebastien :
● dietpi-preboot.service - DietPi-PreBoot
     Loaded: loaded (/etc/systemd/system/dietpi-preboot.service; enabled; preset: enabled)
     Active: active (exited) since Sun 2025-02-16 21:50:47 CET; 17s ago
    Process: 1155 ExecStart=/boot/dietpi/preboot (code=exited, status=0/SUCCESS)
   Main PID: 1155 (code=exited, status=0/SUCCESS)
        CPU: 163ms

févr. 16 21:50:46 DietPi systemd[1]: Starting dietpi-preboot.service - DietPi-PreBoot...
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]:  DietPi-CPU_set
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]: ─────────────────────────────────────────────────────
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]:  Mode: Applying CPU governor settings: performance
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]: [ INFO ] DietPi-CPU_set | Setting CPU frequency limits : Max = Disabled MHz | Min = Disabled MHz
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]: [  OK  ] DietPi-CPU_set | Applied CPU governor settings: performance
févr. 16 21:50:47 DietPi systemd[1]: Finished dietpi-preboot.service - DietPi-PreBoot.
Sebastien@DietPi:~$ cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     aarch64
 Temperature  |     25 °C / 77 °F : Who put me in the freezer!
 Governor     |     schedutil

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1416 MHz      1416 MHz    1800 MHz
 CPU1         |      1416 MHz      1416 MHz    1800 MHz
 CPU2         |      1416 MHz      1416 MHz    1800 MHz
 CPU3         |      1416 MHz      1416 MHz    1800 MHz
 CPU4         |      1800 MHz      408 MHz    2256 MHz
 CPU5         |      1800 MHz      408 MHz    2256 MHz
 CPU6         |      1608 MHz      408 MHz    2304 MHz
 CPU7         |      1608 MHz      408 MHz    2304 MHz

[ INFO ] DietPi-CPU_info | The current CPU frequency may be affected by processing this script itself.

Sebastien@DietPi:~$ sudo journalctl -u dietpi-preboot
févr. 16 21:50:46 DietPi systemd[1]: Starting dietpi-preboot.service - DietPi-PreBoot...
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]:  DietPi-CPU_set
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]: ─────────────────────────────────────────────────────
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]:  Mode: Applying CPU governor settings: performance
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]: [ INFO ] DietPi-CPU_set | Setting CPU frequency limits : Max = Disabled MHz | Min = Disabled MHz
févr. 16 21:50:47 DietPi DietPi-PreBoot[1206]: [  OK  ] DietPi-CPU_set | Applied CPU governor settings: performance
févr. 16 21:50:47 DietPi systemd[1]: Finished dietpi-preboot.service - DietPi-PreBoot.
Sebastien@DietPi:~$
```

MichaIng, 2025-02-16:
Okay that is strange. Do you have the `cpufrequtils` package installed, or another possible tool which overrides the CPU governor?
```sh
dpkg -l cpufrequtils
systemctl status cpufrequtils
```

Seb-tech77, 2025-02-17:
I deactivated it and it works fine. Sorry for the inconvenience

MichaIng, 2025-02-17:
Ah great. It otherwise has its config file in `/etc/default/cpufrequtils` IIRC. Probably we should switch to that one instead of using our own tool. `dietpi-preboot` actually has not much to do anymore. I am just not sure whether it supports individual CPU governor settings, like sampling rate and throttle up frequency for ondemnd governor 🤔.