MichaIng/DietPi #7400: Cannot create user because of invalid option `-g` on `groupadd`

kamikazechaser, 2025-02-11:
### Creating a bug report/issue
- [x] I have searched the existing open and closed issues

#### Required Information
- DietPi version

```bash
G_DIETPI_VERSION_CORE=9
G_DIETPI_VERSION_SUB=10
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'
``` 

- Distro version: `bookworm`
- Kernel version: `Linux timbersaw 6.6.44-current-sunxi64 #1 SMP Sat Aug  3 06:54:42 UTC 2024 aarch64 GNU/Linux`
- SBC model: `Orange Pi Zero 2W (aarch64)`
- Power supply used: `5v 3A`
- SD card used: `Unknown`

#### Additional Information (if applicable)
- Software title: `groupadd`
- Was the software title installed freshly or updated/migrated? `Login triggers dist upgrade so probably updated`
- Can this issue be replicated on a fresh installation of DietPi? `Yes`

```bash
root@timbersaw:~# groupadd --version
groupadd (GNU sed) 4.9
Packaged by Debian
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Jay Fenlason, Tom Lord, Ken Pizzini,
Paolo Bonzini, Jim Meyering, and Assaf Gordon.

This sed program was built with SELinux support.
SELinux is disabled on this system.

GNU sed home page: <https://www.gnu.org/software/sed/>.
General help using GNU software: <https://www.gnu.org/gethelp/>.
E-mail bug reports to: <bug-sed@gnu.org>.
```

#### Steps to reproduce

1. Run any command that calls group add e.g. `adduser usera`, Installing `docker-ce` or any package that creates a user triggers this as well.
2. Error produced:

```bash
root@timbersaw:~# adduser sohail
Adding user `sohail' ...
Adding new group `sohail' (1001) ...
/sbin/groupadd: invalid option -- 'g'
Usage: /sbin/groupadd [OPTION]... {script-only-if-no-other-script} [input-file]...

  -n, --quiet, --silent
                 suppress automatic printing of pattern space
      --debug
                 annotate program execution
  -e script, --expression=script
                 add the script to the commands to be executed
  -f script-file, --file=script-file
                 add the contents of script-file to the commands to be executed
  --follow-symlinks
                 follow symlinks when processing in place
  -i[SUFFIX], --in-place[=SUFFIX]
                 edit files in place (makes backup if SUFFIX supplied)
  -l N, --line-length=N
                 specify the desired line-wrap length for the `l' command
  --posix
                 disable all GNU extensions.
  -E, -r, --regexp-extended
                 use extended regular expressions in the script
                 (for portability use POSIX -E).
  -s, --separate
                 consider files as separate rather than as a single,
                 continuous long stream.
      --sandbox
                 operate in sandbox mode (disable e/r/w commands).
  -u, --unbuffered
                 load minimal amounts of data from the input files and flush
                 the output buffers more often
  -z, --null-data
                 separate lines by NUL characters
      --help     display this help and exit
      --version  output version information and exit

If no -e, --expression, -f, or --file option is given, then the first
non-option argument is taken as the sed script to interpret.  All
remaining arguments are names of input files; if no input files are
specified, then the standard input is read.

GNU sed home page: <https://www.gnu.org/software/sed/>.
General help using GNU software: <https://www.gnu.org/gethelp/>.
adduser: `/sbin/groupadd -g 1001 sohail' returned error code 1. Exiting.
```

#### Expected behaviour

- User should be created successfully

#### Actual behaviour

- `groupadd` command exits

Joulinar, 2025-02-11:
hmm `groupadd` is a regular Debian command and not maintain by us. @MichaIng any ideas?

kamikazechaser, 2025-02-11:
Reinstalling the `passwd` package fixes this. I'm not sure how this got conflicted. Maybe worth investigating using the image above. But for now I'll close this.

MichaIng, 2025-02-15:
Very strange indeed 😄. I just checked the image, and the SHA256 of `groupadd` there is `ca37fbe3525689b0175776ac2962f574da3c93b992b9f27fd152f94d9cdedc7e`.

The SHA256 of `sed` is `e4c67283545df5469a83c58d21e661455ecf2c1e7a8c4e75369cf684e4eef053`, matching your's. So `groupadd` must have been weirdly replaced but a `sed` different than the one installed on the image.